openwrt/packages #14637: libxml2 - unsupported encoding

joddude, 2021-02-03:
Maintainer: @mhei
Environment: armv5, DNS-325, Alt-F firmware + Entware based on OpenWrt

I tried installing TT-RSS on my NAS and noticed an error while updating some news feeds. 
`libxml error 32 at line 1 (column 44): unsupported encoding windows-1251`

Then I tried to reproduce this error in a simple php file.
```
<?php
$doc = new DOMDocument();
$url = "https://4pda.ru/feed/";
$content = file_get_contents($url);
$doc->loadXML($content);
```
`Warning: DOMDocument::loadXML(): Unsupported encoding windows-1251`

Can such an error occur because libxml compiled without iconv support?
Is there any way to fix this?

micmac1, 2021-02-04:
Maybe just try it out.

```
diff --git a/libs/libxml2/Makefile b/libs/libxml2/Makefile
index 1430dc8..696bed0 100644
--- a/libs/libxml2/Makefile
+++ b/libs/libxml2/Makefile
@@ -27,13 +27,14 @@ PKG_BUILD_PARALLEL:=0
 
 include $(INCLUDE_DIR)/host-build.mk
 include $(INCLUDE_DIR)/package.mk
+include $(INCLUDE_DIR)/nls.mk
 
 define Package/libxml2
   SECTION:=libs
   CATEGORY:=Libraries
   TITLE:=Gnome XML library
   URL:=http://xmlsoft.org/
-  DEPENDS:=+libpthread +zlib
+  DEPENDS:=+libpthread +zlib $(ICONV_DEPENDS)
 endef
 
 define Package/libxml2/description
@@ -80,7 +81,6 @@ CONFIGURE_ARGS += \
        --with-html \
        --without-ftp \
        --without-http \
-       --without-iconv \
        --without-iso8859x \
        --without-legacy \
        --with-output \
```

If it fixes your problem send a pull request to maintainer and ask for it to be included.

mhei, 2021-02-04:
This looks like a good proposal - your testing and feedback would be appreciated.

joddude, 2021-02-06:
If I could do it, then I would do it. That is why I am asking for help.
If someone has everything configured for compilation, it will not take much time. It will take me a very long time to figure everything out from scratch.  And most likely, in the end, nothing will work out.
Well, I'll just hope that someday someone more experienced will pay attention to this problem.

micmac1, 2021-02-06:
If you provided the exact steps to reproduce and it's quick/easy to do I may be tempted to try it out for you. I don't know anything about php. But if you make it easy for me I'll do it.

mhei, 2021-02-06:
I already managed to reproduce this. So I can test your proposal as well.

commit d18692c9 (2021-02-08): libxml2: allow building with iconv support (fixes #14637)

Instead of hard disabling iconv support, just rely on the global setting.
closed 2021-02-08 by mhei