openwrt/packages #14648: keepalived: Back port security parameters fixes to v19.07

pgaufillet, 2021-02-04:
Maintainer: @feckert
Environment: mvebu-cortexa9-linksys_wrt3200acm, v19.07.6

Description: commits 1b1ba71 and a8a1fbf should be back ported to v19.07.

feckert, 2021-02-05:
You are right. I also have the problem with the non existing /run directory. I already fixed this in my own build, so it was no longer on my radar. :+1:
I have staged a pullrequest to fixes this and the commits https://github.com/openwrt/packages/commit/1b1ba710632f8f4850f02c22e83734a6c8b5c41a and https://github.com/openwrt/packages/commit/a8a1fbfa0da2676583f0289e7ed2806a3191800c. I have to adjust commit https://github.com/openwrt/packages/commit/a8a1fbfa0da2676583f0289e7ed2806a3191800c so this one applies correctly. Because there where some config changes.
I am therefore not sure if this is correct. Unfortunately, I don't have a setup running at the moment to test this. Can you please check if the configuration is generated correctly?

pgaufillet, 2021-02-28:
I have struggled to merge the PR in my workspace with deceiving results (I think I applied it incorrectly to my source tree), but that said I don't see any difference between this PR and the patched version I am running. I think you can proceed.

feckert, 2021-03-03:
Is fixed with merge https://github.com/openwrt/packages/commit/8c63f6331cb648557ecaae874de00d7ceb782eae

PR #14661 (2021-03-03): keepalived: backport fixes
Maintainer: me 
Compile tested: no
Run tested: x86_64, APU3, Service started -> warning disappears.

Description:
@pgaufillet Thank you for pointing this out. That the changes that you suggest to backport to the stable branch openwrt-19.07. The commit https://github.com/openwrt/packages/commit/d41a0b75aac67da9267386bde61c898d37fb3f17 must be checked. It applies proper but due a config change, I think this is not correct. I do not like to backport the uci config also. So please test this and give me a feedback if it works or not and tell me what needs to be changed.

 [... cut]
closed 2021-03-03 by feckert