openwrt/packages #14660: dnsmasq option 66 dont work

dipol0, 2021-02-04:
Model Asus RT-AC51U
Architecture MediaTek MT7620A ver:2 eco:6
Firmware Version OpenWrt 19.07.6 r11278-8055e38794 / LuCI openwrt-19.07 branch git-21.029.68283-bf8b0bb
Kernel Version 4.14.215

dnsmasq 2.80-16.3 (i try and dnsmasq-full 2.80-16.3)

I set **option 66** = IP-address (192.168.0.2 for example) but DHCP server always set NEXT-SERVER option = dhcp server address (192.168.0.1)

my config lines /etc.config/dhcp (lan section):

```
list dhcp_option '66,192.168.0.2'
list dhcp_option '67,pxelinux.0'

```

but ipxe show next-server = 192.168.0.1

[image]

and INTEL UNDI PXE-2.1 TFTP timeout
[image]
I use TFTP server on anothe PC (witn ip - 192.168.0.2 not on router (192.168.0.1) with dnsmasq.

dipol0, 2021-02-05:
I try and:

```
list dhcp_option '66,192.168.0.2'
list dhcp_option '150,192.168.0.2'
list dhcp_option '67,pxelinux.0'
```
- dont work

lucize, 2021-02-13:
try this in dnsmasq.conf

```
dhcp-match=set:X86PC,option:client-arch,0
dhcp-boot=tag:X86PC,pxelinux.0,,<your_tftpd_ip_server>
dhcp-boot=pxelinux.0

dhcp-match=set:BC_EFI,option:client-arch,7
dhcp-boot=tag:BC_EFI,ipxe.efi,,<your_tftpd_ip_server>

dhcp-match=set:efi-x86_64,option:client-arch,9
dhcp-boot=tag:efi-x86_64,ipxe.efi,,<your_tftpd_ip_server>

```

luizluca, 2021-02-14:
Here is not the place for core bugs. Sorry, it is not your fault but OpenWrt confusion with two bug tracking systems.

I use this conf that works even for both bios and uefi

```
config vendorclass
        option networkid 'pxebios'
        option vendorclass 'PXEClient:Arch:00000'

config vendorclass
        option networkid 'pxeefi'
        option vendorclass 'PXEClient:Arch:00006'

config vendorclass
        option networkid 'pxeefi'
        option vendorclass 'PXEClient:Arch:00007'

config vendorclass
        option networkid 'pxeefi'
        option vendorclass 'PXEClient:Arch:00009'

config userclass
        option networkid 'ipxe'
        option userclass 'iPXE'

config boot
        option networkid 'pxebios'
        option filename 'ipxe/ipxe.pxe'
        option serveraddress '192.168.3.1'
        option servername 'router.lan3'

config boot
        option networkid 'pxeefi'
        option filename 'ipxe/ipxe.efi'
        option serveraddress '192.168.3.1'
        option servername 'router.lan3'

config boot
        option networkid 'ipxe'
        option filename 'http://router.lan3/tftpboot/ipxe/ipxe.cfg/init.ipxe'
```

In my case, tftpserver is the router, but the option is serveraddress is still declared. Change as you wish.

romanchyla, 2021-03-21:
I don't have an account in the other bugtracking system, sorry.

The 'problem', if it can be called that way, is inside stock `/etc/init.d/dnsmasq` script, with following modifications it works:

```
202,209d200
< append_dhcp_boot() {
<         xappend "--dhcp-boot=$1"
< }
< 
< append_dhcp_match() {
<         xappend "--dhcp-match=$1"
< }
< 
---
> 
906,909c897
< 	#append_parm "$cfg" "dhcp_boot" "--dhcp-boot"
<         config_list_foreach "$cfg" "dhcp_boot" append_dhcp_boot
<         config_list_foreach "$cfg" "dhcp_match" append_dhcp_match
<         #append_parm "$cfg" "dhcp_match" "--dhcp-match"
---
> 	append_parm "$cfg" "dhcp_boot" "--dhcp-boot"
```

basically, it looks as if somebody forgot to include the `dhcp_match` params (they are parsed); and `dhcp_boot` only accepts one line

this is the `uci` configuration i'm using (without the modifications, only one option - and only dhcp-boot - can be set via uci):

```
            'uci add_list dhcp.@dnsmasq[0].dhcp_match=set:bios,60,PXEClient:Arch:00000',
            'uci add_list dhcp.@dnsmasq[0].dhcp_boot=tag:bios,netboot.xyz.kpxe,,{}'.format(PXEMachine.ipaddress),
            'uci add_list dhcp.@dnsmasq[0].dhcp_match=set:efi32,60,PXEClient:Arch:00002',
            'uci add_list dhcp.@dnsmasq[0].dhcp_boot=tag:efi32,netboot.xyz.efi,,{}'.format(PXEMachine.ipaddress),

```

but yes, the easiest way is to modify `/etc/dnsmasq.conf` -- I just hope somebody with higher powers can update the stock dnsmasq script

pprindeville, 2021-03-22:
Please use context diffs.

romanchyla, 2023-05-30:
I was reminded of this issue while setting up openwrt 22.03

There now exists a new option to specify list of PXE options, and the dnsmasq init script was updated too.

So if you use a tag in the filename; you can achieve the desired results

Here is part of my `/etc/config/dhcp`

```
config boot
	option serveraddress '192.168.3.1'
	option servername 'openwrt'
	option filename 'tag:bios,netboot.xyz.kpxe'

config boot
	option serveraddress '192.168.3.1'
	option servername 'openwrt'
	option filename 'tag:efi,netboot.xyz.efi'

config match
	option networkid 'bios'
	option match '60,PXEClient:Arch:00000'

config match
	option networkid 'efi'
	option match '60,PXEClient:Arch:00002'

```

`/etc/init.d/netmasq restart` then produces desirable:

```
cat /var/etc/dnsmasq.conf.cfg01411c 
# auto-generated config file from /etc/config/dhcp

....
dhcp-boot=tag:bios,netboot.xyz.kpxe,openwrt,192.168.3.1
dhcp-boot=tag:efi,netboot.xyz.efi,openwrt,192.168.3.1
dhcp-match=set:bios,60,PXEClient:Arch:00000
dhcp-match=set:efi,60,PXEClient:Arch:00002

```

netcrystals, 2024-01-17:
I can confirm that dnsmasq does not use the option 66. I have absolutely no clue why not. Tried to figure this out in the internet. Is this a dnsmasq bug? Are there other DHCP servers which correctly use option 66?
However, the proposed solution with **config boot** in OpenWRT works. So, if someone uses OpenWRT based systems, use this method.
If someone could explain me why dnsmasq does not propagate ip addresse defined by option 66, please contact me. This problem cost me days and weeks of my life.
Thanks.

rob-wilco, 2025-05-13:
Sorry to revive an old thread. I ran into the same behavior - what worked for me was navigating to Network > DHCP and DNS > click the PXE/TFTP tab and set the applicable options there. Save and apply and it worked immediately me, clients now getting the correct TFTP / next server address. Hopefully if anyone else stumbles across this thread it solves their problem too!

DarthJahus, 2025-11-13:
@netcrystals, I face the same issue as you. I've set 66,IP, but it's only set in sname, not in siaddr. And clients try to get the file defined by option 67 from the router / DNS server itself.