openwrt/packages #14662: meson: Incorrect detection of cpu subtype value

zokl, 2021-02-05:
@neheb

I try to compile my own package with meson cross-build function but I have a problem with CPU subtype specification. If I try it by hand. Host_machine configuration which works is:

```
[host_machine]
system = 'linux'
cpu_family = 'x86_64'
cpu = 'native'
endian = 'little'
```
but meson.mk generates this one:

```
[host_machine]
system = 'linux'
cpu_family = 'x86_64'
cpu = 'x86_64'
endian = 'little'
```

The problem is in meson.mk:

```
define Meson/CreateCrossFile
	$(STAGING_DIR_HOST)/bin/sed \
		-e "s|@CC@|$(foreach BIN,$(TARGET_CC),'$(BIN)',)|" \
		-e "s|@CXX@|$(foreach BIN,$(TARGET_CXX),'$(BIN)',)|" \
		-e "s|@AR@|$(TARGET_AR)|" \
		-e "s|@STRIP@|$(TARGET_CROSS)strip|" \
		-e "s|@NM@|$(TARGET_NM)|" \
		-e "s|@PKGCONFIG@|$(PKG_CONFIG)|" \
		-e "s|@CFLAGS@|$(foreach FLAG,$(TARGET_CFLAGS) $(EXTRA_CFLAGS) $(TARGET_CPPFLAGS) $(EXTRA_CPPFLAGS),'$(FLAG)',)|" \
		-e "s|@CXXFLAGS@|$(foreach FLAG,$(TARGET_CXXFLAGS) $(EXTRA_CXXFLAGS) $(TARGET_CPPFLAGS) $(EXTRA_CPPFLAGS),'$(FLAG)',)|" \
		-e "s|@LDFLAGS@|$(foreach FLAG,$(TARGET_LDFLAGS) $(EXTRA_LDFLAGS),'$(FLAG)',)|" \
		-e "s|@ARCH@|$(MESON_ARCH)|" \
		-e "s|@CPU@|$(CONFIG_TARGET_SUBTARGET)|" \
		-e "s|@ENDIAN@|$(if $(CONFIG_BIG_ENDIAN),big,little)|" \
		< $(MESON_DIR)/openwrt-cross.txt.in \
		> $(1)
endef
```

where `-e "s|@CPU@|$(CONFIG_TARGET_SUBTARGET)|" \` is takes from CPU_SUBTARGET.

Is it possible to change this value before compilation? On meson pages https://mesonbuild.com/Cross-compilation.html. They wrote that _cpu_ is highly platform and OS-dependent and for many purposes will be better put inside manually. If I put into meson.mk keyword 'native' manually, my compilation works...

zokl, 2021-02-05:
Hello,

solved by using this retyping in my Makefile: CONFIG_TARGET_SUBTARGET:=native

Thank you, Zbynek

closed 2021-02-05 by zokl

neheb, 2021-02-05:
ping @dhewg 

That sounds like a bug.

dhewg, 2021-02-06:
I'm not sure, `cpu_family` is the important part, and afaict there're no reports that any in-tree package fails with `x86_64` and `64`. `native` sounds wrong too though, because in your case your're probably not even cross compiling.

What does it look like when it doesn't work?

zokl, 2021-02-06:
Hello,

I am preparing a package for DPDK integration into OpenWRT. It has a lot of requirements and uses meson as a compilation framework https://mesonbuild.com/Cross-compilation.html. For the compilation and using is necessary to have several CPU features, for that, I need passthrough access to the builders CPU. In DPDK is solved by native value in cpu parameter in the host_compile section.

This Makefile works for me - https://github.com/k13132/openwrt-dpdk/blob/main/packages/dpdk/Makefile. 

The target platform is PCEngines APU based on x86_64 with glibc.

dhewg, 2021-02-06:
On the other hand `CONFIG_TARGET_SUBTARGET` is definitely not the right thing to use here, that's an internal OpenWRT thing like `nand` or `tiny`.

Does the attached patch work for you?
[meson_cpu.patch.txt](https://github.com/openwrt/packages/files/5936393/meson_cpu.patch.txt)

zokl, 2021-02-06:
Here is without native variable

```
Using 'PKG_CONFIG_PATH' from environment with value: '/mnt/data/ftester/ethershaper_ng/staging_dir/target-x86_64_glibc/usr/lib/pkgconfig:/mnt/data/ftester/ethershaper_ng/staging_dir/target-x86_64_glibc/usr/share/pkgconfig'
The Meson build system
Version: 0.56.0
Source dir: /mnt/data/ftester/ethershaper_ng/build_dir/target-x86_64_glibc/dpdk-20.11
Build dir: /mnt/data/ftester/ethershaper_ng/build_dir/target-x86_64_glibc/dpdk-20.11/openwrt-build
Build type: cross build
Program cat found: YES (/usr/bin/cat)
WARNING: Unknown options: "build.c_args, build.c_link_args, build.cpp_args, build.cpp_link_args"
The value of new options can be set with:
meson setup <builddir> --reconfigure -Dnew_option=new_value ...
Project name: DPDK
Project version: 20.11.0
Using 'CFLAGS' from environment with value: ''
Using 'LDFLAGS' from environment with value: ''
C compiler for the host machine: x86_64-openwrt-linux-gnu-gcc (gcc 8.4.0 "x86_64-openwrt-linux-gnu-gcc (OpenWrt GCC 8.4.0 r15676-c3959cd54f) 8.4.0")
C linker for the host machine: x86_64-openwrt-linux-gnu-gcc ld.bfd 2.34
C compiler for the build machine: gcc (gcc 8.4.1 "gcc (GCC) 8.4.1 20200928 (Red Hat 8.4.1-1)")
C linker for the build machine: gcc ld.bfd 2.30-90
Build machine cpu family: x86_64
Build machine cpu: x86_64
Host machine cpu family: x86_64
Host machine cpu: x86_64
Target machine cpu family: x86_64
Target machine cpu: 64
Program pkg-config found: YES (/mnt/data/ftester/ethershaper_ng/staging_dir/host/bin/pkg-config)
Program gen-pmdinfo-cfile.sh found: YES (/mnt/data/ftester/ethershaper_ng/build_dir/target-x86_64_glibc/dpdk-20.11/buildtools/gen-pmdinfo-cfile.sh)
Program list-dir-globs.py found: YES (/mnt/data/ftester/ethershaper_ng/build_dir/target-x86_64_glibc/dpdk-20.11/buildtools/list-dir-globs.py)
Program check-symbols.sh found: YES (/mnt/data/ftester/ethershaper_ng/build_dir/target-x86_64_glibc/dpdk-20.11/buildtools/check-symbols.sh)
Program options-ibverbs-static.sh found: YES (/mnt/data/ftester/ethershaper_ng/build_dir/target-x86_64_glibc/dpdk-20.11/buildtools/options-ibverbs-static.sh)
Program binutils-avx512-check.sh found: YES (/mnt/data/ftester/ethershaper_ng/build_dir/target-x86_64_glibc/dpdk-20.11/buildtools/binutils-avx512-check.sh)
Program python3 found: YES (/mnt/data/ftester/ethershaper_ng/staging_dir/host/bin/python)
Program cat found: YES (/usr/bin/cat)
Program ../buildtools/symlink-drivers-solibs.sh found: YES (/bin/sh /mnt/data/ftester/ethershaper_ng/build_dir/target-x86_64_glibc/dpdk-20.11/config/../buildtools/symlink-drivers-solibs.sh)
Checking for size of "void *" : 8
Checking for size of "void *" : 8
Library m found: YES
Library numa found: YES
Has header "numaif.h" : YES
Library libfdt found: NO
Found pkg-config: /mnt/data/ftester/ethershaper_ng/staging_dir/host/bin/pkg-config (1.7.3)
Found CMake: /mnt/data/ftester/ethershaper_ng/staging_dir/host/bin/cmake (3.19.1)
Run-time dependency libbsd found: NO (tried pkgconfig and cmake)
Using 'PKG_CONFIG_PATH' from environment with value: '/mnt/data/ftester/ethershaper_ng/staging_dir/target-x86_64_glibc/usr/lib/pkgconfig:/mnt/data/ftester/ethershaper_ng/staging_dir/target-x86_64_glibc/usr/share/pkgconfig'
Run-time dependency libpcap found: YES 1.9.1
Has header "pcap.h" with dependency libpcap: YES
Compiler for C supports arguments -Wextra: YES
../../../../build_dir/target-x86_64_glibc/dpdk-20.11/config/meson.build:231: WARNING: Consider using the built-in warning_level option instead of using "-Wextra".
Compiler for C supports arguments -Wcast-qual: YES
Compiler for C supports arguments -Wdeprecated: YES
Compiler for C supports arguments -Wformat: YES
Compiler for C supports arguments -Wformat-nonliteral: YES
Compiler for C supports arguments -Wformat-security: YES
Compiler for C supports arguments -Wmissing-declarations: YES
Compiler for C supports arguments -Wmissing-prototypes: YES
Compiler for C supports arguments -Wnested-externs: YES
Compiler for C supports arguments -Wold-style-definition: YES
Compiler for C supports arguments -Wpointer-arith: YES
Compiler for C supports arguments -Wsign-compare: YES
Compiler for C supports arguments -Wstrict-prototypes: YES
Compiler for C supports arguments -Wundef: YES
Compiler for C supports arguments -Wwrite-strings: YES
Compiler for C supports arguments -Wno-address-of-packed-member: NO
Compiler for C supports arguments -Wno-packed-not-aligned: YES
Compiler for C supports arguments -Wno-missing-field-initializers: YES

../../../../build_dir/target-x86_64_glibc/dpdk-20.11/config/x86/meson.build:14:6: ERROR: Could not get define '__SSE4_2__'

A full log can be found at /mnt/data/ftester/ethershaper_ng/build_dir/target-x86_64_glibc/dpdk-20.11/openwrt-build/meson-logs/meson-log.txt
make[2]: *** [Makefile:87: /mnt/data/ftester/ethershaper_ng/build_dir/target-x86_64_glibc/dpdk-20.11/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 1
make[2]: Leaving directory '/mnt/data/ftester/ethershaper_ng/feeds/dpdkrepo/packages/dpdk'
time: package/feeds/dpdkrepo/dpdk/compile#4.17#4.35#8.57
    ERROR: package/feeds/dpdkrepo/dpdk failed to build.
make[1]: *** [package/Makefile:114: package/feeds/dpdkrepo/dpdk/compile] Error 1
make[1]: Leaving directory '/mnt/data/ftester/ethershaper_ng'
make: *** [/mnt/data/ftester/ethershaper_ng/include/toplevel.mk:234: package/dpdk/compile] Error 2
```

zokl, 2021-02-06:
> On the other hand `CONFIG_TARGET_SUBTARGET` is definitely not the right thing to use here, that's an internal OpenWRT thing like `nand` or `tiny`.
> 
> Does the attached patch work for you?
> [meson_cpu.patch.txt](https://github.com/openwrt/packages/files/5936393/meson_cpu.patch.txt)

Your patch produce zero value:

```
Using 'CFLAGS' from environment with value: ''
Using 'LDFLAGS' from environment with value: ''
C compiler for the host machine: x86_64-openwrt-linux-gnu-gcc (gcc 8.4.0 "x86_64-openwrt-linux-gnu-gcc (OpenWrt GCC 8.4.0 r15676-c3959cd54f) 8.4.0")
C linker for the host machine: x86_64-openwrt-linux-gnu-gcc ld.bfd 2.34
C compiler for the build machine: gcc (gcc 8.4.1 "gcc (GCC) 8.4.1 20200928 (Red Hat 8.4.1-1)")
C linker for the build machine: gcc ld.bfd 2.30-90
Build machine cpu family: x86_64
Build machine cpu: x86_64
Host machine cpu family: x86_64
Host machine cpu: x86_64
Target machine cpu family: x86_64
Target machine cpu:
Program pkg-config found: YES (/mnt/data/ftester/ethershaper_ng/staging_dir/host/bin/pkg-config)
Program gen-pmdinfo-cfile.sh found: YES (/mnt/data/ftester/ethershaper_ng/build_dir/target-x86_64_glibc/dpdk-20.11/buildtools/gen-pmdinfo-cfile.sh)
Program list-dir-globs.py found: YES (/mnt/data/ftester/ethershaper_ng/build_dir/target-x86_64_glibc/dpdk-20.11/buildtools/list-dir-globs.py)
Program check-symbols.sh found: YES (/mnt/data/ftester/ethershaper_ng/build_dir/target-x86_64_glibc/dpdk-20.11/buildtools/check-symbols.sh)
Program options-ibverbs-static.sh found: YES (/mnt/data/ftester/ethershaper_ng/build_dir/target-x86_64_glibc/dpdk-20.11/buildtools/options-ibverbs-static.sh)
Program binutils-avx512-check.sh found: YES (/mnt/data/ftester/ethershaper_ng/build_dir/target-x86_64_glibc/dpdk-20.11/buildtools/binutils-avx512-check.sh)
Program python3 found: YES (/mnt/data/ftester/ethershaper_ng/staging_dir/host/bin/python)
Program cat found: YES (/usr/bin/cat)
Program ../buildtools/symlink-drivers-solibs.sh found: YES (/bin/sh /mnt/data/ftester/ethershaper_ng/build_dir/target-x86_64_glibc/dpdk-20.11/config/../buildtools/symlink-drivers-solibs.sh)
Checking for size of "void *" : 8
Checking for size of "void *" : 8
```

dhewg, 2021-02-06:
> Your patch produce zero value:

I guess OpenWRT doesn't set it for x86_64. But there's no error in that log, does it work anyway?

If you remove that patch again and try with the initial `64`, what does the error log contain? (/mnt/data/ftester/ethershaper_ng/build_dir/target-x86_64_glibc/dpdk-20.11/openwrt-build/meson-logs/meson-log.txt)

zokl, 2021-02-06:
> > Your patch produce zero value:
> 
> I guess OpenWRT doesn't set it for x86_64. But there's no error in that log, does it work anyway?
> 

Crash as previous with the same problem `../../../../build_dir/target-x86_64_glibc/dpdk-20.11/config/x86/meson.build:14:6: ERROR: Could not get define '__SSE4_2__'`

dhewg, 2021-02-06:
Please paste the meson log, it looks like dpdk passes that cross file cpu value to gcc via -mcpu=, which I'm not convinced is valid

dhewg, 2021-02-06:
How about this one?
[meson_cpu-v2.patch.txt](https://github.com/openwrt/packages/files/5936508/meson_cpu-v2.patch.txt)

zokl, 2021-02-06:
> I guess OpenWRT doesn't set it for x86_64. But there's no error in that log, does it work anyway?
>
> If you remove that patch again and try with the initial `64`, what does the error log contain? [...]

Here is a requested log ...
[meson-log.txt](https://github.com/openwrt/packages/files/5936808/meson-log.txt)

zokl, 2021-02-06:
> How about this one?
> [meson_cpu-v2.patch.txt](https://github.com/openwrt/packages/files/5936508/meson_cpu-v2.patch.txt)

The same problem.
[meson-log_patchv2.txt](https://github.com/openwrt/packages/files/5936818/meson-log_patchv2.txt)

dhewg, 2021-02-06:
Well, this is getting ridiculous. My last attempt:
[meson_cpu-v3.patch.txt](https://github.com/openwrt/packages/files/5936975/meson_cpu-v3.patch.txt)

But I think dpdk can't just use that value for -mcpu=, -march= or -mtune= or whatever they're using the value for. If that patch doesn't work I think patching dpdk is your best bet.

zokl, 2021-02-06:
> Well, this is getting ridiculous. My last attempt:
> [meson_cpu-v3.patch.txt](https://github.com/openwrt/packages/files/5936975/meson_cpu-v3.patch.txt)
>
> But I think dpdk can't just use that value for [...]

Dpdk needs one of this cpu type:

nocona core2 nehalem corei7 westmere sandybridge corei7-avx ivybridge core-avx-i haswell core-avx2 broadwell skylake skylake-avx512 cannonlake icelake-client icelake-server bonnell atom silvermont slm knl knm x86-64 eden-x2 nano nano-1000 nano-2000 nano-3000 nano-x2 eden-x4 nano-x4 k8 k8-sse3 opteron opteron-sse3 athlon64 athlon64-sse3 athlon-fx amdfam10 barcelona bdver1 bdver2 bdver3 bdver4 znver1 btver1 btver2 native

best solution for me is rewrite the Meson_CPU variable to native ...

dhewg, 2021-02-06:
`native` might fix it for you locally, but it definitely isn't the right fix to merge

zokl, 2021-02-06:
> `native` might fix it for you locally, but it definitely isn't the right fix to merge
Yes, you have true. But i think this problem will be in other x86 project with dependencies to cpu architecture. You need to know which features could you use before you compile the project. Rewriting one variable is quite good solution. Thank you veru much for you help. 

Do you update the meson.mk based on our discussion?

dhewg, 2021-02-06:
depends, does v3 work for you? I didn't test that at all if it regresses other packages/archs

zokl, 2021-02-06:
> depends, does v3 work for you? I didn't test that at all if it regresses other packages/archs

Yes, patch3 does not work too. The same problem. The problem is, that all platform based on x86_64 has the same CPU architecture profile. Others like arm or mips differ based on used cpu. 

Properly solution will be add CPU profiles into x86_64 architecture in the same way as in ARM or MIPS. Without it the workaround of my issue is CONFIG_TARGET_SUBTARGET:=native or CONFIG_TARGET_SUBTARGET:=btver2 in case of my PC Engines APU board ...

dhewg, 2021-02-06:
v3 sets the cpu type to `x86-64 `, which is part of the list you pasted. Sp that should work, but it seems dpdk does more weird stuff with that value.

commit a7865a4a (2021-03-18): meson: fix host cpu setting

CONFIG_TARGET_SUBTARGET is bogus for this, it can contain values such as
'nand' or 'tiny', use the real cpu type instead.

Fixes #14662.