openwrt/packages #14671: Commit Access?

PolynomialDivision, 2021-02-06:
Could you imagine giving commit access in the near future?

**Why?**

I am part of the ["Freifunk Berlin"](https://berlin.freifunk.net/) group, which is a "free wireless community mesh". Unfortunately, there was a development that people moved away from Upstream-OpenWrt. I didn't like this development. That's why I made a fork and tried to close the gap between OpenWrt and the "Freifunk firmware" again. There is still a lot to do to close the gap completely, e.g. we have to rewrite all our luci-apps in javascript. However, I was able to convince a few people and we have now interesting development:

- Ansible For Building Automatic Images
- Own OPKG-Feed
- Using OpenWrt-SDK to Build our Images

I am in favor of getting more and more upstream (of course only useful tools that everyone can use). Further, I always urge people to find upstream solutions. Nevertheless, some people are against it because they fear they will no longer have sufficient control over their own packages. That is the main reason why I would like to have commit access, because then I can always point out to the people that I do have commit access. Another reason is simply that I really enjoy the development and community and plan to spend more of my free looking at this GitHub repro. ;)

**Contributions**

I mainly contribute to
- collectd
- owipcalc
- prometheus-node-exporter-lua

"Own" packages
- dawn
- sipcalc
- libradiotap
- libuwifi 
- generate-ipv6-addresses
- ... (stuff that I was not able to port successfully on OpenWrt for now, e.g. xdp-tools)

**Openwrt**
You can see my commits here:
- https://git.openwrt.org/?p=openwrt%2Fopenwrt.git&a=search&h=HEAD&st=author&s=Nick+Hainke
- kernel(-patches)
- netifd
- odhcpd
- hostapd
- ath79, ...
- umdns

You may know me of this "decentralized wireless controller" (dawn), since I started to work on radio resource management in OpenWrt/hostapd. This 802.11v/k extensions to steer clients and stuff like this. I wrote here some shitty blog entry you can read (but a bit outdated: https://polynomialdivision.github.io/)

Currently, I was working on ipv6 (since we introduce it in our mesh network), xdp and segment-routing (ipv6 feature).

**Routing-Feed**
- babeld (adding ubus bindings)
- olsr
- olsrv2

**Luci**
- own luci apps (dawn...)
- luci-app-statistics

https://github.com/Freifunk-Spalter

I think there would not be that much difference at first, except that I could contribute to my own packages faster and review other PRs officially. Not that I have a problem with the current state, because I really value this good working community. I also like to look through open issues. When I have more time again, I will be happy to go through pull-requests and review them as well.

Maybe if you do not feel comfortable, you could say me what else you expect. :)

dangowrt, 2021-02-06:
Not much thinking required on that one, I think it's obvious that giving you access to packages will make things better and easier for everyone involved. So welcome!
You should have received an invite to the package-maintainers group. That gives you commit access to packages as well as packages-abandonned. For access to luci and routing feeds you got to ask the teams there.

closed 2021-02-06 by dangowrt

champtar, 2021-02-06:
Not even time to say +1. Just keep up the good work !

PolynomialDivision, 2021-02-06:
Thanks for the nice words! :)