openwrt/packages #14674: tangd: Cannot change port of xinetd service

rugk, 2021-02-06:
Maintainer: @Tiboris @neheb @cotequeiroz
Environment: OpenWrt 19.07.6 r11278-8055e38794, GL.iNet GL-B1300

Description: If you change the port of the xinetd to anything else (which you'd likely actually may want to do if you want to use a port differently etc.), you get an error at the xinetdx startup.

## STR

1. Edit the file [`/etc/xinetd.d/tangdx`](https://github.com/openwrt/packages/blob/master/utils/tang/files/tangdx).
2. Change the port to 8889 e.g.:
   ```
   service tangd
   {
       port            = 8889
       socket_type     = stream
       wait            = no
       user            = root
       server          = /usr/libexec/tangdw
       server_args     = /usr/share/tang/cache
       log_on_success  += USERID
       log_on_failure  += USERID
       disable         = no
   }
   ```
3. Now starting xinetdx (`service xinetd start` or `service xinetd restart`) 

### What happens

It logs this into the system log:
> Sat Feb  6 16:46:47 2021 daemon.debug xinetd[10847]: Reading included configuration file: /etc/xinetd.d/tangdx [file=/etc/xinetd.conf] [line=6]
Sat Feb  6 16:46:47 2021 daemon.err xinetd[10847]: Service tangd expects port 8888, not 8889
Sat Feb  6 16:46:47 2021 daemon.crit xinetd[10847]: 10847 {init_services} no services. Exiting...

### What should happen

AFAIK this has worked in the past.
Of course it should (freely) let me change the port to whatever I want.
There is no reason to require port 8888, [e.g. see this for comparison](https://bugzilla.redhat.com/show_bug.cgi?id=1650909).

Tiboris, 2021-02-08:
Heloo @rugk i believe you need to change this missing piece in `/etc/services` as installation does for you.
https://github.com/openwrt/packages/pull/5447/files#diff-448a2884ef422a3550078f305e2eb4a1ace5d69551469f8d98c6427215f7a636R63 
```bash
echo -e "tangd\t\t8889/tcp" >> /etc/services
```

rugk, 2021-02-09:
Thanks a lot, this works!

closed 2021-02-09 by rugk

Tiboris, 2021-02-10:
@rugk yw glad it helped.