openwrt/packages #14675: odhcpd: doesn't set M flag nor O flag in router advertisment when using isc-dhcp-server-ipv6

eltorio, 2021-02-06:
Hi,
On OpenWRT 19.07.6 I config my /etc/config/dhcp like this
config dhcp 'lan'
        option interface 'lan'
        option start '100'
        option limit '150'
        option leasetime '12h'
        option ra 'server'
        option ra_mininterval '1'
        option ra_maxinterval '20'
        option ra_management '2'

but odhcpd does not set M flag nor O flag
I tried to add option ignore '1' without success
This is a Wireshark capture
[image]

PolynomialDivision, 2021-02-06:
I think odhpcd is part of the core and not the packages feed. You should bug report here: https://bugs.openwrt.org/

eltorio, 2021-02-07:
> I think odhpcd is part of the core and not the packages feed. You should bug report here: https://bugs.openwrt.org/

Thanks for your help, you're probably right so I create a new task in the base system:
[https://bugs.openwrt.org/index.php?do=details&task_id=3609](url)

closed 2021-02-09 by BKPepe