openwrt/packages #14692: tang: bump to v8

rugk, 2021-02-08:
Maintainer: @Tiboris @neheb @cotequeiroz

Description:
Tang has a new v8 release: https://github.com/latchset/tang/releases/tag/v8

Can we bump that?

**Important:** It changed how the keys are generated, so you also need to adjust (actually _remove_) tang-update and the whole cache dir. See https://github.com/latchset/tang/pull/53

rugk, 2021-05-05:
Oh it seems v8 has been released with https://github.com/openwrt/packages/commit/b9236dc173ba93dc7531103beb8eda389348ec4f

v10 has now been released however: https://github.com/latchset/tang/releases/tag/v10

kinafu, 2021-06-09:
[This commit](https://github.com/openwrt/packages/commit/a02d79327bdaa2bb80a884ffce5f0ec71bf709d8) seems to have bumped the version to v10.

kinafu, 2021-06-09:
Even tough the version number was bumped to [v10](https://github.com/openwrt/packages/issues/14692), I believe the installation should fail for users wanting to install tang freshly. (I cannot try it on my own since I do not have an openwrt environment.)
A change to the Makefile has to be made to reflect [the changes in the tang package](https://github.com/latchset/tang/pull/53).

I am trying to install the package by the third party package manager `entware`, which provides its packages in sync with the openwrt packages. The [`entware` tang package](https://github.com/Entware/entware-packages/blob/616cb55ab9e33ffb151a363870cfc26695fd7977/utils/tang/Makefile) currently does not work. 
And since the `entware` Makefile should equal the one from `openwrt`, it should not work in `openwrt`, too, I believe.
The [issue I encountered upon the installation ](https://github.com/Entware/Entware/issues/652) was:
`/opt/libexec/tangd-update: not found`

`tangd-update` and the `cache` directory are not being used anymore, see the reference to the tang repo by @rugk.
But this change has not yet been applied to the Makefile in `openwrt`.

rugk, 2021-06-09:
> The commit seems to have bumped the version to v10.

You did not link a commit?

But ah, I found it: https://github.com/openwrt/packages/commit/a02d79327bdaa2bb80a884ffce5f0ec71bf709d8

As for the tang-update thing, yes, I did not try a fresh installation, but is certainly not there anymore, @neheb. So let's remove that?

rugk, 2021-06-12:
Okay, so seems to be fixed by https://github.com/openwrt/packages/pull/15814

BTW @neheb you can (automatically) [let issues close](https://help.github.com/articles/closing-issues-via-commit-messages/) when a PR is merged by adding some "magic" text to your PR _body_.

closed 2021-06-12 by rugk