openwrt/packages #14706: pagekitec error when restarting the router

predators46, 2021-02-09:
before i competed on pagekite i tried it free. and I tried I found the error below.

`Wed Feb 10 07:17:21 2021 user.err pagekitec: tid=77a71d74; ll=29; msg=pkmanager.c: Rejected as a duplicate by front-end
Wed Feb 10 07:17:21 2021 user.notice pagekitec: tid=77a71d74; ll=2a; msg=Network is up but kites are not ready.
Wed Feb 10 07:18:21 2021 user.notice pagekitec: tid=77ab7d74; ll=2b; msg=Tick!  [repeating=yes, next=86, status=50, tunnels=0, v=0.91.200718C]
Wed Feb 10 07:18:21 2021 user.notice pagekitec: tid=77a94d74; ll=2c; msg=Connecting to [69.164.211.158]:443 (session=602317ff:d1c5c35eadc845c3cc80e88d5701e5a30ddb6f74, in DNS)
Wed Feb 10 07:18:22 2021 user.notice pagekitec: tid=77a94d74; ll=2d; msg=7: TLSv1.3 connection established: TLS_CHACHA20_POLY1305_SHA256 TLSv1.3 Kx=any Au=any Enc=CHACHA20/POLY1305(256) Mac=AEAD
Wed Feb 10 07:18:23 2021 user.notice pagekitec: tid=77a94d74; ll=2e; msg=Connecting to [69.164.211.158]:443 (session=6023183f:16706dd51e5c91f4cdb87cb91769a17eea6bec31, in DNS)
Wed Feb 10 07:18:24 2021 user.notice pagekitec: tid=77a94d74; ll=2f; msg=7: TLSv1.3 connection established: TLS_CHACHA20_POLY1305_SHA256 TLSv1.3 Kx=any Au=any Enc=CHACHA20/POLY1305(256) Mac=AEAD
Wed Feb 10 07:18:26 2021 user.notice pagekitec: tid=77a94d74; ll=30; msg=Connect failed: -1  `

predators46, 2021-02-10:
seems like pagekitec often experience disconnection
[image]

karlp, 2021-02-10:
What version are you running? (of openwrt and pagekitec?) does it resolve itself after some time? or require a manual restart every time you restart the router?

predators46, 2021-02-10:
> What version are you running? (of openwrt and ~pagekitec~ (0.91.200718C) ?) does it resolve itself after some time? or require a manual restart every time you restart the router?

Thank you for your reply. I have tried the 2007 and 2011 versions all the same the connection is lost and temporarily sometimes the connection will connect itself without restarting and sometimes I have to restart pagekitec to be able to reconnect

predators46, 2021-02-10:
> What version are you running? (of openwrt and ~pagekitec~ (0.91.200718C) ?) does it resolve itself after some time? or require a manual restart every time you restart the router?

I use luci-app-pagekitec. and I found the problem quite strange so I can reconnect I have to go to openwrt using LAN or wifi on my router and open my router ip address, a few minutes later pagekitec will reconnect and sometimes pagekitec does not reconnect. It looks like Pagekitec isn't perfect walking in Luci

karlp, 2021-02-10:
"open my ip address"? I'm not really following...  does your router regularly get new ip addresses? I've asked the pagekite people to take a look at your log, but it's a little hard to follow what's going on right now.

predators46, 2021-02-10:
> "open my ip address"? I'm not really following... does your router regularly get new ip addresses? I've asked the pagekite people to take a look at your log, but it's a little hard to follow what's going on right now.

I will speak to be clear. the problem that I found pagekitec a few minutes later the connection was lost I can only wait for pagekitec to reconnect without restarting pagekitec. and the worst thing is pagekitec can not connect anymore will experience disconnected connection I have to restart pagekitec again after a few minutes then pagekitec is not connected anymore. and I see system.log on openwrt no logs pagekitec the connection is lost

predators46, 2021-02-10:
> "open my ip address"? I'm not really following... does your router regularly get new ip addresses? I've asked the pagekite people to take a look at your log, but it's a little hard to follow what's going on right now.

you better try it yourself as everyone experiences this when using luci pagekitec. for now I'm wearing zerotier

predators46, 2021-02-10:
> "open my ip address"? I'm not really following... does your router regularly get new ip addresses? I've asked the pagekite people to take a look at your log, but it's a little hard to follow what's going on right now.

Pagekitec often experiences temporarily unavailable

predators46, 2021-02-11:
> "open my ip address"? I'm not really following... does your router regularly get new ip addresses? I've asked the pagekite people to take a look at your log, but it's a little hard to follow what's going on right now.

I'm currently trying to come back but still get a temporarily unavailable error

predators46, 2021-02-11:
I feel like if I run Pagekite with Pyhton it will work better without a temporarily unavailable problem. this is what I did install the screen and run the command below screen pagekite.py 80 yourname.pagekite.me by running the above command pagekitec will run in the background

karlp, 2021-02-11:
You will have to take this up with pagekite project I'm afraid.

predators46, 2021-02-11:
> You will have to take this up with pagekite project I'm afraid.

it's easier to use pagekitec in luci i hope you can improve so that pagekitec can run in the background

karlp, 2021-02-11:
well, I'm running pagekitec in production very happily, so ... this just works.  I don't know what you're seeing.

closed 2021-02-12 by predators46