openwrt/packages #14718: mwan3: iptables user rules chain not populated on boot up

dl12345, 2021-02-11:
Maintainer: @aaronjg @feckert  
Environment: x86_64 glibc, OpenWrt master (custom build)

Immediately after bootup, the mwan3_rules chain in the mangle table is empty

```
Chain mwan3_rules (1 references)
 pkts bytes target     prot opt in     out     source               destination
```

Likewise, the chains that should be created for those rules that have the sticky option set are also missing.

I suspect it might be a race condition of some form, although have spent some time looking at it and can't see what's causing it. 

Issuing `/etc/init.d/mwan3 restart` fixes the problem, albeit not a very elegant solution.

Putting some debug statements in `/lib/mwan3/mwan3.sh`, it's clear that the function `mwan3_set_user_rules()` is not being called during the boot process from the init script. 

When manually restarting, the function is called and thus the rules are added.

Enabling the log debug function, the only debug messages emitted during boot show the caller as being either `mwan3rtmon` or `mwan3-hotplug`. Nothing from `mwan3-init`, unlike when the init script is manually invoked from the command line.

dl12345, 2021-02-11:
And the init script is definitely enabled and it runs. Debug code shows entry into the start_service() function.

dl12345, 2021-02-11:
Located the problem. It was actually a race condition with the creation of the ipsets that mwan3 relies on in my user rules. 4 out of 5 were created, but the 5th one was still apparently in the process of creation and so the mwan3 iptables rule creation failed for all of the rules as a result.

I moved creation of the ipsets much earlier in the boot process and the problem is fixed.

closed 2021-02-11 by dl12345

aaronjg, 2021-02-12:
Thanks! Was there nothing in the error log that indicated the failure? If not, we definitely should add something for this...

feckert, 2021-02-12:
As already mentioned by @aaronjg, it would not be bad, since you have this use case, to tell us where the whole thing happens. We would then have to issue a warning. So that the problem is communicated to the user. Maybe we can make the ipset handling more smarter.

dl12345, 2021-02-12:
It turns out that there was actually a single error logged. 

```
Fri Feb 12 16:53:44 2021 user.err mwan3-init[5172]: set_user_rules: iptables-restore v1.8.7 (legacy): Set zoom doesn't exist.  Error occurred at line: 3 Try `iptables-restore -h' or 'iptables-restore --help' for more information.
```

Having done a some more investigation since I posted the issue, I now have a clearer handle on the cause of the error. I recently upgraded from r13372 to r15724. During that time, it appears that mwan3 has undergone an almost total rewrite and the behaviour of the hotplug handler has changed. 

Previously, I invoked ipset creation, loading and refresh in `mwan3.user`. 

At boot time, I relied on the hotplug handlers to make a call to `mwan3.user` on ifup, which loaded my ipsets and the mwan3 iptables user rules chains. So when start_service() ran, the rules were not loaded initially, but they were loaded shortly after because there are mutiple hotplug calls when the interfaces come up after `/etc/init.d/network` starts. `/etc/hotplug.d/iface/16-mwan3` loads the ipset on ifup of the interfaces, then the next call to `/etc/hotplug.d/iface/15-mwan3` loads the rules correctly and mwan3 is up and running. 

It was arguably a bit ugly to rely on this behaviour. 

Checking out my branch for the r13372 build, I see that this is the old `/etc/hotplug.d/iface/15-mwan3`. Note how it loads the user rules with a call to mwan3_set_user_rules

```
        ifup)
                mwan3_set_general_rules
                mwan3_set_general_iptables
                mwan3_create_iface_iptables $INTERFACE $DEVICE
                mwan3_create_iface_rules $INTERFACE $DEVICE
                mwan3_create_iface_route $INTERFACE $DEVICE
                if [ ${running} -eq 1 -a "${status}" = "online" ]; then
                        $LOG notice "Starting tracker on interface $INTERFACE (${DEVICE:-unknown})"
                        mwan3_set_iface_hotplug_state $INTERFACE "online"
                        mwan3_track $INTERFACE $DEVICE "online" "$src_ip"
                else
                        $LOG notice "Starting tracker on interface $INTERFACE (${DEVICE:-unknown})"
                        mwan3_set_iface_hotplug_state $INTERFACE "offline"
                        mwan3_track $INTERFACE $DEVICE "offline" "$src_ip"
                fi
                mwan3_set_policies_iptables
                mwan3_set_user_rules
        ;;
        ifdown)
                mwan3_set_iface_hotplug_state $INTERFACE "offline"
                mwan3_delete_iface_ipset_entries $INTERFACE
                mwan3_track_signal $INTERFACE $DEVICE
                mwan3_set_policies_iptables
                mwan3_set_user_rules
        ;;
esac
```

The new mwan3 hotplug handler does not call `mwan3_set_user_rules()` anymore. They're loaded once only at boot time or on explicit command line invocation of the init script. 

Arguably, the hotplug handler should call `mwan3_set_user_rules()` because not doing so means that mwan3.user cannot be used to create and load ipsets anymore. They will need to be created and loaded by an explicit init script with higher precedence than mwan3.

```
case "$ACTION" in
    connected)
        mwan3_set_iface_hotplug_state $INTERFACE "online"
        mwan3_set_policies_iptables
        ;;
    ifup)
        mwan3_create_iface_iptables $INTERFACE $DEVICE
        mwan3_create_iface_rules $INTERFACE $DEVICE
        mwan3_set_iface_hotplug_state $INTERFACE "$status"
        if [ "$MWAN3_STARTUP" != "init" ]; then
            mwan3_create_iface_route $INTERFACE $DEVICE
                    mwan3_set_general_rules
            [ "$status" = "online" ] && mwan3_set_policies_iptables
        fi
        [ "$ACTION" = ifup ] && procd_running mwan3 "track_$INTERFACE" && procd_send_signal mwan3 "track_$INTERFACE" USR2
        ;;
    disconnected)
        mwan3_set_iface_hotplug_state $INTERFACE "offline"
        mwan3_set_policies_iptables
        ;;
    ifdown)
        mwan3_set_iface_hotplug_state $INTERFACE "offline"
        mwan3_delete_iface_ipset_entries $INTERFACE
        mwan3_delete_iface_rules $INTERFACE
        mwan3_delete_iface_route $INTERFACE
        mwan3_delete_iface_iptables $INTERFACE
        procd_running mwan3 "track_$INTERFACE" && procd_send_signal mwan3 "track_$INTERFACE" USR1
        mwan3_set_policies_iptables
    ;;
esac
exit 0
```

So this was essentially a configuration error that introduced itself as a result of an unnoticed change in the behaviour of the mwan3 hotplug handler.

The net result with the new mwan3: my ipsets are loaded by the hotplug handler, but the user rules chains are not. 

Other than the single error in the logs, the whole process failed silently. I think it would be a good idea to have a more explicit error in the logs about failure to load the mwan3 iptables user rules. Something along the lines of "Failure to load mwan3 iptables user rules chains", followed by the actual error as per above

@feckert yes, that would be a nice enhancement. Perhaps give mwan3 a configuration option to create any ipsets referenced in a rule with an optional file containing the set data to load? 

Alternatively, have the hotplug handler make a call to `mwan3_set_user_rules()` as it used to do so that `mwan3.user` can be used to create and load the ipsets.

dl12345, 2021-02-21:
@aaronjg @feckert 

My use case is that I use multiple ipsets in user rules, independently of dnsmasq. 

These are ipsets that contain addresses of traffic destined for Zoom and Webex servers that I route out my second link, which is a slower link but has much lower latency than the main link. I also an ipset to route certain traffic through my wireguard interface.

As a result, these sets need to be created and populated manually. 

I made the following modification to my mwan3 to handle the ipset creation so that user rules don't fail to get added because of missing ipsets. If my use case is not uncommon, you may want to consider adding something like this. 

It would need a luci component as well to make it truly user friendly.

```
diff --git a/net/mwan3/files/lib/mwan3/mwan3.sh b/net/mwan3/files/lib/mwan3/mwan3.sh
index 24af3dad5..e7d9dd7a5 100644
--- a/net/mwan3/files/lib/mwan3/mwan3.sh
+++ b/net/mwan3/files/lib/mwan3/mwan3.sh
@@ -148,6 +148,27 @@ mwan3_set_custom_ipset()
        error=$(echo "$update" | $IPS restore 2>&1) || LOG error "set_custom_ipset: $error"
 }
 
+mwan3_create_user_ipset()
+{
+       local update=""
+       local family="inet"
+       local name file
+
+       config_get name $1 name
+       config_get family $1 family
+       config_get file $1 file
+
+       mwan3_push_update -! create $name hash:net family $family
+       [ -f $file ] && for entry in $(cat $file); do
+               mwan3_push_update -! add $name $entry
+       done
+       error=$(echo "$update" | $IPS restore 2>&1) || LOG error "create_user_ipset: $error"
+}
+
+mwan3_create_user_ipsets()
+{
+        config_foreach mwan3_create_user_ipset ipset
+}
 
 mwan3_set_connected_ipv4()
 {
@@ -893,6 +914,8 @@ mwan3_set_user_rules()
        local IPT IPTR ipv
        local current update error
 
+       mwan3_create_user_ipsets
+
        for ipv in ipv4 ipv6; do
                if [ "$ipv" = "ipv4" ]; then
                        IPT="$IPT4"
```

`/etc/config/mwan3` would have entries thus

```
config ipset 'zoom_ipset'
	option name 'zoom'
	option family 'inet'
	option file '/etc/mwan3.d/zoom'
```

and the named files will just contain a list of networks in CIDR format, such as with banIP.

aaronjg, 2021-02-21:
I think allowing ipset creation in mwan3 is a great idea. Is your use case a static list or is it programmatically created? I had a patch that would automatically create the ipset when multiple ips were specified in the routing rule, would that work for your use case? If so, I can update the patch to work with the latest version of mwan3.

#13088

https://github.com/openwrt/packages/commit/4f0089f7d9ec3e10395aff93ed1b9c8eed187322

dl12345, 2021-02-21:
> Is your use case a static list or is it programmatically created? 

5 out of 6 are static and the 6th is a combination of static and dynamic added by dnsmasq. So it get initialized with a list of addresses at boot time and then dnsmasq adds to the same ipset.

Some of them have quite a lot of entries, so not really practical to specify explicitly in a rule. The list of Zoom servers, for example, contains 75 entries.

feckert, 2021-02-21:
In principle, I would also say that we should extend and fix the ipset handling.

The openwrt firewall (fw3) already supports ipset creation with [uci](https://openwrt.org/docs/guide-user/firewall/fw3_configurations/fw3_config_ipset).
Maybe we can hook into it.

Although [mwan3](
https://github.com/openwrt/packages/blob/master/net/mwan3/files/etc/init.d/mwan3#L7) and the [firewall](
https://github.com/openwrt/openwrt/blob/master/package/network/config/firewall/files/firewall.init#L3) have the init number 19, we start alphabetically after the firewall during boot up, so the IPset should already be installed.

dl12345, 2021-02-21:
> The openwrt firewall (fw3) already supports ipset creation with [uci](https://openwrt.org/docs/guide-user/firewall/fw3_configurations/fw3_config_ipset).
> Maybe we can hook into it.

I just tried that and I can confirm that it works. The ipsets are created before mwan3 starts, so hooking into fw3 for this purpose would clearly be better than duplicating ipset handling in mwan3.

Use cases for ipsets in mwan3 user rules that I can think of

- static ipset that gets loaded from a file
- dynamic ipsets that are referenced in both a mwan3 rule and a dnsmasq configuration
- implicit ipsets that get created by mwan3 when multiple IPs are specified in a rule as @aaronjg noted above

Given the existing implementation in fw3, which I didn't even know about (note to self: RTFM - thanks @feckert), maybe what's needed is just a bit of light handling from mwan3 for use case 3 and ultimately a luci firewall tab to manage fw3 ipsets with a couple of help tags on the mwan3 pages to direct the user to the firewall tab for the ipset configuration?