openwrt/packages #14731: geth: can not find -liconv

BKPepe, 2021-02-14:
Maintainer: @ecsv 
Compile tested: Turris Omnia, mvebu (cortex-a9), OpenWrt master

Description:

```
SHELL= flock /foo/build/tmp/.go-ethereum-1.9.22.tar.gz.flock -c '  	/foo/build/scripts/download.pl "/foo/build/dl" "go-ethereum-1.9.22.tar.gz" "cda67de8e76afaad28b8bb4d3e5a95d01f88f69ab2c25964bc8ee2d368794d8c" "" "https://codeload.github.com/ethereum/go-ethereum/tar.gz/v1.9.22?"    '
rm -f /foo/build/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/go-ethereum-1.9.22/.built
touch /foo/build/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/go-ethereum-1.9.22/.built_check
CONFIG_GOLANG_MOD_CACHE_WORLD_READABLE="" GO_BUILD_CACHE_DIR="/foo/build/tmp/go-build" GO_MOD_CACHE_DIR="/foo/build/dl/go-mod-cache" GO_MOD_ARGS="-modcacherw" GO_PKG="github.com/ethereum/go-ethereum" GO_INSTALL_EXTRA="" GO_INSTALL_ALL="" GO_SOURCE_ONLY="" GO_BUILD_PKG="github.com/ethereum/go-ethereum/cmd/geth" GO_EXCLUDES="" GO_GO_GENERATE="" GO_INSTALL_BIN_PATH="/usr/bin" BUILD_DIR="/foo/build/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/go-ethereum-1.9.22" GO_BUILD_DIR="/foo/build/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/go-ethereum-1.9.22/.go_work/build" GO_BUILD_BIN_DIR="/foo/build/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/go-ethereum-1.9.22/.go_work/build/bin/linux_arm" GO_BUILD_DEPENDS_PATH="/usr/share/gocode" GO_BUILD_DEPENDS_SRC="/foo/build/staging_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/usr/share/gocode/src" GOOS="linux" GOARCH="arm" GO386="" GOARM="7" GOMIPS="" GOMIPS64="" CGO_ENABLED=1 CC="ccache_cc" CXX="ccache_cxx" CGO_CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -fmacro-prefix-map=/foo/build/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/go-ethereum-1.9.22=go-ethereum-1.9.22 -Wformat -Werror=format-security -fpic -fstack-protector-strong -D_FORTIFY_SOURCE=2 -Wl,-z,now -Wl,-z,relro -I/foo/build/staging_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/usr/lib/libiconv-full/include -I/foo/build/staging_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/usr/lib/libintl-full/include" CGO_CPPFLAGS="-I/foo/build/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-8.4.0_musl_eabi/usr/include -I/foo/build/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-8.4.0_musl_eabi/include/fortify -I/foo/build/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-8.4.0_musl_eabi/include -I/foo/build/staging_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/usr/lib/libiconv-full/include -I/foo/build/staging_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/usr/lib/libintl-full/include" CGO_CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -fmacro-prefix-map=/foo/build/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/go-ethereum-1.9.22=go-ethereum-1.9.22 -Wformat -Werror=format-security -fpic -fstack-protector-strong -D_FORTIFY_SOURCE=2 -Wl,-z,now -Wl,-z,relro -I/foo/build/staging_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/usr/lib/libiconv-full/include -I/foo/build/staging_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/usr/lib/libintl-full/include" CGO_LDFLAGS="-L/foo/build/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-8.4.0_musl_eabi/usr/lib -L/foo/build/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-8.4.0_musl_eabi/lib -fpic -specs=/foo/build/include/hardened-ld-pie.specs -znow -zrelro -L/foo/build/staging_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/usr/lib/libiconv-full/lib -Wl,-rpath-link=/foo/build/staging_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/usr/lib/libiconv-full/lib -L/foo/build/staging_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/usr/lib/libintl-full/lib -Wl,-rpath-link=/foo/build/staging_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/usr/lib/libintl-full/lib -liconv" GOPATH="/foo/build/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/go-ethereum-1.9.22/.go_work/build" GOCACHE="/foo/build/tmp/go-build" GOMODCACHE="/foo/build/dl/go-mod-cache" GOENV=off /usr/bin/env bash ../../lang/golang//golang-build.sh build -v -trimpath -ldflags "all=-buildid '1613235521' -linkmode external -extldflags '-L/foo/build/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-8.4.0_musl_eabi/usr/lib -L/foo/build/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-8.4.0_musl_eabi/lib -fpic -specs=/foo/build/include/hardened-ld-pie.specs -Wl,-z,now -Wl,-z,relro -L/foo/build/staging_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/usr/lib/libiconv-full/lib -Wl,-rpath-link=/foo/build/staging_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/usr/lib/libiconv-full/lib -L/foo/build/staging_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/usr/lib/libintl-full/lib -Wl,-rpath-link=/foo/build/staging_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/usr/lib/libintl-full/lib -liconv'"   -buildmode pie -installsuffix "v7"    -ldflags "  -buildid '1613235521' -linkmode external -extldflags '-L/foo/build/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-8.4.0_musl_eabi/usr/lib -L/foo/build/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-8.4.0_musl_eabi/lib -fpic -specs=/foo/build/include/hardened-ld-pie.specs -Wl,-z,now -Wl,-z,relro -L/foo/build/staging_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/usr/lib/libiconv-full/lib -Wl,-rpath-link=/foo/build/staging_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/usr/lib/libiconv-full/lib -L/foo/build/staging_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/usr/lib/libintl-full/lib -Wl,-rpath-link=/foo/build/staging_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/usr/lib/libintl-full/lib -liconv'"  
Finding targets

Building targets
runtime/cgo
# runtime/cgo
/foo/build/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-8.4.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/8.4.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: cannot find -liconv
collect2: error: ld returned 1 exit status
```

ecsv, 2021-02-14:
what is geth an why I am the maintainer of it?

BKPepe, 2021-02-14:
@ecs oh, sorry for mentioning you! While looking into the history of changes done in [geth Makefile](https://github.com/openwrt/packages/commits/master/net/geth/Makefile) I found two people you and neheb

closed 2021-02-14 by neheb