openwrt/packages #14734: openconnect: undefined reference to libiconv

BKPepe, 2021-02-14:
Maintainer: @nmav
Compile tested: Turris Omnia, mvebu (cortex-a9), OpenWrt master
Enabled hardening, NLS and other stuff.

Description:

```
SHELL= flock /foo/build/tmp/.openconnect-8.10.tar.gz.flock -c '  	/foo/build/scripts/download.pl "/foo/build/dl" "openconnect-8.10.tar.gz" "30e64c6eca4be47bbf1d61f53dc003c6621213738d4ea7a35e5cf1ac2de9bab1" "" "ftp://ftp.infradead.org/pub/openconnect/"    '
rm -f /foo/build/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/openconnect-8.10/.built
touch /foo/build/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/openconnect-8.10/.built_check
CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -fmacro-prefix-map=/foo/build/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/openconnect-8.10=openconnect-8.10 -Wformat -Werror=format-security -fpic -fstack-protector-strong -D_FORTIFY_SOURCE=2 -Wl,-z,now -Wl,-z,relro  -I/foo/build/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-8.4.0_musl_eabi/usr/include -I/foo/build/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-8.4.0_musl_eabi/include/fortify -I/foo/build/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-8.4.0_musl_eabi/include " CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -fmacro-prefix-map=/foo/build/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/openconnect-8.10=openconnect-8.10 -Wformat -Werror=format-security -fpic -fstack-protector-strong -D_FORTIFY_SOURCE=2 -Wl,-z,now -Wl,-z,relro  -I/foo/build/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-8.4.0_musl_eabi/usr/include -I/foo/build/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-8.4.0_musl_eabi/include/fortify -I/foo/build/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-8.4.0_musl_eabi/include " LDFLAGS="-L/foo/build/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-8.4.0_musl_eabi/usr/lib -L/foo/build/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-8.4.0_musl_eabi/lib -fpic -specs=/foo/build/include/hardened-ld-pie.specs -znow -zrelro " make -j1 -C /foo/build/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/openconnect-8.10/. AR="arm-openwrt-linux-muslgnueabi-gcc-ar" AS="ccache_cc -c -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -fmacro-prefix-map=/foo/build/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/openconnect-8.10=openconnect-8.10 -Wformat -Werror=format-security -fpic -fstack-protector-strong -D_FORTIFY_SOURCE=2 -Wl,-z,now -Wl,-z,relro" LD=arm-openwrt-linux-muslgnueabi-ld NM="arm-openwrt-linux-muslgnueabi-gcc-nm" CC="ccache_cc" GCC="ccache_cc" CXX="ccache_cxx" RANLIB="arm-openwrt-linux-muslgnueabi-gcc-ranlib" STRIP=arm-openwrt-linux-muslgnueabi-strip OBJCOPY=arm-openwrt-linux-muslgnueabi-objcopy OBJDUMP=arm-openwrt-linux-muslgnueabi-objdump SIZE=arm-openwrt-linux-muslgnueabi-size CROSS="arm-openwrt-linux-muslgnueabi-" ARCH="arm" ;
make[3]: Entering directory '/foo/build/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/openconnect-8.10'
make  all-recursive
make[4]: Entering directory '/foo/build/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/openconnect-8.10'
Making all in tests
make[5]: Entering directory '/foo/build/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/openconnect-8.10/tests'
/bin/bash ../libtool  --tag=CC   --mode=link ccache_cc  -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -fmacro-prefix-map=/foo/build/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/openconnect-8.10=openconnect-8.10 -Wformat -Werror=format-security -fpic -fstack-protector-strong -D_FORTIFY_SOURCE=2 -Wl,-z,now -Wl,-z,relro   -L/foo/build/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-8.4.0_musl_eabi/usr/lib -L/foo/build/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-8.4.0_musl_eabi/lib -fpic -specs=/foo/build/include/hardened-ld-pie.specs -znow -zrelro  -o serverhash serverhash.o ../libopenconnect.la -L/foo/build/staging_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/usr/lib -lgnutls   
libtool: link: ccache_cc -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -fmacro-prefix-map=/foo/build/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/openconnect-8.10=openconnect-8.10 -Wformat -Werror=format-security -fpic -fstack-protector-strong -D_FORTIFY_SOURCE=2 -Wl,-z -Wl,now -Wl,-z -Wl,relro -fpic -specs=/foo/build/include/hardened-ld-pie.specs -znow -zrelro -o serverhash serverhash.o  -L/foo/build/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-8.4.0_musl_eabi/usr/lib -L/foo/build/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-8.4.0_musl_eabi/lib ../.libs/libopenconnect.a -L/foo/build/staging_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/usr/lib -lxml2 -lz -lgnutls
/foo/build/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-8.4.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/8.4.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: warning: libiconv.so.2, needed by /foo/build/staging_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/usr/lib/libxml2.so, not found (try using -rpath or -rpath-link)
/foo/build/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-8.4.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/8.4.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: /foo/build/staging_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/usr/lib/libxml2.so: undefined reference to `libiconv'
/foo/build/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-8.4.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/8.4.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: /foo/build/staging_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/usr/lib/libxml2.so: undefined reference to `libiconv_close'
/foo/build/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-8.4.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/8.4.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: /foo/build/staging_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/usr/lib/libxml2.so: undefined reference to `libiconv_open'
collect2: error: ld returned 1 exit status
make[5]: *** [Makefile:768: serverhash] Error 1
make[5]: Leaving directory '/foo/build/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/openconnect-8.10/tests'
make[4]: *** [Makefile:1489: all-recursive] Error 1
make[4]: Leaving directory '/foo/build/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/openconnect-8.10'
make[3]: *** [Makefile:753: all] Error 2
make[3]: Leaving directory '/foo/build/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/openconnect-8.10'
make[2]: *** [Makefile:89: /foo/build/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/openconnect-8.10/.built] Error 2
make[2]: Leaving directory '/foo/build/feeds/packages/net/openconnect'
time: package/feeds/packages/openconnect/compile#0.40#0.10#0.47
    ERROR: package/feeds/packages/openconnect failed to build.
make[1]: *** [package/Makefile:114: package/feeds/packages/openconnect/compile] Error 1
```

nmav, 2021-02-14:
This looks like an issue due to libxml2:
```
libtool: link: ccache_cc -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -fmacro-prefix-map=/foo/build/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/openconnect-8.10=openconnect-8.10 -Wformat -Werror=format-security -fpic -fstack-protector-strong -D_FORTIFY_SOURCE=2 -Wl,-z -Wl,now -Wl,-z -Wl,relro -fpic -specs=/foo/build/include/hardened-ld-pie.specs -znow -zrelro -o serverhash serverhash.o  -L/foo/build/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-8.4.0_musl_eabi/usr/lib -L/foo/build/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-8.4.0_musl_eabi/lib ../.libs/libopenconnect.a -L/foo/build/staging_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/usr/lib -lxml2 -lz -lgnutls
/foo/build/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-8.4.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/8.4.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: warning: libiconv.so.2, needed by /foo/build/staging_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/usr/lib/libxml2.so, not found (try using -rpath or -rpath-link)
/foo/build/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-8.4.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/8.4.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: /foo/build/staging_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/usr/lib/libxml2.so: undefined reference to `libiconv'
```

BKPepe, 2021-02-14:
That's weird as I compiled libxml2 and even I installed on my host system libxml2 adnd libxml2-dev and still getting this error. Maybe related to https://github.com/openwrt/packages/commit/d18692c92638cdc1e643554699cb21e893abf2e9?

BKPepe, 2021-02-14:
Reverted that commit and I can compile `openconnect` successfully.

nmav, 2021-02-14:
@mhei ? Does libxml2 miss a direct link to iconv, or is that something that @BKPepe should have done in his setup?

BKPepe, 2021-02-14:
Btw `openconnect` is not only the package which suffers from the issue, which I described. Also, the same issue I experience with packages  `lttng-tools`, `augeas`, `hs20`, `stoken`.

commit d6510824 (2021-02-20): Merge pull request #14741 from mhei/openconnect-iconv-intl

openconnect: fix dependencies to iconv/intl (fixes #14734)
commit f2877e81 (2021-02-20): openconnect: fix dependencies to iconv/intl (fixes #14734)

This fixes the issue raised after d18692c (libxml2: allow building with iconv support).
closed 2021-02-20 by mhei