openwrt/packages #14735: isc-dhcp-server-ipv4: disappear from package feeds and menuconfig

carlo2007, 2021-02-14:
Maintainer: @pprindeville
Environment: mipsel_24kc, MIPS 1004Kc V2.15, OpenWrt (all versions)

Description:

From Openwrt System Build is no more possible to compile the package: it has disappeared from feeds and as a result from menuconfig.

pprindeville, 2021-02-15:
This seems to be a broader problem resulting from changes either to `opkg` or the broader build machinery itself...

plntyk, 2021-03-05:
the compile / install error "incompatible with the architectures" should be fixed

afaik with DEPENDS+=+bind-server +bind-client the package will show up and bind-server will be selected/installed too

carlo2007, 2021-03-06:
My envroinment is an embedded equipment with limited resources. Actually I don't have a "local bind server" and I don't want to install a DNS server. Introducing the dependency on the bind-server package means for my envroinment that I will no longer be able to use isc-dhcp-server as DHCP only and will probably have to change software or keep an old version of it without upgrading.

pprindeville, 2021-03-19:
> Collected errors:
> 
> pkg_hash_fetch_best_installation_candidate: Packages for isc-dhcp-server-ipv4 found, but incompatible with the architectures configured
> opkg_install_cmd: Cannot install package isc-dhcp-server-ipv4.

This was actually the result of a regression @nbd168 introduced via commit c92165038217e49075098479704da58a2a3a89bd.

pprindeville, 2021-03-19:
> My envroinment is an embedded equipment with limited resources. Actually I don't have a "local bind server" and I don't want to install a DNS server. Introducing the dependency on the bind-server [...]

Point taken.  Fixed with PR #15161.

closed 2025-11-14 by GeorgeSapkin