openwrt/packages #14738: klish: can not find libxml2 library

BKPepe, 2021-02-14:
Maintainer: @t-umeno 
Environment: Compile tested: Turris Omnia, mvebu (cortex-a9), OpenWrt master and enabled hardening, NLS

Description:

```
f06bb485e15b2ff27b856e46cd76fee1eac7e0f62a8d8ac0ad413694614" "" "http://libcode.org/attachments/download/77/"    '
+ curl -f --connect-timeout 20 --retry 5 --location --insecure http://libcode.org/attachments/download/77/klish-2.2.0.tar.xz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  368k    0  368k    0     0   487k      0 --:--:-- --:--:-- --:--:--  487k
touch /foo/build/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/klish-2.2.0/.prepared_22343697721ba309e67251855531327d_6664517399ebbbc92a37c5bb081b5c53_check
. /foo/build/include/shell.sh; xzcat /foo/build/dl/klish-2.2.0.tar.xz | tar -C /foo/build/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/klish-2.2.0/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/. /foo/build/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/klish-2.2.0

Applying ./patches/010-shell_execute_fix.patch using plaintext: 
patching file clish/shell/shell_execute.c
touch /foo/build/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/klish-2.2.0/.prepared_22343697721ba309e67251855531327d_6664517399ebbbc92a37c5bb081b5c53
rm -f /foo/build/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/klish-2.2.0/.configured_*
rm -f /foo/build/staging_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/stamp/.klish_installed
(cd /foo/build/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/klish-2.2.0/./; if [ -x ./configure ]; then find /foo/build/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/klish-2.2.0/ -name config.guess | xargs -r chmod u+w; find /foo/build/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/klish-2.2.0/ -name config.guess | xargs -r -n1 cp --remove-destination /foo/build/scripts/config.guess; find /foo/build/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/klish-2.2.0/ -name config.sub | xargs -r chmod u+w; find /foo/build/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/klish-2.2.0/ -name config.sub | xargs -r -n1 cp --remove-destination /foo/build/scripts/config.sub; AR="arm-openwrt-linux-muslgnueabi-gcc-ar" AS="ccache_cc -c -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -fmacro-prefix-map=/foo/build/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/klish-2.2.0=klish-2.2.0 -Wformat -Werror=format-security -fpic -fstack-protector-strong -D_FORTIFY_SOURCE=2 -Wl,-z,now -Wl,-z,relro -D_GNU_SOURCE" LD=arm-openwrt-linux-muslgnueabi-ld NM="arm-openwrt-linux-muslgnueabi-gcc-nm" CC="ccache_cc" GCC="ccache_cc" CXX="ccache_cxx" RANLIB="arm-openwrt-linux-muslgnueabi-gcc-ranlib" STRIP=arm-openwrt-linux-muslgnueabi-strip OBJCOPY=arm-openwrt-linux-muslgnueabi-objcopy OBJDUMP=arm-openwrt-linux-muslgnueabi-objdump SIZE=arm-openwrt-linux-muslgnueabi-size CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -fmacro-prefix-map=/foo/build/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/klish-2.2.0=klish-2.2.0 -Wformat -Werror=format-security -fpic -fstack-protector-strong -D_FORTIFY_SOURCE=2 -Wl,-z,now -Wl,-z,relro -D_GNU_SOURCE " CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -fmacro-prefix-map=/foo/build/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/klish-2.2.0=klish-2.2.0 -Wformat -Werror=format-security -fpic -fstack-protector-strong -D_FORTIFY_SOURCE=2 -Wl,-z,now -Wl,-z,relro -D_GNU_SOURCE " CPPFLAGS="-I/foo/build/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-8.4.0_musl_eabi/usr/include -I/foo/build/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-8.4.0_musl_eabi/include/fortify -I/foo/build/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-8.4.0_musl_eabi/include " LDFLAGS="-L/foo/build/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-8.4.0_musl_eabi/usr/lib -L/foo/build/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-8.4.0_musl_eabi/lib -fpic -specs=/foo/build/include/hardened-ld-pie.specs -znow -zrelro "   ./configure --target=arm-openwrt-linux --host=arm-openwrt-linux --build=x86_64-pc-linux-gnu --program-prefix="" --program-suffix="" --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --libexecdir=/usr/lib --sysconfdir=/etc --datadir=/usr/share --localstatedir=/var --mandir=/usr/man --infodir=/usr/info   --with-libxml2 ; fi; )
configure: loading site script /foo/build/include/site/arm
checking for arm-openwrt-linux-gcc... ccache_cc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... yes
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether ccache_cc accepts -g... yes
checking for ccache_cc option to accept ISO C89... none needed
checking whether ccache_cc understands -c and -o together... yes
checking how to run the C preprocessor... ccache_cc -E
checking for grep that handles long lines and -e... /foo/build/staging_dir/host/bin/grep
checking for egrep... /foo/build/staging_dir/host/bin/grep -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking minix/config.h usability... no
checking minix/config.h presence... no
checking for minix/config.h... no
checking whether it is safe to define __EXTENSIONS__... yes
checking for arm-openwrt-linux-gcc... (cached) ccache_cc
checking whether we are using the GNU C compiler... (cached) yes
checking whether ccache_cc accepts -g... (cached) yes
checking for ccache_cc option to accept ISO C89... (cached) none needed
checking whether ccache_cc understands -c and -o together... (cached) yes
checking build system type... x86_64-pc-linux-gnu
checking host system type... arm-openwrt-linux-gnu
checking how to print strings... printf
checking for a sed that does not truncate output... /foo/build/staging_dir/host/bin/sed
checking for fgrep... /foo/build/staging_dir/host/bin/grep -F
checking for ld used by ccache_cc... arm-openwrt-linux-muslgnueabi-ld
checking if the linker (arm-openwrt-linux-muslgnueabi-ld) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... arm-openwrt-linux-muslgnueabi-gcc-nm
checking the name lister (arm-openwrt-linux-muslgnueabi-gcc-nm) interface... BSD nm
checking whether ln -s works... yes
checking the maximum length of command line arguments... 1572864
checking how to convert x86_64-pc-linux-gnu file names to arm-openwrt-linux-gnu format... func_convert_file_noop
checking how to convert x86_64-pc-linux-gnu file names to toolchain format... func_convert_file_noop
checking for arm-openwrt-linux-muslgnueabi-ld option to reload object files... -r
checking for arm-openwrt-linux-objdump... arm-openwrt-linux-muslgnueabi-objdump
checking how to recognize dependent libraries... pass_all
[... 29 lines cut ...]
checking whether to build shared libraries... yes
checking whether to build static libraries... yes
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for arm-openwrt-linux-strip... (cached) arm-openwrt-linux-muslgnueabi-strip
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking whether make sets $(MAKE)... yes
checking for style of include used by make... GNU
checking whether make supports nested variables... yes
checking dependency style of ccache_cc... gcc3
checking size of int... (cached) 4
checking size of long... (cached) 4
checking size of size_t... (cached) 4
checking if LD -Wl,--version-script works... yes
checking for libexpat support... no
checking for libroxml support... no
checking for xmlNewDoc in -lxml2... no
configure: error: cannot find libxml2 library
make[2]: *** [Makefile:73: /foo/build/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/klish-2.2.0/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 1
make[2]: Leaving directory '/foo/build/feeds/packages/utils/klish'
time: package/feeds/packages/klish/compile#1.25#0.46#2.38
    ERROR: package/feeds/packages/klish failed to build.
make[1]: *** [package/Makefile:114: package/feeds/packages/klish/compile] Error 1

```

neheb, 2021-02-15:
Missing include $(INCLUDE_DIR)/nls.mk

BKPepe, 2021-02-15:
@neheb yes, I can confirm that I was able to compile it with your suggested approach. Shall I create a PR or will you?  It appears to be related to include https://github.com/openwrt/packages/commit/d18692c92638cdc1e643554699cb21e893abf2e9, because if I revert it, I don't need to add it there.

neheb, 2021-02-15:
Go ahead.

Yeah that commit causes mass breakage.

BKPepe, 2021-02-15:
I noticed just a few packages (~5 packages), which are broken, not what I would say a mass breakage. But it seems that buildbots are fine with that and they don't report any failure which I see these days. :(

neheb, 2021-02-15:
Are you building with BUILD_NLS?

BKPepe, 2021-02-15:
Yes as mentioned in OP, which seems to be an issue in my case.

BKPepe, 2021-02-15:
Fixed by https://github.com/openwrt/packages/pull/14756

Closing.

closed 2021-02-15 by BKPepe