openwrt/packages #14754: github-workflow: add clang for xdp-support?

PolynomialDivision, 2021-02-15:
As you maybe already noticed, some people are working on XDP Support in OpenWrt.

As loader we currently have:
- iproute2
- [ubpf](https://github.com/blogic/ubpf)
- [WIP xdp-tools](https://github.com/openwrt/packages/pull/14739)

Here are some hacky OpenWrt examples from me:
- [XDP-IP-Stats](https://github.com/PolynomialDivision/xdp-ipstats/)
- [Remove Routing Header Extension](https://github.com/PolynomialDivision/xdp-srv6-remover/)

Soon, there will be some OpenWrt tutorial. Further, I highly recommend [XDP-Tutorial](https://github.com/xdp-project/xdp-tutorial).
As u can see in the [WIP xdp-tools](https://github.com/openwrt/packages/pull/14739), everything just fails, since I use clang to cross-compile.
```
*** ERROR: Cannot find tool clang
```
Can we add clang to the workflow to have support for packages that want to make use of xdp?

PolynomialDivision, 2021-02-19:
https://github.com/openwrt/docker/pull/77

neheb, 2021-04-17:
OTOH the OpenWrt buildbot do not have clang AFAIK.

PolynomialDivision, 2021-04-17:
> OTOH the OpenWrt buildbot do not have clang AFAIK.

I wanted to add it, but then suddely I had no free time to work on it anymore. :/
But I will continue! :)