openwrt/packages #14755: gcc: missing link from gcc to cc

flyn-org, 2021-02-15:
Maintainer: @noblepepper 
Environment: x86_64 master

Description:
The gcc package does not create a symbolic link from /usr/bin/gcc to /usr/bin/cc. This breaks some Makefiles that use $(CC), as GNU make sets CC to cc by default. I think POSIX required the presence of cc as some point.

pprindeville, 2021-03-04:
Why not just pass `CC=gcc make ...` into those packages?

plntyk, 2021-03-07:
for reference: 

[CC in Single Unix v2](https://pubs.opengroup.org/onlinepubs/7908799/xcu/cc.html) - marked Legacy
[C99 in Single Unix v4](http://pubs.opengroup.org/onlinepubs/9699919799/utilities/c99.html) - no CC

via [stackexchange](https://unix.stackexchange.com/questions/378890/does-posix-mention-cc-or-only-c99)

with CC: Debian, Arch; FreeBSD
no CC: Fedora

personally I dont use gcc on OpenWrt anyway - feel free to reject

flyn-org, 2021-03-07:
@pprindeville, `CC=gcc make` would work, but requires an explicit action. I use OpenWrt to host a build service of sorts, and my use case benefits from the cc symbolic link. Symbolic links are cheap, so I thought this would be an acceptable change. I agree with @plntyk that it is unclear if cc is actually required by POSIX or The Single Unix Specification (does the legacy designation mean that C < C99 is legacy or the requirement for cc is legacy?). Moreover, I am not sure OpenWrt presumes to implement either (e.g., does busybox drop some requirements?). All of that said, the make we ship sets CC to cc, so I think our gcc package should provide the same.

PR #15008 (2021-03-12): gcc: add cc symlink
fixes #14755

Maintainer: @noblepepper
Compile tested: x86-64 VM
Run tested: x86-64 VM

fixes #14755
commit 55cb4d1a (2021-03-12): gcc: add cc symlink

fixes #14755
commit bd1fcf1d (2022-08-08): gcc: add cc symlink

fixes #14755

(cherry picked from commit 55cb4d1aee0055e26f19b968beaed83a2668ee51)
closed 2021-03-12 by neheb