openwrt/packages #14771: [19.07]has a dependency on 'kmod-crypto-arc4', which does not exist

openwrt-diy, 2021-02-17:
Maintainer: @neheb
Environment: (put here arch, model, OpenWrt version)
OpenWrt-19.07.7

Description:
./scripts/feeds install -a A prompt error occurred during the update installation process
When make V=99 starts to compile, an error appears

WARNING: Makefile'package/feeds/packages/ksmbd/Makefile' has a dependency on'kmod-crypto-arc4', which does not exist

neheb, 2021-02-17:
Under openwrt 19.07 , target/linux/generic/config-4.14:CONFIG_CRYPTO_ARC4=y meaning that this dependency can be dropped.

The module also lists this as a softdep.

```
MODULE_SOFTDEP("pre: arc4");
```

I don't remember what ARC4 is used for. ping @Andy2244

Andy2244, 2021-02-17:
> I don't remember what ARC4 is used for. ping @Andy2244

Don't look at me i simply tried to follow [upstream](https://github.com/cifsd-team/cifsd/blob/master/Kconfig).
I think its for some older protocol negotiation, so i guess this might not be used in reality and just follows some older standard?

Quick search came up with this for samba-4.4 patch-notes, so seems its used somewhere:
```
Crypto.Cipher.ARC4 is not available on some platforms, fallback to M2Crypto.RC4.RC4 then.
```

BKPepe, 2021-02-17:
Guys, let's solve this, right? Dropping kmod-crypto-arc4 seems enough for me. Right now, I can not install `kmod-fs-ksmbd` due to this error:
```
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-fs-ksmbd:
 * 	kmod-crypto-arc4
 * opkg_install_cmd: Cannot install package kmod-fs-ksmbd.
```

closed 2021-02-17 by BKPepe

NoUserNameForYou, 2021-05-16:
Why is this issue closed? smb isn't installable without it.

Andy2244, 2021-05-16:
> 
> 
> Why is this issue closed? smb isn't installable without it.

Because the issue was solved via #14778 starting with 19.07.7, so you have to upgrade if you run a older version.

NoUserNameForYou, 2021-05-16:
Hmm, I'm already using that?

"Hostname	OpenWrt
Model	TP-Link Archer C7 v1
Architecture	Qualcomm Atheros QCA9558 ver 1 rev 0
Firmware Version	OpenWrt 19.07.7 r11306-c4a6851c72 / LuCI openwrt-19.07 branch git-21.132.36199-d0cf6e4"

I'll try upgrading now.

sirinsidiator, 2021-05-17:
Same problem here.
Model TP-Link TL-WDR3600 v1
Architecture Atheros AR9344 rev 2
Firmware Version OpenWrt 19.07.7 r11306-c4a6851c72 / LuCI openwrt-19.07 branch git-21.101.59933-c56d9f2
Kernel Version 4.14.221

[image]

jeromeof, 2021-05-17:
Same problem for me on wrt32x

BKPepe, 2021-05-17:
Guys, this is fixed, as you can see from the pull request. Also, it is mentioned on the OpenWrt's 19.07.7 as a [regression](https://openwrt.org/releases/19.07/notes-19.07.7#regressions) with the know-how what you should do to install it. This is basically fixed, and nothing we can do now about it here. Because the dependency is not present anymore in OpenWrt 19.07 branch. Take a look here: https://github.com/openwrt/packages/blob/2543c33b0b05db79954124b469dedc8717d7c584/kernel/ksmbd/Makefile

1. Follow the instructions on the website.
2. If it does not work, you can compile the package yourself and install it.
Yes, that's for experienced users. You can install the latest development version of OpenWrt 19.07 branch to have a recent version https://github.com/openwrt/openwrt/commit/c4a6851c7276084137d91a91ab598e5109452961 as recent versions includes many security fixes related to OpenVPN, wolfSSL, mbedTLS, dropbear and so on.

Also, OpenWrt 19.07.8 will follow soon due to FragAttacks security vulnerabilities found in wireless.

As said, nothing more to do here from our side.