openwrt/packages #14780: adblock: [feature request] add oisd.nl basic list to Blocklist Sources

digital-mystik, 2021-02-17:
Maintainer: @dibdot et al

Environment: all

****

Description:

Including the oisd.nl basic list in the default sources would be beneficial for all users not using beefier x86 builds, since the full list is far too large for the majority of devices; the basic list is a good resource and would be a nice addition.

Thank you for your time.

dibdot, 2021-02-18:
yep, currently adblock 4.1 is in preparation which includes a much more flexible list handling (e.g. you can select categories in archives like shallalist/utcapitole or list variants from energized or stevenblack) ... the basic list from oisd is also included.

digital-mystik, 2021-02-18:
Sounds great, much appreciated!

closed 2021-02-18 by digital-mystik

dibdot, 2021-02-27:
done with https://github.com/openwrt/packages/pull/14913

ansonl, 2021-04-25:
When I copy the current `adblock.sources` in the repo and zip it to `adblock.sources.gz` and select `oisd_basic `as the source, the adblock log says:

> Sun Apr 25 14:20:12 2021 user.info adblock-4.0.7[20591]: download of 'oisd_basic' failed, url: https://dbl.oisd.nl/basic, rule: /^([[:alnum:]_-]{1,63}\.)+[[:alpha:]]+([[:space:]]|$)/{print tolower($1)}, categories: -, rc: 8, log: Downloading 'https://dbl.oisd.nl/basic' Connecting to 172.67.198.135:443 HTTP error 503

It worked before but I'm not sure why it is failing now. My current adblock version is 4.0.7-8

dibdot, 2021-04-25:
A temporary server issue, i.e. https://developer.mozilla.org/en-US/docs/Web/HTTP/Status/503. Just wait ... currently at works for me (from Germany).

ansonl, 2021-04-25:
Ah, I see the issue now, it loads fine from my computer, but the domain has cloudflare protection and the Cloudflare page is saying "Please turn JavasScript on and reload the page." when the router tries to fetch/curl it.

ansonl, 2021-04-26:
@dibdot I contacted the oisd maintainer and he changed the Cloudflare settings. However the current adblock.sources list oisd_basic url of `https://dbl.oisd.nl/basic` returns an 301 redirect and should be changed to reflect the new url at [`https://dbl.oisd.nl/basic/`](https://dbl.oisd.nl/basic/) so that adblock can parse it.