openwrt/packages #14799: open-vm-tools: change openwrt network config on vmware clone

feckert, 2021-02-19:
Maintainer: @kuoruan @srchack 
Environment: x86_64, Vmware, openwrt-master
Vmware-Environment:
Name:         VMware vCenter Server
Vendor:       VMware, Inc.
Version:      6.7.0
Build:        17137327
OS type:      linux-x64
API type:     VirtualCenter
API version:  6.7.3
Product ID:   vpx

Description:

I'm currently working with [terraform](https://www.terraform.io/) and the [vsphere provider](https://registry.terraform.io/providers/hashicorp/vsphere/latest/docs/resources/virtual_machine#network-interface-options) and I want to push the created VMDK from OpenWrt into a vcenter and create a template. This works so far :tada: 

But now I have a problem that I can not change the IP address on cloning this template.
I think it's somehow the open-vm-tools, because I think they don't support OpenWrt.
Can you help me maybe there further.
I have also created a Debian VMs and there the IP are adjusted.
So this must be somehow related to OpenWrt and the open-vm-tools.
Do you have any hint on what this could be.
And how it's all connected.
Google was not exactly a help.

plntyk, 2021-03-10:
from what I understand :

supposed to happen: every clone has a different IP address

what happens: every clone has the same IP address

likely culprit:
there is a firstboot step / generating config files that creates the config in firstboot - so every config is generated the same in your case

- change the scripts source code ( package/base-files/files/bin/config_generate )
- use or manipulate board.json (havent done / used that way myself yet - check git commits /sources for supposed usage)
- provide FILES=  - can be done Imagebuilder for example

feckert, 2021-03-16:
@plntyk thanks for the feedback.

>     * change the scripts source code ( package/base-files/files/bin/config_generate )
>     * use or manipulate board.json (havent done / used that way myself yet - check git commits /sources for supposed usage)
>     * provide FILES=  - can be done Imagebuilder for example

Your suggestion would not solve my problems. 

Unfortunately I haven't found out yet how vmware does it to change the IP when cloning a template. It would help to understand it. Then you could also apply this to OpenWrt. But at the moment I don't know where to start. Is the problem with vmware. Since it doesn't know the distribution. Or do I have to change something in OpenWrt. Maybe the data is already provided via the open-vm-tools and I just have to set them and reboot.

I am grateful for any advice.

kuoruan, 2021-03-16:
https://github.com/openwrt/packages/blob/23e0bc9f9ca9d1ce86b997483f16383e8e00017b/utils/open-vm-tools/Makefile#L73

I'm not sure if this cause by the build option `--without-dnet`. You can just have a try.

brada4, 2021-09-29:
It will never work, the script that vsphere injects checks /etc/*release against list of expensive linux, Ubuntu - yes, Debian - no.

feckert, 2021-09-29:
@brada4 can you please go into more detail?
Is there any documentation on this where I can look it up?

brada4, 2021-09-29:
https://partnerweb.vmware.com/programs/guestOS/guest-os-customization-matrix.pdf

brada4, 2021-09-29:
When you try to apply host customisations it says debian is woring OS. Same goes with Ubuntu 20 + ifupdown. It is really fragile facility