openwrt/packages #14812: Ralink kmod-rt2x00-* very low speed WiFi

EagletUSSR, 2021-02-20:
Hello all!
I have **very low WiFi speed** on MT7620A with use **kmod-rt2x00-*** packages for WiFi adapter. My router is Youku-L1C, i use firmware for ZBT-826-32M device. I use OpenWRT version 19.07.x. WiFi speed not increase above 15-20 Mbit/sec in 802.11g/n mode. I found additional information about this issue in: https://bugs.openwrt.org/index.php?do=details&task_id=929 This issue not fixed with 25.07.2017 year.
Please fix this issue for full speed WiFi by use kmod-rt2x00-* packages!

Best regards!

EagletUSSR, 2021-02-26:
Current Speed: download is 20-24 Mbit/sec, upload is 50-60 Mbit/sec. Is very strange results! In driver kmod-rt2x00-* problem with Tx rate?

feckert, 2021-03-01:
This is not related to the package feed.
We can't do anything there.
if then this must be fixed in openwrt Base.
And there you should open an issue on  https://bugs.openwrt.org.
I think this is a driver problem with the kernel.

closed 2021-03-01 by feckert