openwrt/packages #14825: TOR missing required folder

erdoukki, 2021-02-21:
missing /etc/tor/data crash tor service : TOR do not start because of a missing required folder

## FIXME
mkdir /etc/tor/data/
chown tor:tor /etc/tor/data/
service tor restart
–> OK

https://bugs.openwrt.org/index.php?do=details&task_id=3636

ja-pa, 2021-02-23:
Hi,
I think that this error is specific to your torrc configuration. There is no ``/etc/tor/data`` directory in the default torrc config https://github.com/torproject/tor/blob/master/src/config/torrc.sample.in

erdoukki, 2021-03-02:
no, there is no reference of /etc/tordata in my /etc/tor/torrc config file.
content of /etc/tor/data : 
`/etc/tor/data:
drwx------    3 tor      tor         4.0K Mar  2 11:50 .
drwxr-xr-x    4 root     root        4.0K Feb 21 18:47 ..
-rw-------    1 tor      tor        20.0K Feb 24 00:13 cached-certs
-rw-------    1 tor      tor         2.3M Mar  2 11:50 cached-microdesc-consensus
-rw-------    1 tor      tor         5.9M Feb 21 10:53 cached-microdescs
-rw-------    1 tor      tor         2.5M Mar  1 19:21 cached-microdescs.new
drwx------    2 tor      tor         4.0K Feb 21 10:52 keys
-rw-------    1 tor      tor            0 Feb 21 19:13 lock
-rw-r-----    1 tor      tor       435.6K Mar  2 07:13 notices.log
-rw-------    1 tor      tor        10.8K Mar  2 09:03 state
`