openwrt/packages #14830: hplip: lack of plugin which handles scanning for my HP Multi Function Printer

szero, 2021-02-21:
Maintainer: @luizluca 
Environment: MediaTek MT7621 ver:1 eco:3 , Xiaomi Mi Router 3 Pro, OpenWrt 19.07.7 r11306-c4a6851c72
hplip-common version - 3.19.6-2
hplip-sane version - 3.19.6-2

Description:

Hello, so I own an HP Laserjet M1132 MFP and it works flawlessly while printing, but scanning cannot be setup, my troubleshooting below:

Scanner is detected by `scanimage -L ` but the scanning itself fails:
[image]

`logread` output:
[image]

Ok so some sort of HP-MARVELL-SCAN is initialized but fails because file `/var/lib/hp/hplip.state` doesn't exist.
I've looked how this file looks like on my Debian system and created this file on the router with contents of:
```
[plugin]
installed = 1
eula = 1
version = 3.19.6
```
`scanimage` fails with the same error as before so we go to `logread` again:
[image]

Ok so shared library thats responsible for handling my printer model isn't compiled in this package, I'm not sure if its part of the proprietary HP plugin and thats why it's not compiled in by default. I'm willing to work on this but I need some basic insight whether it would be creating custom package build or just crosscompiling missing parts of the package that are proprietary and cannot be inserted into official package.

szero, 2021-02-21:
Exact same issue mentioned on the openwrt forums: https://forum.openwrt.org/t/sane-issue-on-hp-laserjet-m1132/57698

luizluca, 2021-02-22:
I'll take a look

luizluca, 2021-02-22:
It looks like bb_marvel (and others) are from https://developers.hp.com/sites/default/files/hplip-3.21.2-plugin.run.
They are proprietary drivers and distributed as a binary-only library/module. I guess the hp-plugin script downloads this file on-demand.

$ ./hplip-3.21.2-plugin.run --list | grep bb_mar
-rwxrwxr-x hptest/hptest   34347 2021-02-14 20:59 ./bb_marvell-x86_32.so
-rwxrwxr-x hptest/hptest   13984 2021-02-14 20:59 ./bb_marvell-arm32.so
-rwxrwxr-x hptest/hptest   20952 2021-02-14 20:59 ./bb_marvell-arm64.so
-rwxrwxr-x hptest/hptest   37957 2021-02-14 20:59 ./bb_marvell-x86_64.so

There is no MIPS version. So, there is no hope for Xiaomi Mi Router 3 Pro (and many others including my router). For OpenWrt targets that match those archs, you probably could use these files as they are statically linked. However, I do not have a scanner for those drivers. The one I have works with OSS drivers. If those binary-only files do work with a compatible OpenWrt target, I could work together with a scanner-owner tester to implement a "kind-of' hp-plugin script that would download those files.

I'm closing this issue as I only HP can solve it.

closed 2021-02-22 by luizluca

szero, 2021-02-22:
Oh crap, I guess asking on HP forums is futile because as I glimpsed through OpenWRT buildsystem, they would have to crosscompile their stuff against OpenWRT kernel, and even in the end I expect to get corporate answer of "if you need this functionality pls buy one of our products with print and scan servers built in". Will end up connecting this thing to NAS or something in the end. Thanks for figuring this out quickly.