openwrt/packages #14840: [19.07]https-dns-proxy and mwan3 have compatibility issues, and DNS forwarding cannot be resolved!

openwrt-diy, 2021-02-22:
@feckert @aaronjg @jamesmacwhite @brianjmurrell @stangri @aparcar

Environment: x86
openwrt version: 19.07.7
mwan3 version: 2.8.15-1
https-dns-proxy: 2021-01-17-1

Description:
When mwan3 or https-dns-proxy is used separately, the routing network is normal!

When using https-dns-proxy and mwan3 at the same time, when the router WAN and other interfaces are normally online, the LAN interface (including PPPoE dial-up terminal) can access the Internet normally!

When the WAN interface is disconnected, the other WANB, WANC, and WAND interfaces are all online normally! At this time, all users under all LAN interfaces (including PPPoE dial-up terminals) cannot access the Internet! The browser web page prompts that DNS resolution cannot be obtained

stangri, 2021-02-22:
Is https (port 443) traffic allowed under WANB/WANC/WAND?

If you want more qualified help (from others, this is the extent of what I can suggest), you need to post configs for both packages, as well as dhcp/network/firewall configs.

openwrt-diy, 2021-02-23:
**The following is a screenshot of the online network**

[image]
[image]
[image]
[image]

openwrt-diy, 2021-02-23:
The following is a screenshot of the network DNS forwarding failure after the WAN interface is offline.

[image]

openwrt-diy, 2021-02-23:
/$ uci show https-dns-proxy
https-dns-proxy.config=main
https-dns-proxy.config.update_dnsmasq_config='-'
https-dns-proxy.@https-dns-proxy[0]=https-dns-proxy
https-dns-proxy.@https-dns-proxy[0].bootstrap_dns='223.5.5.5,223.6.6.6,2400:3200::1,2400:3200:baba::1'
https-dns-proxy.@https-dns-proxy[0].resolver_url='https://dns.alidns.com/dns-query'
https-dns-proxy.@https-dns-proxy[0].listen_addr='0.0.0.0'
https-dns-proxy.@https-dns-proxy[0].listen_port='5053'
https-dns-proxy.@https-dns-proxy[1]=https-dns-proxy
https-dns-proxy.@https-dns-proxy[1].bootstrap_dns='119.29.29.29,119.28.28.28'
https-dns-proxy.@https-dns-proxy[1].resolver_url='https://doh.pub/dns-query'
https-dns-proxy.@https-dns-proxy[1].listen_addr='0.0.0.0'
https-dns-proxy.@https-dns-proxy[1].listen_port='5054'
https-dns-proxy.@https-dns-proxy[2]=https-dns-proxy
https-dns-proxy.@https-dns-proxy[2].bootstrap_dns='1.1.1.1,1.0.0.1,2606:4700:4700::1111,2606:4700:4700::1001'
https-dns-proxy.@https-dns-proxy[2].resolver_url='https://cloudflare-dns.com/dns-query'
https-dns-proxy.@https-dns-proxy[2].listen_addr='0.0.0.0'
https-dns-proxy.@https-dns-proxy[2].listen_port='5055'
https-dns-proxy.@https-dns-proxy[3]=https-dns-proxy
https-dns-proxy.@https-dns-proxy[3].bootstrap_dns='8.8.8.8,8.8.4.4'
https-dns-proxy.@https-dns-proxy[3].resolver_url='https://dns.google/dns-query'
https-dns-proxy.@https-dns-proxy[3].listen_addr='0.0.0.0'
https-dns-proxy.@https-dns-proxy[3].listen_port='5056'

/$ uci show firewall
firewall.@defaults[0]=defaults
firewall.@defaults[0].synflood_protect='1'
firewall.@defaults[0].input='ACCEPT'
firewall.@defaults[0].output='ACCEPT'
firewall.@defaults[0].forward='REJECT'
firewall.@zone[0]=zone
firewall.@zone[0].name='lan'
firewall.@zone[0].network='lan'
firewall.@zone[0].input='ACCEPT'
firewall.@zone[0].output='ACCEPT'
firewall.@zone[0].forward='ACCEPT'
firewall.@zone[0].device='ppp0' 'ppp1' 'ppp2' 'ppp3' 'ppp4' 'ppp5' 'ppp6' 'ppp7' 'ppp8' 'ppp9' 'ppp10' 'ppp11' 'ppp12' 'ppp13' 'ppp14' 'ppp15' 'ppp16' 'ppp17' 'ppp18' 'ppp19' 'ppp20' 'ppp21' 'ppp22' 'ppp23' 'ppp24' 'ppp25' 'ppp26' 'ppp27' 'ppp28' 'ppp29' 'ppp30' 'ppp31' 

'ppp32' 'ppp33' 'ppp34' 'ppp35' 'ppp36' 'ppp37' 'ppp38' 'ppp39' 'ppp40' 'ppp41' 'ppp42' 'ppp43' 'ppp44' 'ppp45' 'ppp46' 'ppp47' 'ppp48' 'ppp49' 'ppp50' 'ppp51' 'ppp52' 'ppp53' 'ppp54' 'ppp55' 'ppp56' 'ppp57' 'ppp58' 'ppp59' 'ppp60' 'ppp61' 'ppp62' 'ppp63' 'ppp64' 'ppp65' 

'ppp66' 'ppp67' 'ppp68' 'ppp69' 'ppp70' 'ppp71' 'ppp72' 'ppp73' 'ppp74' 'ppp75' 'ppp76' 'ppp77' 'ppp78' 'ppp79' 'ppp80' 'ppp81' 'ppp82' 'ppp83' 'ppp84' 'ppp85' 'ppp86' 'ppp87' 'ppp88' 'ppp89' 'ppp90' 'ppp91' 'ppp92' 'ppp93' 'ppp94' 'ppp95' 'ppp96' 'ppp97' 'ppp98' 'ppp99' 

'ppp100' 'ppp101' 'ppp102' 'ppp103' 'ppp104' 'ppp105' 'ppp106' 'ppp107' 'ppp108' 'ppp109' 'ppp110' 'ppp111' 'ppp112' 'ppp113' 'ppp114' 'ppp115' 'ppp116' 'ppp117' 'ppp118' 'ppp119' 'ppp120' 'ppp121' 'ppp122' 'ppp123' 'ppp124' 'ppp125' 'ppp126' 'ppp127' 'ppp128' 'ppp129' 

'ppp130' 'ppp131' 'ppp132' 'ppp133' 'ppp134' 'ppp135' 'ppp136' 'ppp137' 'ppp138' 'ppp139' 'ppp140' 'ppp141' 'ppp142' 'ppp143' 'ppp144' 'ppp145' 'ppp146' 'ppp147' 'ppp148' 'ppp149' 'ppp150' 'ppp151' 'ppp152' 'ppp153' 'ppp154' 'ppp155' 'ppp156' 'ppp157' 'ppp158' 'ppp159' 

'ppp160' 'ppp161' 'ppp162' 'ppp163' 'ppp164' 'ppp165' 'ppp166' 'ppp167' 'ppp168' 'ppp169' 'ppp170' 'ppp171' 'ppp172' 'ppp173' 'ppp174' 'ppp175' 'ppp176' 'ppp177' 'ppp178' 'ppp179' 'ppp180' 'ppp181' 'ppp182' 'ppp183' 'ppp184' 'ppp185' 'ppp186' 'ppp187' 'ppp188' 'ppp189' 

'ppp190' 'ppp191' 'ppp192' 'ppp193' 'ppp194' 'ppp195' 'ppp196' 'ppp197' 'ppp198' 'ppp199' 'ppp200' 'ppp201' 'ppp202' 'ppp203' 'ppp204' 'ppp205' 'ppp206' 'ppp207' 'ppp208' 'ppp209' 'ppp210' 'ppp211' 'ppp212' 'ppp213' 'ppp214' 'ppp215' 'ppp216' 'ppp217' 'ppp218' 'ppp219' 

'ppp220' 'ppp221' 'ppp222' 'ppp223' 'ppp224' 'ppp225' 'ppp226' 'ppp227' 'ppp228' 'ppp229' 'ppp230' 'ppp231' 'ppp232' 'ppp233' 'ppp234' 'ppp235' 'ppp236' 'ppp237' 'ppp238' 'ppp239' 'ppp240' 'ppp241' 'ppp242' 'ppp243' 'ppp244' 'ppp245' 'ppp246' 'ppp247' 'ppp248' 'ppp249' 

'ppp250' 'ppp251' 'ppp252' 'ppp253' 'ppp254'
firewall.@zone[1]=zone
firewall.@zone[1].name='wan'
firewall.@zone[1].network='wan' 'wanb' 'wanc' 'wand'
firewall.@zone[1].input='REJECT'
firewall.@zone[1].output='ACCEPT'
firewall.@zone[1].forward='REJECT'
firewall.@zone[1].masq='1'
firewall.@zone[1].mtu_fix='1'
firewall.@forwarding[0]=forwarding
firewall.@forwarding[0].src='lan'
firewall.@forwarding[0].dest='wan'
firewall.@rule[0]=rule
firewall.@rule[0].name='Allow-DHCP-Renew'
firewall.@rule[0].src='wan'
firewall.@rule[0].proto='udp'
firewall.@rule[0].dest_port='68'
firewall.@rule[0].target='ACCEPT'
[... 403 lines cut ...]
mwan3.wand_wanb.use_member='wand_m1_w6' 'wanb_m2_w8' 'wand6_m1_w6' 'wanb6_m2_w8'
mwan3.wand_wanb.last_resort='unreachable'
mwan3.wand_wanc=policy
mwan3.wand_wanc.use_member='wand_m1_w6' 'wanc_m2_w7' 'wand6_m1_w6' 'wanc6_m2_w7'
mwan3.wand_wanc.last_resort='unreachable'
mwan3.ip_1_62=rule
mwan3.ip_1_62.proto='all'
mwan3.ip_1_62.src_ip='10.0.1.0/26'
mwan3.ip_1_62.sticky='1'
mwan3.ip_1_62.use_policy='wan_only'
mwan3.ip_65_126=rule
mwan3.ip_65_126.proto='all'
mwan3.ip_65_126.src_ip='10.0.1.64/26'
mwan3.ip_65_126.sticky='1'
mwan3.ip_65_126.use_policy='wanb_only'
mwan3.ip_129_190=rule
mwan3.ip_129_190.proto='all'
mwan3.ip_129_190.src_ip='10.0.1.128/26'
mwan3.ip_129_190.sticky='1'
mwan3.ip_129_190.use_policy='wanc_only'
mwan3.ip_193_254=rule
mwan3.ip_193_254.proto='all'
mwan3.ip_193_254.src_ip='10.0.1.192/26'
mwan3.ip_193_254.sticky='1'
mwan3.ip_193_254.use_policy='wand_only'

closed 2021-02-25 by openwrt-diy