openwrt/packages #14855: Removing uClibc++

neheb, 2021-02-22:
ping @openwrt/package-maintainers 

I plan on doing a treewide nuke of uClibc++. Most of the reason uClibc++ is in here is size (66.6 KB vs 287.3 KB). Problem is that uClibc++ is poorly maintained and targets only C++98. Note: https://git.busybox.net/uClibc++/ . Many of the recent commits are mine with several pending for months.

uClibc++ also has issues with speed in some cases: https://github.com/openwrt/packages/blob/master/net/nmap/Makefile#L9

Posted to get opinions.

aparcar, 2021-02-22:
Please clean up things

nobk, 2021-02-24:
Do I need to change these contents in my package Makefile?

```
include $(INCLUDE_DIR)/uclibc++.mk

DEPENDS:= $(CXX_DEPENDS)

```

neheb, 2021-03-04:
yes

closed 2021-03-06 by neheb

pprindeville, 2021-03-19:
Nothing useful to say, but once upon a time uClibc was the reference library for certain processors architectures...  A shame it's fallen into disrepair.

neheb, 2021-03-19:
This is for uClibc++, not uClibc. The latter survives as uClibc-ng.