openwrt/packages #14900: sane-backends: please update to recent version

lumag, 2021-02-25:
Maintainer: @luizluca 
Environment: mips, 74Kc, TP-Link Archer C7 v5, OpenWrt 19.07.6

Description:

Could you please update sane-backends package to more recent version? Package is currently stuck at version 1.0.27 (released in 2017), while the latest version 1.0.32 was released few days ago. Thus the package lacks several updated backends (e.g. pixma backend fails to support Canon LiDE 300 scanner).

luizluca, 2021-02-25:
Hi @lumag ,

We avoid doing feature updates to a stable branch. The [dev branch](https://downloads.openwrt.org/snapshots/packages/x86_64/packages/) does provide 1.0.32 while the [next openwrt stable](https://downloads.openwrt.org/releases/packages-21.02/x86_64/packages/) release 21.02 will offer 1.0.31 (it was branched before sane was updated).

There were some significant change in packaging, specially that now sane will run as non-root. You could install the snapshot version package but you'll need to manually download sane packages and force deps as libraries (libusb) changed the name. It should work (it is how I'm currently testing it). A cleaner solution would be to download 19.07 SDK and recompile it.

neheb, 2021-02-25:
You could backport to 21.02 at least since it's not out yet.

luizluca, 2021-02-27:
> You could backport to 21.02 at least since it's not out yet.

I created a new PR for 21.02 #14920

lumag, 2021-02-27:
Hi @luizluca ,

Thank you for your suggestions. I've locally backported 1.0.32, works perfectly.

Another suggestion for the packages in question. As the sane-daemon doesn't provide init.d script, I think that it should depend (or at least recommend) xinetd.  Also it might be great if by default sane-daemon would be configured to allow lan zone network access.

closed 2021-02-27 by BKPepe