openwrt/packages #14909: copy of /etc/unbound/unbound.conf to /var/lib/unbound/ gets overwritten

vollkommenheit, 2021-02-26:
Package: unbound-daemon
Version: 1.13.1-1 and maybe older too
Depends: Depends: libc, ca-bundle, libopenssl1.1, libunbound
Status: install user installed
Section: net
Architecture: mips_24kc

Maintainer: @EricLuehrsen
Environment: Seen on ramips-mt7621 but *not* on ath79-generic-netgear_wndr3800

Description:
On the new ramips-mt7621 based router running 
OpenWrt SNAPSHOT r15963-512229ce49 I'm setting up,  /usr/lib/unbound/unbound.sh does not copy /etc/unbound/unbound.conf to /var/lib/unbound/, or alternatively, that it does so, only to have the unbound.conf from /etc/unbound/ overwritten later by the stock unbound.conf generated by /etc/init.d/unbound. If I add the "cp -p $UB_ETCDIR/*.conf $UB_VARDIR/" block after the last "if...fi" but before "resolv_setup", then the unbound.conf from /etc/unbound/ survives being overwritten in /var/lib/unbound/.

This is 
I don't have this issue on ath79-generic-netgear_wndr3800 running OpenWrt SNAPSHOT r15949-ce8b535ed3 (which is only a few days newer) or any snapshot versions before.

EricLuehrsen, 2021-02-28:
Yes that is the design intent. If you use `manual_conf` in UCI, then your `/etc/unbound/unbound.conf` is used. Otherwise UCI writes a new one. If you want to amend the UCI generated one, then you can add to `/etc/unbound/unbound_[srv/ext].conf.` Unbound loads certificates before it drops root privleges, so they can (should) remain in `/etc/...` or some more secure area. The only exception is the DNS root key which is regularly maintained using RFC5011. That would burn ROM so it needs to be in `/var/...` and copied back at a much slower rate.

vollkommenheit, 2021-03-01:
Ah, I see. I need to add:
>uci set 'unbound.@unbound[0].manual_conf=1'
>uci commit

to my sysupgrade routine then.

closed 2021-03-01 by vollkommenheit