openwrt/packages #14921: strongswan: gmpdh can't be built with static flags

pprindeville, 2021-02-27:
Maintainer: @stintel 
Environment: x86_64, generic, HEAD

Description:

Strongswan won't build in CI/CD because `gmpdh` is being built with "baked-in" flags rather than `libstrongswan_gmpdh_la_LDFLAGS` inheriting from the global settings.

`$TARGET_CFLAGS` and `$TARGET_LDFLAGS` need to go into `gmpdh`'s recipe, or it will not be built as `CONFIG_PKG_CHECK_FORMAT_SECURITY`, etc. (see `include/hardening.mk`) dictate, including `-fPIC` and `-fstack-protector`.

pprindeville, 2021-02-27:
@brainsucker-na @micmac1 Did I forget anything?

cotequeiroz, 2021-03-11:
I think I've found the problem and a solution for this.

The problem is that gmp C sources are turned into PIC code, but their asm sources are not.  They need `-DPIC` in `CFLAGS` to enable PIC code.  I've sent a patch to the ML fixing gmp, but it will benefit from further testing.

pprindeville, 2021-03-19:
Per my email, can we just add `-DPIC` to `$(FPIC)` in `rules.mk`?

pprindeville, 2021-03-19:
See PR #4006 in openwrt.

pprindeville, 2023-09-14:
Fixed via https://github.com/openwrt/openwrt/pull/4006

closed 2023-09-14 by pprindeville