openwrt/packages #14922: ca-certificates: stopping CI/CD builds from completing

pprindeville, 2021-02-27:
Maintainer: unknown
Environment: all, master

Description:

This should probably go into flyspray but I don't know how often that gets checked...

In any case, CI/CD is failing with the following:

```
make[2]: Entering directory '/home/build/openwrt/feeds/base/package/system/ca-certificates'
13645
mkdir -p /home/build/openwrt/dl
13646
SHELL= flock /home/build/openwrt/tmp/.ca-certificates_20210119.tar.xz.flock -c '  	echo "ERROR: No download method available"; false'
13647
ERROR: No download method available
13648
make[2]: *** [Makefile:71: /home/build/openwrt/dl/ca-certificates_20210119.tar.xz] Error 1
13649
make[2]: Leaving directory '/home/build/openwrt/feeds/base/package/system/ca-certificates'
13650
time: package/feeds/base/ca-certificates/compile#0.16#0.06#0.19
13651
    ERROR: package/feeds/base/ca-certificates failed to build.
```

@zhaojh329 @chunkeey @BKPepe Adding a few people who have touched this package recently so that they might add some insight.  Thanks

BKPepe, 2021-02-27:
I'd say that is related to https://github.com/openwrt/openwrt/commit/9a9cf40dd968f409cf36c9537e79438d35322d95 and @blocktrron is looking into it.

blocktrron, 2021-02-27:
This should be fixed as soon as updated docker containers have been pushed to docker hub for each arch.

The reason for build failure (if I'm not mistaken) are outdated SDKs used with updated base feeds. The SDK download logic does not know how to handle the `@DEBIAN` mirror alias and fails building on these packages.

So this will fix itself soon-ish.

neheb, 2021-02-28:
This is already fixed.

closed 2021-02-28 by neheb