openwrt/packages #14924: mwan3 ping checks depend on DNS

davidelang, 2021-02-27:
Maintainer: @\feckert
Environment: Turris omni openwrt 19.07 mwan3 version 2.8.15-1

Description:
the ping checks don't have -n (skip DNS lookups), so when there is an outage on the main link, DNS may fail for a bit before lookups succeed on the second link. During this time the ping checks on the second link can also fail, causing it to be marked down.

adding -n to the ping command line eliminates this failure mode

jamesmacwhite, 2021-02-27:
This will be an issue for Busybox ping, as it doesn't have the `-n` flag. `iputils-ping` will likely support this, but it's not a strict requirement of mwan3.

jamesmacwhite, 2021-02-27:
Ping for @feckert, as the maintainer info has a typo.

aaronjg, 2021-02-28:
It looks like busy box ping ignores the -n option, so we could add it for iputils ping and it will be ignored when iputils ping is not installed.

https://git.busybox.net/busybox/tree/networking/ping.c?id=9b6bcfda0e11c0e73a966a77110f6c68425cff34#n401

@davidelang Care to make a pull request?

davidelang, 2021-03-01:
I don't know how to tell which ping is being used, so if you don't mind, I'll watch the commit and learn rather than taking more maintainer time with my fumbling :-/

aaronjg, 2021-03-01:
It should be just these lines here:

https://github.com/openwrt/packages/blob/267b65ad9f74c854e790f0771e280b763df5f07e/net/mwan3/files/usr/sbin/mwan3track#L245

https://github.com/openwrt/packages/blob/267b65ad9f74c854e790f0771e280b763df5f07e/net/mwan3/files/usr/sbin/mwan3track#L250

Happy to help and review the P/R if you are interested in contributing. Otherwise, I can get to it once I finish debugging an LT2P mwan3 issue that I am working on.

davidelang, 2021-03-01:
that's equivalent to where I added the -n on my earlier version, but how do you make it conditional on what version of ping is being run? or does the busybox version ignore the -n so it's safe to have there in any case?

aaronjg, 2021-03-01:
> does the busybox version ignore the -n so it's safe to have there in any case?

Exactly. Busybox ping ignores the -n, so it can safely be added regardless of which ping version is used.

jamesmacwhite, 2021-03-01:
Good to know about Busybox! The only other thing I remember about iputils-ping for 19.07 is the package is super old, in snapshot its been split from the main source and much more up to date, but I thought using the 2010 version caused issues on 19.07 once installed. Unless maybe the recent kernel patch resolved the issues that occurred with iputils-ping.

feckert, 2021-03-01:
@aaronjg done

closed 2021-03-01 by feckert

BlessDeix92, 2021-08-24:
I know this issue is closed but I have some valuable info regarding this subject, which is that this issue is probably moot.

The dnsmasq DNS server/client in OpenWRT has the "allservers" option set by default. Note that the wiki currently says this option is set "0" default, but it's wrong. This option is, in fact, set on/active by default, which means that all upstream DNS servers get queried and the first received reply is used.

Further, when dnsmasq sends these queries, it send them out of the associated interfaces the DNS servers were configured for.

The result is that even if your default network route is down, as long as one of them is up, you should get a DNS reply back, even without mwan3 installed. This can be pretty confusing behavior while troubleshooting!

This can all be verified by doing a "tcpdump -i any dst port 53 &" and then doing a nslookup.

commit 7cfb2824 (2021-03-01): mwan3: disable DNS lookups for ping checks

By default, ping does a reverse DNS of the IP that you are pinging.
When you have a network issue (such as when a link has just gone down
and you haven't yet marked it down), this lookup can cause failures on
tests for links that are still good.

This option only works for iputils ping.
For busybox the option is not evaluated, but it is accepted without
throwing an error.

Fixes: #14968
Fixes: #14924

Suggested-by: David Lang <david@lang.hm>
commit 0042f2df (2021-03-01): mwan3: disable DNS lookups for ping checks

By default, ping does a reverse DNS of the IP that you are pinging.
When you have a network issue (such as when a link has just gone down
and you haven't yet marked it down), this lookup can cause failures on
tests for links that are still good.

This option only works for iputils ping.
For busybox the option is not evaluated, but it is accepted without
throwing an error.

Fixes: #14968
Fixes: #14924

Suggested-by: David Lang <david@lang.hm>
(cherry picked from commit 7cfb282432dad52aca0c2c578b43a99da4f8522a)