openwrt/packages #14925: gzip: sysupgrade error after installation

fanck0605, 2021-02-27:
Maintainer: Christian Beier <dontmind@freeshell.org>

Environment:
- FriendlyElec NanoPi R2S
- openwrt 21.02

Description:

After I installed `gzip`, the command `zcat` would actually call `gzip`.

I execute `zcat` in the terminal, it works fine.

However, `zcat` does not seem to work properly when it is called by the sysupgrade script

https://github.com/openwrt/openwrt/blob/3feef9c555e6f13fd225176c95dc5b3ce165c97d/target/linux/rockchip/armv8/base-files/lib/upgrade/platform.sh#L52

Since I am using a gzip-compressed image, the function [`get_image`](https://github.com/openwrt/openwrt/blob/master/package/base-files/files/lib/upgrade/common.sh#L91) will call zcat to decompress the firmware, but it says `/bin/zcat: exec: line 51: gzip: not found`,  thus causing the entire sysupgrade process to fail.

After I removed the gzip, everything works will.

Here is the sysupgrade log

```shell
root@OpenWrt:/# sysupgrade -v -n /tmp/openwrt-rockchip-armv8-friendlyarm_nanopi-
r2s-squashfs-sysupgrade.img.gz 
Thu Jan 21 17:10:37 CST 2016 upgrade: Reading partition table from bootdisk...

gzip: stdout: Broken pipe
Thu Jan 21 17:10:37 CST 2016 upgrade: Reading partition table from image...
Thu Jan 21 17:10:37 CST 2016 upgrade: Commencing upgrade. Closing all shell sessions.
killall: telnetd: no process killed
Thu Jan 21 17:10:38 CST 2016 upgrade: Sending TERM to remaining processes ... uhttpd vsftpd dbus-daemon avahi-daemon thd miniupnpd dnsmasq ubusd crond urngd smbd nmbd ntpd netdata netdata ttyd wsdd2 logd rpcd netifd odhcpd
Thu Jan 21 17:10:41 CST 2016 upgrade: Sending KILL to remaining processes ...
[  105.137201] sh (4300): drop_caches: 3
Thu Jan 21 17:10:43 CST 2016 upgrade: Switching to ramdisk...
[  106.938611] EXT4-fs (mmcblk0p2): re-mounted. Opts: (null)
Thu Jan 21 09:10:44 UTC 2016 upgrade: Performing system upgrade...
Thu Jan 21 09:10:45 UTC 2016 upgrade: Reading partition table from bootdisk...
/bin/zcat: exec: line 51: gzip: not found
0+0 records in
0+0 records out
Thu Jan 21 09:10:45 UTC 2016 upgrade: Reading partition table from image...
Thu Jan 21 09:10:45 UTC 2016 upgrade: Invalid partition table on /tmp/image.bs
[  107.231703] reboot: Restarting system

U-Boot TPL 2021.01 (Feb 25 2021 - 14:13:13)
```

bk138, 2021-02-27:
It seems from the [Makefile](https://github.com/openwrt/packages/blob/master/utils/gzip/Makefile) that zcat is actually installed into /usr/libexec/ and then some kind of linking is done to make /bin/zcat point there. What does `ls -la /bin/zcat` yield for you? I guess that during sysupgrade /usr/ might me unmounted, but that needs verification from someone more into it.

neheb, 2021-03-05:
ping @zhanhb

zhanhb, 2021-03-05:
IMT, the best way to solve this problem is to replace zcat with busybox zcat.
https://github.com/openwrt/openwrt/pull/3940#pullrequestreview-605439894

closed 2021-08-08 by fanck0605