openwrt/packages #14930: Joe: Syntax highlighting & other features broken

VA1DER, 2021-02-27:
Maintainer: @aTanW  (Vitaly Protsko)
Environment: All
Description: Syntax highlighting does not work on the package joe (Joe's Own Editor). The package is missing all the files that would normally go in /opt/share/joe. Specifically, all the charmaps, color schemes, language files, and most importantly the syntax highlighting files. Without them syntax highlighting doesn't work, nor are other languages available. Copying the /usr/share/joe folder from a working installation to /opt/share/joe on a target causes joe to function properly.

aTanW, 2021-02-27:
They are not missing. They are stripped to keep firmware size smaller.
If you plan development in OpenWRT, you can copy this files yourself, right?

VA1DER, 2021-02-27:
Thank-you very much for the reply!

> They are not missing. They are stripped to keep firmware size smaller.

I'm curious, is there really a pressing need to keep the joe package that small?  I did not think joe was in core.  I would have expected that the vast majority of joe installations would be on systems where space isn't such a premium.   One would think nano would be the editor people would reach for if space is that much of a premium.

If there really is a pressing need to keep the joe package that small, then is there a possibility of getting the rest of its installation included in something like joe_extras?

> If you plan development in OpenWRT...

It's not so much a case of doing development, but syntax highlighting is a very useful feature to have even just for bash editing.  Don't you find this to be the case?  Personally I feel like I'm trying to edit a bowl of oatmeal when I have a complex script in an editor without syntax highlighting.  Everything looks the same.

> ...you can copy this files yourself, right?

The issue with asking users to get the added files themselves is that it won't be immediately obvious to those not very familiar with joe's workings why syntax highlighting isn't working.  Many will just assume support was compiled out and not go to the bother of looking further.  Those that do go further then must find a working full-featured installation to move the missing files from.

I would suggest that it would be a benefit to have the full install available.  Please consider this a request for the added package.

aTanW, 2021-02-28:
New pull req:
joe: new package joe-extras #14943