openwrt/packages #14936: meta: if commit acces -> enable 2FA

aparcar, 2021-02-28:
Hi @openwrt/package-maintainers , this goes out to everyone with commit access: Please enable 2FA for your GitHub account.

There was a recent security breach where an admin account of the OpenWrt forum was taken over by stealing a password. To avoid this in the future, the core developers agreed within the last meeting to require 2FA for all OpenWrt services/infrastructure logins via password.

GitHub offers multiple ways to enable 2FA, e.g. via an app or phone number. Full information is available [here](https://docs.github.com/en/github/authenticating-to-github/securing-your-account-with-two-factor-authentication-2fa).

Sunshine

nmeyerhans, 2021-03-08:
Is it possible for the organization admins to require 2FA of all members, or to suspend access for members that don't have it set up? Asking nicely in the form of an issue is fine, but without enforcement it's not worth much.

If there's already a plan to enforce 2FA, what's the timeline for implementation?

aparcar, 2021-03-08:
Hi @nmeyerhans thanks for your interest in OpenWrts security. There are no access suspensions planed, since opening this issues many members followed. Since it's been a week I'll now ping the rest.

> Asking nicely in the form of an issue is fine, but without enforcement it's not worth much.

I hope everyone understands the advantages and no enforcement is required at any point :)

> If there's already a plan to enforce 2FA, what's the timeline for implementation?

I'll bring that up on the next meeting, I guess activating any new member receiving commit access anywhere should have it enabled as a requirement.

nmeyerhans, 2021-03-16:
> > Asking nicely in the form of an issue is fine, but without enforcement it's not worth much.
> 
> I hope everyone understands the advantages and no enforcement is required at any point :)

I would hope so, too, but even the most well intentioned people can forget. Maybe it'd be helpful to send periodic polite reminders to people who haven't yet enabled 2FA, without outright restricting them?

aparcar, 2021-04-08:
Everyone enabled 2FA by now or lost it's commit access (due to not enabling it and not responding to emails). I'll close this issue and kindly request anyone who grants commit access to future contributors to make sure 2FA is enabled.

closed 2021-04-08 by aparcar