openwrt/packages #14940: ntfs-3g fails to compile: undefined reference to 'minor'

hnyman, 2021-02-28:
cc @thess @neheb 

master up-to-date r16017-dfa0a38d1f,  ipq806x/R7800,  default musl, gcc, etc.,  Ubuntu 20.10 
compilation after "make dirclean".

ntfs-3g fails to compile. Error message is a missing "major" and "minor" at the linking stage.

```
OpenWrt-libtool: link: arm-openwrt-linux-muslgnueabi-gcc -I../include/ntfs-3g -Os -pipe -g3 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -fmacro-prefix-map=/Openwrt/r7800/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/ntfs-3g-2017.3.23-4-fuseint=ntfs-3g-2017.3.23-4-fuseint -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z -Wl,now -Wl,-z -Wl,relro -Wall -znow -zrelro -o .libs/ntfs-3g.probe ntfs_3g_probe-ntfs-3g.probe.o  -L/Openwrt/r7800/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-8.4.0_musl_eabi/usr/lib -L/Openwrt/r7800/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-8.4.0_musl_eabi/lib ../libntfs-3g/.libs/libntfs-3g.so -lc
/Openwrt/r7800/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-8.4.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/8.4.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: ../libntfs-3g/.libs/libntfs-3g.so: undefined reference to `minor'
/Openwrt/r7800/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-8.4.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/8.4.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: ../libntfs-3g/.libs/libntfs-3g.so: undefined reference to `major'
collect2: error: ld returned 1 exit status
make[6]: *** [Makefile:574: ntfs-3g.probe] Error 1
make[6]: Leaving directory '/Openwrt/r7800/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/ntfs-3g-2017.3.23-4-fuseint/src'
make[5]: *** [Makefile:504: all-recursive] Error 1
```

I last successfully compiled it two days ago for r15980-cc2d61edc3.

As there has been no change related to nfts-3g itself, my first suspect is actually something in the main OpenWrt. Looking at the commit history since cc2d61edc3, there is nothing obvious. Ignoring the unrelated mediatek changes etc., there is possibly the site.config and autoconf-lean related stuff, merged last night with 
https://github.com/openwrt/openwrt/commit/32c664ff02910bf39a3fbd5a5a4a8bff3191dd03 and
https://github.com/openwrt/openwrt/commit/f439e291304a93b982e912dc91b80ca950a594f3

neheb, 2021-02-28:
minor and major are related to a change in musl header handling. That's very strange.

neheb, 2021-02-28:
Try compiling with TARGET_CFLAGS += -Werror=implicit-function-declaration

hnyman, 2021-02-28:
Didn't help.

I copied the full package compile error log to 
https://gist.github.com/hnyman/51325860a415bc896c165a2f3bf6e1a7

This seems to be to the more detailed error section.
https://gist.github.com/hnyman/51325860a415bc896c165a2f3bf6e1a7#file-ntfs-3g_compilation_error-txt-L436-L460

```
OpenWrt-libtool: compile:  arm-openwrt-linux-muslgnueabi-gcc -DHAVE_CONFIG_H -I. -I.. -I../include/fuse-lite -I/Openwrt/r7800/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-8.4.0_musl_eabi/usr/include -I/Openwrt/r7800/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-8.4.0_musl_eabi/include/fortify -I/Openwrt/r7800/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-8.4.0_musl_eabi/include -I../include/ntfs-3g -Os -pipe -g3 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -fmacro-prefix-map=/Openwrt/r7800/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/ntfs-3g-2017.3.23-4-fuseint=ntfs-3g-2017.3.23-4-fuseint -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -Werror=implicit-function-declaration -Wall -MT libntfs_3g_la-dir.lo -MD -MP -MF .deps/libntfs_3g_la-dir.Tpo -c dir.c  -fPIC -DPIC -o .libs/libntfs_3g_la-dir.o
In file included from ../include/ntfs-3g/layout.h:28,
                 from ../include/ntfs-3g/inode.h:32,
                 from ../include/ntfs-3g/attrib.h:33,
                 from dir.c:56:
dir.c: In function '__ntfs_create':
dir.c:1640:31: error: implicit declaration of function 'major' [-Werror=implicit-function-declaration]
     data->major = cpu_to_le64(major(dev));
                               ^~~~~
../include/ntfs-3g/endians.h:125:27: note: in definition of macro '__cpu_to_le64'
 #define __cpu_to_le64(x) (x)
                           ^
dir.c:1640:19: note: in expansion of macro 'cpu_to_le64'
     data->major = cpu_to_le64(major(dev));
                   ^~~~~~~~~~~
```

Btw, I notice that there is "-Wall" coping from somewhere to be after "-Werror=implicit-function-declaration"
`-Werror=implicit-function-declaration -Wall`

neheb, 2021-02-28:
Try

```
CONFIGURE_VARS += ac_cv_header_sys_sysmacros_h=yes
```

hnyman, 2021-02-28:
Didn't help.
I also tried disabling "Wall" from configure.ac, but that diodn't help, either. The linking fails at the end.

neheb, 2021-02-28:
The issue is that https://github.com/boundarydevices/ntfs-3g/blob/imx-l5.1.1_2.1.0-ga/libntfs-3g/dir.c#L43 is not being included. There's something wrong with your setup.

hnyman, 2021-02-28:
> There's something wrong with your setup.

Hard to believe as it is a straight ipq806x R7800 build without any compilation tweaks.
The same build host has succeeded 19.07, 21.02 and master builds until today's master.

And the build was done after a nice make dirclean.

I still suspect some impact from that recent autoconf-lean thing.

We will see soon if this surfaces in buildbot.

hnyman, 2021-02-28:
Reverting the two autoconf-lean commits (regarding site config) enables the ntfs-3g compilation again. (I re-compiled toolchain from scratch).

The autoconf-lean commits have been authored ages ago by @nbd168  and were merged yesterday by @dangowrt 

The buildhost is Ubuntu 20.10 in Virtualbox in Windows. I have used it to compile 19.07, 21.02 and master without problems until today. 
Apparently last night's autoconf-lean stuff somehow affects just ntfs-3g, but none of other packages in my builds. Curious. Likely here is something peculiar/rare in nfts-3g's configure stuff.

ptpt52, 2021-03-01:
same issue.

EdiTurn, 2021-03-01:
I got similar issues when compiling rsync and bash.

neheb, 2021-03-01:
Seems GitHub CI does too...

hnyman, 2021-03-01:
Buildbot has  various breakages on different architectures, e.g. ath10k-ct, mac80211, mwlwifi, rsync, ntfs-3g.

Likely the errors depnds on the buildhost to some extent, and thus autoconf-lean provides too varied results.

cc @nbd168

hnyman, 2021-03-01:
I filed also a bug to the main OpenWrt, as this seems to be about autoconf-lean

https://bugs.openwrt.org/index.php?do=details&task_id=3655

small-5, 2021-03-01:
I have restored autoconf-lean to autoconf and reported an error:
make  install-exec-hook
make[6]: Entering directory '/home/small_5/openwrt-master/build_dir/target-x86_64_musl/ntfs-3g-2017.3.23-4-fuseext/src'
/sbin/ldconfig
/sbin/ldconfig.real: Can't create temporary cache file /etc/ld.so.cache~: Permission denied
Makefile:979: recipe for target 'install-exec-hook' failed
make[6]: *** [install-exec-hook] Error 1

neheb, 2021-03-01:
I mostly fixed it with the makedev thing. Now I get this:

```
Can't create temporary cache file /etc/ld.so.cache~: Permission denied
```

All the paths are wrong too. I don't feel like spending more time on this.

hnyman, 2021-03-02:
Looks like the autoconf-lean stuff is heading for reverting after the comments from Hauke and Felix. 

http://lists.openwrt.org/pipermail/openwrt-devel/2021-March/034120.html

I will keep this open until that is done.

httpstorm, 2021-03-02:
Caused by this commit: https://github.com/openwrt/openwrt/commit/f439e291304a93b982e912dc91b80ca950a594f3
We may have to either adapt the affected packages, fix or revert `build: use config.site generated by autoconf-lean, drop hardcoded sitefiles`.

hnyman, 2021-03-02:
Sure it is caused by it, I mentioned that commit already on my first message here.

autoconf-lean has now been reverted, so I close this.

closed 2021-03-02 by hnyman