openwrt/packages #14956: nft-qos: unable to install properly using image builder

storm1er, 2021-02-28:
Maintainer: @rosysong 
Environment: target `rpi-4`, OpenWrt latest snapshot **but may apply to every target**

## Description

When adding the nft-qos package to image builder script, we get some configuration error.  
This is due to the fact that install scripts are refering to host root instead of builder folder (*virtual system root folder?*).

Command is something like `opkg install $OPTIONS $PACKAGES` which I can't give as it's build automatically by the image builder, here's the output

```
Enabling ...
Enabling ...
Enabling ...
/home/user/openwrt-imagebuilder-bcm27xx-bcm2711.Linux-x86_64/build_dir/target-aarch64_cortex-a72_musl/root-bcm27xx/lib/nft-qos/core.sh: line 7: /lib/functions/procd.sh: No such file or directory
/home/user/openwrt-imagebuilder-bcm27xx-bcm2711.Linux-x86_64/build_dir/target-aarch64_cortex-a72_musl/root-bcm27xx/lib/nft-qos/monitor.sh: line 6: /lib/nft-qos/core.sh: No such file or directory
/home/user/openwrt-imagebuilder-bcm27xx-bcm2711.Linux-x86_64/build_dir/target-aarch64_cortex-a72_musl/root-bcm27xx/lib/nft-qos/dynamic.sh: line 6: /lib/nft-qos/core.sh: No such file or directory
/home/user/openwrt-imagebuilder-bcm27xx-bcm2711.Linux-x86_64/build_dir/target-aarch64_cortex-a72_musl/root-bcm27xx/lib/nft-qos/static.sh: line 6: /lib/nft-qos/core.sh: No such file or directory
/home/user/openwrt-imagebuilder-bcm27xx-bcm2711.Linux-x86_64/build_dir/target-aarch64_cortex-a72_musl/root-bcm27xx/lib/nft-qos/mac.sh: line 6: /lib/nft-qos/core.sh: No such file or directory
/home/user/openwrt-imagebuilder-bcm27xx-bcm2711.Linux-x86_64/build_dir/target-aarch64_cortex-a72_musl/root-bcm27xx/lib/nft-qos/priority.sh: line 6: /lib/functions/network.sh: No such file or directory
/home/user/openwrt-imagebuilder-bcm27xx-bcm2711.Linux-x86_64/build_dir/target-aarch64_cortex-a72_musl/root-bcm27xx/lib/nft-qos/priority.sh: line 7: /lib/nft-qos/core.sh: No such file or directory
Enabling nft-qos
Enabling ...
Enabling ...
```

Should be reproductible with this (did not try)
```bash
wget -q --show-progress https://downloads.openwrt.org/snapshots/targets/bcm27xx/bcm2711/openwrt-imagebuilder-bcm27xx-bcm2711.Linux-x86_64.tar.xz
tar -xf openwrt-imagebuilder-bcm27xx-bcm2711.Linux-x86_64.tar.xz
cd openwrt-imagebuilder-bcm27xx-bcm2711.Linux-x86_64
make image PACKAGES="nft-qos"
```

storm1er, 2021-02-28:
Updated reproduction step, bug confirmed

I will update if the builds work or not.

storm1er, 2021-02-28:
Update: I'm unable to activate traffic priority feature

wehagy, 2025-09-25:
Package removed in the commit https://github.com/openwrt/packages/commit/14bff6e90ef8ebe22d72b4ede398e306ea3b46da.