openwrt/packages #15003: gnunet: logspam

plntyk, 2021-03-03:
Maintainer: @dangowrt 
Environment: x86-64 VM

Description:
default / invalid configuration lead to serious log spam in logread
errors are

    Wed Mar  3 13:14:30 2021 daemon.err gnunet-service-arm[4193]: Mar 03 13:14:30-728068 namestore-8449 ERROR Could not load database backend `libgnunet_plugin_namestore'
    Wed Mar  3 13:14:30 2021 daemon.err gnunet-service-arm[4193]: Mar 03 13:14:30-732033 namestore-8450 ERROR Configuration fails to specify option `FILENAME' in section!
    Wed Mar  3 13:14:30 2021 daemon.err gnunet-service-arm[4193]: Mar 03 13:14:30-732129 namestore-8450 ERROR Unable to initialize file: (null)`

maybe disable by default (like wifi)

diffconfig.sh output for creating the VM image:
https://gist.github.com/plntyk/8b7523b883cd0be32b4c3a9c45d5825a

dangowrt, 2022-03-18:
Should be fixed now.

closed 2022-03-18 by dangowrt

DavidASmith072, 2022-03-23:
I am having this issue too with the OpenWRT version 19.07.7-c4a6851c73. I am getting the following in the system.log file during bootup:

namestore-17682 ERROR Could not load database backend `libgnunet_plugin_namestore_sqlite'
Mon Mar 21 15:58:47 2022 daemon.err gnunet-service-arm[2739]: Mar 21 15:58:47-673062 util-plugin-17683 ERROR `lt_dlopenext' failed for library `libgnunet_plugin_namestore_sqlite' with error: file not found
Mon Mar 21 15:58:47 2022 daemon.err gnunet-service-arm[2739]: Mar 21 15:58:47-673126 namestore-17683 ERROR Could not load database backend `libgnunet_plugin_namestore_sqlite'
Mon Mar 21 15:58:47 2022 daemon.err gnunet-service-arm[2739]: Mar 21 15:58:47-681408 util-plugin-17684 ERROR `lt_dlopenext' failed for library `libgnunet_plugin_namestore_sqlite' with error: file not found
Mon Mar 21 15:58:47 2022 daemon.err gnunet-service-arm[2739]: Mar 21 15:58:47-681609 namestore-17684 ERROR Could not load database backend `libgnunet_plugin_namestore_sqlite'
Mon Mar 21 15:58:47 2022 daemon.err gnunet-service-arm[2739]: Mar 21 15:58:47-690161 util-plugin-17685 ERROR `lt_dlopenext' failed for library `libgnunet_plugin_namestore_sqlite' with error: file not found
Mon Mar 21 15:58:47 2022 daemon.err gnunet-service-arm[2739]: Mar 21 15:58:47-690226 namestore-17685 ERROR Could not load database backend `libgnunet_plugin_namestore_sqlite'
Mon Mar 21 15:58:47 2022 daemon.err gnunet-service-arm[2739]: Mar 21 15:58:47-698493 util-plugin-17686 ERROR `lt_dlopenext' failed for library `libgnunet_plugin_namestore_sqlite' with error: file not found
Mon Mar 21 15:58:47 2022 daemon.err gnunet-service-arm[2739]: Mar 21 15:58:47-698694 namestore-17686 ERROR Could not load database backend `libgnunet_plugin_namestore_sqlite'
Mon Mar 21 15:58:47 2022 daemon.err gnunet-service-arm[2739]: Mar 21 15:58:47-707090 util-plugin-17687 ERROR `lt_dlopenext' failed for library `libgnunet_plugin_namestore_sqlite' with error: file not found
Mon Mar 21 15:58:47 2022 daemon.err gnunet-service-arm[2739]: Mar 21 15:58:47-707154 namestore-17687 ERROR Could not load database backend `libgnunet_plugin_namestore_sqlite'
Mon Mar 21 15:58:47 2022 daemon.err gnunet-service-arm[2739]: Mar 21 15:58:47-715464 util-plugin-17688 ERROR `lt_dlopenext' failed for library `libgnunet_plugin_namestore_sqlite' with error: file not found
Mon Mar 21 15:58:47 2022 daemon.err gnunet-service-arm[2739]: Mar 21 15:58:47-715661 namestore-17688 ERROR Could not load database backend `libgnunet_plugin_namestore_sqlite'
Mon Mar 21 15:58:47 2022 daemon.err gnunet-service-arm[2739]: Mar 21 15:58:47-724162 util-plugin-17689 ERROR `lt_dlopenext' failed for library `libgnunet_plugin_namestore_sqlite' with error: file not found
Mon Mar 21 15:58:47 2022 daemon.err gnunet-service-arm[2739]: Mar 21 15:58:47-724227 namestore-17689 ERROR Could not load database backend `libgnunet_plugin_namestore_sqlite'
Mon Mar 21 15:58:47 2022 daemon.err gnunet-service-arm[2739]: Mar 21 15:58:47-732494 util-plugin-17690 ERROR `lt_dlopenext' failed for library `libgnunet_plugin_namestore_sqlite' with error: file not found
Mon Mar 21 15:58:47 2022 daemon.err gnunet-service-arm[2739]: Mar 21 15:58:47-732693 namestore-17690 ERROR Could not load database backend `libgnunet_plugin_namestore_sqlite'
Mon Mar 21 15:58:47 2022 daemon.err gnunet-service-arm[2739]: Mar 21 15:58:47-741194 util-plugin-17691 ERROR `lt_dlopenext' failed for library `libgnunet_plugin_namestore_sqlite' with error: file not found
Mon Mar 21 15:58:47 2022 daemon.err gnunet-service-arm[2739]: Mar 21 15:58:47-741259 namestore-17691 ERROR Could not load database backend `libgnunet_plugin_namestore_sqlite'

What software package can be removed to stop this?

dangowrt, 2022-03-24:
The GNUnet life network of today (0.16.x) is incompatible with GNUnet version in OpenWrt 19.07 (GNUnet 0.11.5) and 21.02 (GNUnet 0.13.3). I may backport up-to-date GNUnet to 21.02, as current 0.16.02 actually works pretty well...