openwrt/packages #15026: *** Failed to download the package list from http://downloads.lede-project.org/releases/17.01.4/targets/ramips/mt7620/packages/Packages.gz

predators46, 2021-03-05:
` root@LEDE:~# opkg update
Downloading http://downloads.lede-project.org/releases/17.01.4/targets/ramips/mt7620/packages/Packages.gz
*** Failed to download the package list from http://downloads.lede-project.org/releases/17.01.4/targets/ramips/mt7620/packages/Packages.gz   `

neheb, 2021-03-05:
Seems so. I can access it now.

predators46, 2021-03-06:
> Seems so. I can access it now.

i just tried and still got error. a few days ago everything was no problem

`  root@LEDE:~# opkg update                                                  Downloading http://downloads.lede-project.org/releases/17.01.4/targets/ramips/mt7620/packages/Packages.gz
*** Failed to download the package list from http://downloads.lede-project.org/releases/17.01.4/targets/ramips/mt7620/packages/Packages.gz

Downloading http://downloads.lede-project.org/releases/17.01.4/packages/mipsel_24kc/base/Packages.gz
Updated list of available packages in /var/opkg-lists/reboot_base
Downloading http://downloads.lede-project.org/releases/17.01.4/packages/mipsel_24kc/base/Packages.sig
Signature check passed.
Downloading http://downloads.lede-project.org/releases/17.01.4/packages/mipsel_24kc/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/reboot_luci
Downloading http://downloads.lede-project.org/releases/17.01.4/packages/mipsel_24kc/luci/Packages.sig
Signature check passed.
Downloading http://downloads.lede-project.org/releases/17.01.4/packages/mipsel_24kc/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/reboot_packages
Downloading http://downloads.lede-project.org/releases/17.01.4/packages/mipsel_24kc/packages/Packages.sig
Signature check passed.
Downloading http://downloads.lede-project.org/releases/17.01.4/packages/mipsel_24kc/routing/Packages.gz
Updated list of available packages in /var/opkg-lists/reboot_routing
Downloading http://downloads.lede-project.org/releases/17.01.4/packages/mipsel_24kc/routing/Packages.sig
Signature check passed.
Downloading http://downloads.lede-project.org/releases/17.01.4/packages/mipsel_24kc/telephony/Packages.gz
Updated list of available packages in /var/opkg-lists/reboot_telephony
Downloading http://downloads.lede-project.org/releases/17.01.4/packages/mipsel_24kc/telephony/Packages.sig
Signature check passed.
Collected errors:
 * opkg_download: Failed to download http://downloads.lede-project.org/releases/17.01.4/targets/ramips/mt7620/packages/Packages.gz, wget returned 8.root@LEDE:~# `
Plisss fixxx

neheb, 2021-03-06:
DNS issue on your end?

predators46, 2021-03-06:
> DNS issue on your end?

not a dns problem. i've tried imagebuilder 17.01.4 openwrt on my amazon vps and got the same error

predators46, 2021-03-06:
> DNS issue on your end?

It looks like the Packages.gz file on the repo doesn't match and gets an error. I tried again at imagebuilder openwrt 17.01.4 getting stuck error and can't download at this url http://downloads.lede-project.org/releases/17.01.4/targets/ramips/mt7620/packages/Packages.gz

predators46, 2021-03-06:
> > DNS issue on your end?
> 
> not a dns problem. i've tried imagebuilder 17.01.4 openwrt on my amazon vps and got the same error

Solved

http://mirror.sjtu.edu.cn/openwrt/releases/17.01.4/targets/ramips/mt7620/packages/

closed 2021-03-06 by predators46