openwrt/packages #15032: meson: Meson-based builds do not use changed options on reconfigure

NeoRaider, 2021-03-06:
Maintainer: @dhewg
Environment: (host build, 19.07 branch, newer versions likely affected as well)

Description:

When changing the options of a package (e.g. via menuconfig) and the package sets PKG_CONFIG_DEPENDS correctly, all build steps starting with `Configure` are repeated with the new settings. For Meson-based builds (observed with the `fastd` package), this doesn't have any effect, because `meson setup` ignores new passed options when it has already been configured and `--reconfigure` is not passed. When `--reconfigure` is passed, the passed options are handled as expected (but only when a configured build directory already exists - otherwise passing `--reconfigure` will make the setup fail).

Unfortunately, this solution isn't perfect either: While `meson setup` updates the build config using the flags passed in `MESON_ARGS`, options that were in `MESON_ARGS` in a previous build, but have disappeared now, are not reverted to their default values, which seems very fragile to me. Not even the `--wipe` option reverts to default settings.

I haven't found any option in the docs to "revert everything to the defaults, then apply the passed options". Should we just remove the whole build directory for the package at the start of `Configure`? This might lead to more rebuilds than necessary, but at least there won't be any surprises.

Just deleting `meson-*` seems to have the desired effect as well, and doesn't require a full rebuild afterwards, but it feels more like a hack than a proper solution...

dhewg, 2021-03-09:
[0001-meson-reconfigure.patch.gz](https://github.com/openwrt/packages/files/6107342/0001-meson-reconfigure.patch.gz)
I don't really like it, but maybe like this?

NeoRaider, 2021-03-09:
As I wrote, this is not sufficient. It will not revert options to the default that are not in MESON_ARGS anymore.

dhewg, 2021-03-10:
Yeah, but is that a problem? As far as packages are concerned they always pass `-Dfeature=$(enabled or disabled)`, so the arg set is the same, the values may change. That fixes toggling a config value, which is the important user facing issue.

It's obviously a problem if we as package maintainers add/remove options and then try to build incremental. But I don't see a nice solution either, but since it's not a user facing issue I'd rather not add that rm hack.

commit f1c40e38 (2021-03-18): meson: reconfigure on incremental builds

When toggling a package specific config knob, we need to pass --reconfigure
to the meson setup command if an earlier build already exists.

Fixes #15032.
closed 2021-03-18 by neheb