openwrt/packages #15041: mwan3: ifdown does not affect the selection of the default gateway.

DaniilMir, 2021-03-07:
Maintainer: @\feckert
Environment: 
Model	                Netgear Nighthawk X4S R7800
Firmware Version	OpenWrt 19.07.7 r11306-c4a6851c72 / LuCI openwrt-19.07 branch git-21.044.30835-34e0d65
Kernel Version  	4.14.221
mwan3 Version 	2.8.15-1

Description:
I installed and configured an USB LTE Modem for a backup link. Successfully tested it by disabling the wired connection.
After that I installed and configured the mwan3 package including the Luci interface module.
Scope is to failover on 4G link while wired connection is down and failback on wired connection state changed to Up. 
It works pretty well when I manually disable and enable connection from the Lucy UI. 
Tracert command is showing that mobile operator's gateway is in use while wired connection is down. 
I started testing the ping check mechanism of the mwan3 and if is not changing anything. 
I used an IP address which does not respond on ICMP as a Tracking IP for wired connection. 
UI is showing : 

```
Interface: wan
Status: Offline
```

But routing is still going trough the wired connection gateway. 
I also tried to keep the gateway IP address and increase option reliability value to simulate the link failure. 
It gave me the same results. 

**Expected behavior :** 
Once interface changed status to "offline" on the load balancing page it should redirect all the traffic to a backup interface 
regardless of physical status of the link or default gateway ping. 

**Configuration :** 

```
cat /etc/config/mwan3
config member 'WAN'
        option interface 'wan'
        option metric '1'
        option weight '10'

config member 'LTE'
        option interface '4G'
        option metric '250'
        option weight '1'        

config globals 'globals'
        option rtmon_interval '3'
        option logging '1'
        option loglevel 'debug'
        option mmx_mask '0xf0'

config interface 'wan'
        option enabled '1'
        option family 'ipv4'
        option track_method 'ping'
        option count '4'
        option size '56'
        option max_ttl '60'
        option check_quality '0'
        option failure_interval '5'
        option recovery_interval '5'
        option down '3'
        option up '3'
        option initial_state 'offline'
        list track_ip '%default gateway address%'
        option timeout '1'
        option interval '1'
        option reliability '2'

config interface '4G'
        option enabled '1'
[... 144 lines cut ...]
Chain mwan3_ifaces_in (1 references)
 pkts bytes target     prot opt in     out     source               destination         
 3053  428K mwan3_iface_in_4G  all  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0xf0
 2767  402K mwan3_iface_in_wan  all  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0xf0

Chain mwan3_policy_Both_UP (0 references)
 pkts bytes target     prot opt in     out     source               destination         
    0     0 MARK       all  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0xf0 /* 4G 1 1 */ MARK xset 0x20/0xf0

Chain mwan3_policy_LTE_Only (0 references)
 pkts bytes target     prot opt in     out     source               destination         
    0     0 MARK       all  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0xf0 /* 4G 1 1 */ MARK xset 0x20/0xf0

Chain mwan3_policy_Wan_Only (0 references)
 pkts bytes target     prot opt in     out     source               destination         
    0     0 MARK       all  --  *      eth0.2  0.0.0.0/0            0.0.0.0/0            mark match 0x0/0xf0 /* out wan eth0.2 */ MARK or 0xf0
    0     0 MARK       all  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0xf0 /* unreachable */ MARK xset 0xe0/0xf0

Chain mwan3_rules (1 references)
 pkts bytes target     prot opt in     out     source               destination
`

```
Formating code blocks by wrapping them with pairs of ```
```

feckert, 2021-03-13:
I can not see a rule in your configuration? https://github.com/openwrt/packages/blob/master/net/mwan3/files/etc/config/mwan3#L115

DaniilMir, 2021-03-14:
@feckert Thanks you very much for your advise!!!
After adding a default rule it started working as expected. 
I also mentioned a weird behavior of tracert command in windows. 
When I was simulating "link down" on wan connection my steps were : 

- trace route to address say google.com
- check if it is using wan connection 
- change Tracking reliability value to make wan link fail.
- trace route to google.com again.

Second trace was still using wan link (it was not down physically)
But when I changed the address to say microsoft.com - it used the LTE link. 
Exactly after this check I was checking again trace route google.com and it still was using the wan link) 
Pingplotter was able to track the routing change without any issues.

Offtopic : 
Also I tried to use default state "offline" for both links. And it was hanging on failback change 
WAN->LTE ok LTE->WAN hang. Luci UI was showing both interfaces as "disabled" on Status -> Load Balancing page.
When I tried to click on "detail" tab UI became irresponsible. Console was OK no memory leaks detected.
Only a full restart helped to recover this state. I was able to reproduce it twice, but didn't capture the logs. 
After I switched both interface to have default state "online" - it started working without any issues.

feckert, 2021-03-15:
> I also mentioned a weird behavior of tracert command in windows.
> When I was simulating "link down" on wan connection my steps were :
> 
>     * trace route to address say google.com
> 
>     * check if it is using wan connection
> 
>     * change Tracking reliability value to make wan link fail.
> 
>     * trace route to google.com again.
> 
> 
> Second trace was still using the wan link (it was not down physically)
> But when I changed the address to say microsoft.com - it used the LTE link.
> Exactly after this check I was checking again trace route google.com and it still was using the wan link)
> Pingplotter was able to track the routing change without any issues.

I think this has to do with conntrack. You have to add the option `list flush_conntrack '<value>'` to your interface section.
The value could be connectd, disconnected, ifup or ifdown. This value flushes the iptables connctrack on interfaces events.

> Offtopic :
> Also I tried to use default state "offline" for both links. And it was hanging on failback change
> WAN->LTE ok LTE->WAN hang. Luci UI was showing both interfaces as "disabled" on Status -> [...]

That's news to me. I use exactly this setup and it works for me.

DaniilMir, 2021-03-15:
I enabled all options on both interfaces for the "Flush conntrack table" setting. 
Switched interfaces back to default "online" option and it works stable right now. 
Don't know what caused this issue. Sometimes even a DBA can break everything like QA does ) 
Thank you again for your help. 
I'm new to openwrt, but more I use it - more I like it.

closed 2023-05-03 by feckert