openwrt/packages #15045: Problem with "https-dns-proxy" after reconnecting WAN

M33FT, 2021-03-08:
Please make sure that the issue subject starts with `<package-name>: `

Also make sure that the package is maintained in this repository and not in base which should be submitted at https://bugs.openwrt.org or in the LuCI repository which should be submitted at https://github.com/openwrt/luci/issues.

Issues related to releases below 18.06 and forks are not supported or maintained and will be closed.

# Issue template (remove lines from top till here)

Maintainer: @\<github-user> (find it by checking history of the package Makefile)
Environment: (put here arch, model, OpenWrt version)

Description:

```
Formating code blocks by wrapping them with pairs of ```
```

Aloki, 2021-10-14:
My ISP uses PPPoE protocol, and sometimes the connection drops (either manually or by the ISP). After reconnection, the https-dns-proxy is **reloaded** and the service stops responding for 15 minutes. Messages "Connecting and sending request to resolver was successful, but no response was sent back" appear in the log during this time. Only **restarting** the service helps (or the problem disappears after 15 minutes). In the https://github.com/aarond10/https_dns_proxy/issues/106#issuecomment-858459047, the developer specified restart, not reload.

@M33FT you can edit /etc/hotplug.d/iface/90-https-dns-proxy file and replace reload with restart.

stangri, 2021-10-14:
This has been addressed in https://github.com/openwrt/packages/pull/16776 and will be backported to 21.02 once merged for master branch.