openwrt/packages #15060: podman: should require runc package

flyn-org, 2021-03-08:
Maintainer: @dangowrt
Environment: x86_64 master

Description:
I installed the podman package, but I found that running `podman pull alpine` produced the error "Error: no valid executable found for OCI runtime /usr/sbin/runc: invalid argument." I installed the runc package, and then the `pull` succeeded. Should the podman package depend on the runc package?

dangowrt, 2021-03-08:
Actually that still doesn't make it a strict dependency, as one can use other runtime environments as well. I'm working on `uxc` to be suitable for that (it's not yet).

dangowrt, 2021-03-29:
Now we got both `runc` and `crun` in available with https://github.com/openwrt/packages/commit/01d1a4969e258f7b08a1284e30bfa9bceba62b9c , so users can choose.

closed 2021-03-29 by dangowrt

paper42, 2021-08-17:
@dangowrt @oskarirauta Do you think it would be possible to make podman depend on crun (because it's 18x smaller than runc) and optionally on runc? Right now users have to see the error and manually install either runc or crun which isn't ideal.

```
size    package
 169442 crun_0.20-1_arm_cortex-a9_vfpv3-d16.ipk
3176827 runc_1.0.1-1_arm_cortex-a9_vfpv3-d16.ipk
```