openwrt/packages #15063: podman: segfault when running container image

flyn-org, 2021-03-08:
Maintainer: @dangowrt 
Environment: x86_64 master

Description:
I find that podman/conmon crashes when I try to execute a container image with an interactive shell:
```
# podman pull alpine
# podman run -it alpine
[  101.682694] IPv6: ADDRCONF(NETDEV_CHANGE): veth01ab2596: link becomes ready
[  101.687520] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[  101.705175] cni-podman0: port 1(veth01ab2596) entered blocking state
[  101.709576] cni-podman0: port 1(veth01ab2596) entered disabled state
[  101.715579] device veth01ab2596 entered promiscuous mode
[  101.721729] cni-podman0: port 1(veth01ab2596) entered blocking state
[  101.725899] cni-podman0: port 1(veth01ab2596) entered forwarding state
[  104.044390] conmon[2261]: segfault at 8 ip 00007fa08d8a0877 sp 00007ffcb018be10 error 4 in libglib-2.0.so.0.6600.]
[  104.052939] Code: eb d2 31 c0 5b 5d 41 5c 41 5d 41 5e c3 48 89 d1 31 d2 e9 9b ff ff ff 90 41 55 49 89 d5 41 54 49f
[  104.674975] cni-podman0: port 1(veth01ab2596) entered disabled state
[  104.701286] device veth01ab2596 left promiscuous mode
[  104.706079] cni-podman0: port 1(veth01ab2596) entered disabled state
Error: container create failed (no logs from conmon): EOF
```

The same works fine with docker on the same host:
```
# docker run -it alpine
Unable to find image 'alpine:latest' locally
latest: Pulling from library/alpine
ba3557a56b15: Pull complete 
Digest: sha256:a75afd8b57e7f34e4dad8d65e2c7ba2e1975c795ce1ee22fa34f8cf46f96a3be
Status: Downloaded newer image for alpine:latest
[  496.116917] docker0: port 1(veth0f1b27d) entered blocking state
[  496.120632] docker0: port 1(veth0f1b27d) entered disabled state
[  496.125985] device veth0f1b27d entered promiscuous mode
[  501.403819] eth0: renamed from veth05f5e12
[  501.435133] IPv6: ADDRCONF(NETDEV_CHANGE): veth0f1b27d: link becomes ready
[  501.439869] docker0: port 1(veth0f1b27d) entered blocking state
[  501.442986] docker0: port 1(veth0f1b27d) entered forwarding state
[  501.476341] IPv6: ADDRCONF(NETDEV_CHANGE): docker0: link becomes ready
/ # exit
```
I ran podman again after docker, just in case docker installed something podman needed. The outcome remained a crash.

dangowrt, 2021-03-08:
Can you try building `conmon` and `glib2` with `CONFIG_DEBUG=y` and then attach it to gdb to see what's going on?

flyn-org, 2021-03-09:
I am trying to gather some information using gdb. Catching runc/conmon with GDB, as run by podman, is difficult because podman is written in Go. I played a game where I moved runc to runc.orig and replaced runc with
```
#!/bin/sh
                                                                                                                     
gdb /usr/sbin/runc.orig -ex "set height unlimited" -ex "set follow-fork-mode parent" -ex "run $*" -ex bt -ex quit 2>&1
```, but this does not work.

I am now trying to capture the full runc command with strace so that I can try to run it directly ...

dangowrt, 2021-03-29:
@oskarirauta fixed this in https://github.com/openwrt/packages/pull/15305 with https://github.com/openwrt/packages/commit/463227f70cc75579a93a59c09cb6b372c48c13cc

closed 2021-03-29 by dangowrt