openwrt/packages #15071: acme, nginex: acme can't stop nginex instance

majkrzak, 2021-03-09:
Maintainer: acme: @yangfl  nginx: @neheb, @peter-stadler 
Environment: (put here arch, model, OpenWrt version)

Description:
---------------

Acme is not able to restart nginex server, because it is not getting properly detected.

```
port80 listens: 2868/uci.conf 2868/uci.conf
Cannot run in standalone mode; another daemon is listening on port 80.
```

In `netstat -nptl` command (used by acme to determinate what service is listening) command it is decribed as:

```
tcp        0      0 0.0.0.0:80              0.0.0.0:*               LISTEN      2868/uci.conf -g da
```

in `ps` it figgures as:

```
nginx: master process /usr/sbin/nginx -c /etc/nginx/uci.conf -g daemon off;
```

yangfl, 2021-03-09:
acme maintainer is not me but tohojo

majkrzak, 2021-03-09:
Sorry,  package Makefile history tricked me.

tohojo, 2021-03-11:
Piotr Majkrzak ***@***.***> writes:

> Maintainer: acme: @tohojo  nginx: @neheb, @peter-stadler 
> Environment: OpenWrt SNAPSHOT r16142-12dbad1a86 / LuCI Master git-21.060.51374-cd06e70
>
> Description:
> ---------------
>
> Acme is not able to restart nginex server, because it is not getting properly detected.
>
> ```
> port80 listens: 2868/uci.conf 2868/uci.conf
> Cannot run in standalone mode; another daemon is listening on port 80.
> ```
>
> In `netstat -nptl` command (used by acme to determinate what service is listening) command it is decribed as:
>
> ```
> tcp        0      0 0.0.0.0:80              0.0.0.0:*               LISTEN      2868/uci.conf -g da
> ```

Huh, that's odd. Where is your 'netstat' coming from? (i.e., what's the
output of `which netstat` and `netstat --version`?)

majkrzak, 2021-03-11:
It is busybox builtin,

```
root@circle:~# opkg info busybox
Package: busybox
Version: 1.33.0-1
Depends: libc
Conflicts: busybox-selinux
Status: install user installed
Section: base
Essential: yes
Architecture: arm_cortex-a15_neon-vfpv4
Size: 235551
Filename: busybox_1.33.0-1_arm_cortex-a15_neon-vfpv4.ipk
Description: The Swiss Army Knife of embedded Linux.
 It slices, it dices, it makes Julian Fries.
Installed-Time: 1615196766
```

tohojo, 2021-03-11:
So why is it not just output'ing the binary name? Is this a recent change?

majkrzak, 2021-03-13:
It looks like it got truncated in weird way. `/proc/../cmdline` contains proper value. And it is how I fixed it on my local.

tohojo, 2021-03-26:
Could you please test if that PR fixes it? :)

PR #15275 (2021-03-26): acme: Get listener name from /proc/PID/exe instead of netstat output
It seems the command name output from netstat can be truncated in weird
ways, so let's get the binary name from /proc instead and use that for
matching which listener we have.

Fixes #15071.

Maintainer: me

Posting this as a draft so @majkrzak can test it :)
commit c9312769 (2021-03-26): acme: Get listener name from /proc/PID/exe instead of netstat output

It seems the command name output from netstat can be truncated in weird
ways, so let's get the binary name from /proc instead and use that for
matching which listener we have.

Fixes #15071.
closed 2021-03-26 by tohojo