openwrt/packages #15077: json-glib: compilation with full language support fails

plntyk, 2021-03-10:
Maintainer: @micmac1
Environment: x86_64 VM

Description: reading to #13038 it seems that this should be in a fixed state

Maybe the ninja change exposed it ?
3/25 fails

```
FAILED: json-glib/libjson-glib-1.0.so.0.600.2.p/json-path.c.o 

x86_64-openwrt-linux-musl-gcc 
-Ijson-glib/libjson-glib-1.0.so.0.600.2.p -Ijson-glib -I../json-glib -I. -I.. 
-IXXX/source/staging_dir/target-x86_64_musl/usr/include 
-IXXX/source/staging_dir/target-x86_64_musl/usr/include/glib-2.0 
-IXXX/source/staging_dir/target-x86_64_musl/usr/lib/glib-2.0/include 
-IXXX/source/staging_dir/target-x86_64_musl/usr/lib/libiconv-full/include 
-IXXX/source/staging_dir/target-x86_64_musl/usr/lib/libintl-full/include 
-IXXX/source/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/include 
-IXXX/source/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify 
-IXXX/source/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include 
-fdiagnostics-color=always 
-pipe -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -std=c99 -Os -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result 
-fmacro-prefix-map=XXX/source/build_dir/target-x86_64_musl/json-glib-1.6.2=json-glib-1.6.2 
-Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -fPIC -pthread 
'-DG_LOG_DOMAIN="Json"' -DG_LOG_USE_STRUCTURED=1 -DJSON_COMPILATION 
'-DJSON_LOCALEDIR="/usr/share/locale"' -Wcast-align -Wlogical-op -Wmissing-declarations -Wmissing-format-attribute -Wmissing-prototypes -Wmissing-noreturn -Wold-style-definition -Wpointer-arith -Wshadow -Wstrict-prototypes -Wunused -Wno-discarded-qualifiers -Wno-int-conversion -fno-strict-aliasing -Wno-uninitialized -Werror=address -Werror=array-bounds -Werror=empty-body -Werror=implicit -Werror=init-self -Werror=int-to-pointer-cast -Werror=main -Werror=missing-braces -Werror=nested-externs -Werror=nonnull -Werror=pointer-to-int-cast -Werror=return-type -Werror=sequence-point -Werror=trigraphs -Werror=undef -Werror=write-strings 
-fvisibility=hidden -MD -MQ 
json-glib/libjson-glib-1.0.so.0.600.2.p/json-path.c.o -MF json-glib/libjson-glib-1.0.so.0.600.2.p/json-path.c.o.d -o json-glib/libjson-glib-1.0.so.0.600.2.p/json-path.c.o -c ../json-glib/json-path.c

In file included from XXX/source/staging_dir/target-x86_64_musl/usr/include/glib-2.0/glib/gi18n-lib.h:23,
                 from ../json-glib/json-path.c:160:
XXX/source/staging_dir/target-x86_64_musl/usr/lib/libintl-full/include/libintl.h:465:8: error: unknown type name 'locale_t'
 extern locale_t newlocale (int, const char *, locale_t);
        ^~~~~~~~
```

also
```
find . -name "glib-2.0.pc"
./build_dir/target-x86_64_musl/glib-2.66.4/openwrt-build/meson-private/glib-2.0.pc
./build_dir/target-x86_64_musl/glib-2.66.4/ipkg-install/usr/lib/pkgconfig/glib-2.0.pc
./build_dir/hostpkg/glib-2.66.4/openwrt-build/meson-private/glib-2.0.pc
./build_dir/hostpkg/vala-0.50.2/tests/version/glib-2.0.pc
./staging_dir/target-x86_64_musl/usr/lib/pkgconfig/glib-2.0.pc
./staging_dir/hostpkg/lib/pkgconfig/glib-2.0.pc
```
the meson-private does not have -lintl flag

micmac1, 2021-03-10:
Hello @plntyk,

Thanks for reporting. This is not an issue with json-glib, though. This is due to commit [b59905f](https://github.com/openwrt/openwrt/commit/b59905f04598008458b6545f297e742a3b7414d4) in OpenWrt. If I'm not mistakend @neheb is already aware, as he has a [patch](https://patchwork.ozlabs.org/project/openwrt/patch/20210306221655.25924-1-rosenp@gmail.com/) up in patchwork.

Kind regards,
Seb

neheb, 2021-03-10:
Correct. gettext-full needs to be fixed. It’s an issue here since meson does not use gnulib I think.

plntyk, 2021-03-11:
okay thank you . locales/gettext/intl "NLS" complex is a minefield and rabbit hole

regarding "HAVE_NAMELESS_LOCALES" change - it seems to be Solaris related / with per-thread locales

ref: https://lists.gnu.org/archive/html/bug-gnulib/2018-10/msg00122.html

neheb, 2021-03-27:
This was fixed.

closed 2021-03-27 by neheb