openwrt/packages #15099: fail2ban: new package, missing lib2to3 in checks of PR from buildroot

erdoukki, 2021-03-11:
https://github.com/openwrt/packages/pull/15098

Checks failed because of missing library lib2to3 from buildroot images.

Is there any possibilities to add a host python pypi library at compile to have checks validating this PR ?

aparcar, 2021-03-11:
Isn't the correct way to add it as a package to this repo and then depend on it?

erdoukki, 2021-03-12:
Sorry, I don’t understand what you mean.
I have made a PR for the package of fail2ban which is not ok on additional build tests.
It is okay on my host, and on CI test.
Lib2to3 is necessary at build time only for these buildbot tests.
On my ubuntu buildroot the python package is already here.
Can someone help me to fix this ?

erdoukki, 2021-03-12:
```
PKG_BUILD_DEPENDS:=python3/host
HOST_BUILD_DEPENDS:=python3-lib2to3/host
```
will this be correct and sufficient ?
What the difference between PKG_B and UILD_DEPENDS and HOST_BUILD_DEPENDS ?

erdoukki, 2021-03-12:
looks like to fix the problem !
PR is no check/building correctly.

erdoukki, 2021-03-13:
```
cd /home/build/openwrt/build_dir/target-arc_archs_glibc/fail2ban-0.11.2 && ./fail2ban-2to3
Traceback (most recent call last):
  File "/home/build/openwrt/staging_dir/hostpkg/bin/2to3", line 3, in <module>
    from lib2to3.main import main
ModuleNotFoundError: No module named 'lib2to3'
Fail!
```

erdoukki, 2021-03-13:
> looks like to fix the problem !
> PR is no check/building correctly.

Not fixed, wrking without understanding why, today KO again : 
```
Traceback (most recent call last):
  File "/home/build/openwrt/staging_dir/hostpkg/bin/2to3", line 3, in <module>
    from lib2to3.main import main
ModuleNotFoundError: No module named 'lib2to3'
Fail!
```

erdoukki, 2021-03-13:
https://github.com/openwrt/packages/pull/15098/files/ffa8cbf11d3899e69cd32968857aa3aad216a64e#r593729377
After removing python3-lib2to3 dependency, the checks build do not work no more and made error happen again

erdoukki, 2021-03-14:
still not resolved with adding : HOST_BUILD_DEPENDS:=python3-lib2to3 !
I cannot add to PKG_BUILD_DEPENDS because of an error if I try to : 
`WARNING: Makefile 'package/feeds/packages/fail2ban/Makefile' has a build dependency on 'python3-lib2to3', which does not exist
`

erdoukki, 2021-03-14:
unsolved but the python package do not need no more this part, so build correctly now, tested ok on snapshot.

closed 2021-03-14 by erdoukki