openwrt/packages #15118: APCUPSD: doshutdown not being called when Initiating system shutdown

rs6b, 2021-03-14:
Maintainer: @tru7 @gentoo-root 
Environment: (wrt1200ac, OpenWrt 19.07.7)

Description:
Basically, `/etc/apcupsd/apccontrol doshutdown` does not get called when UPS reaches run-time TIMEOUT, or even when it reaches specified battery percentage for example.
This is all i see in syslog:
```
daemon.crit apcupsd[1234]: Power failure.
daemon.alert apcupsd[1234]: Battery charge below low limit.
daemon.alert apcupsd[1234]: Initiating system shutdown!
daemon.alert apcupsd[1234]: User logins prohibited
daemon.crit apcupsd[1234]: Running on UPS batteries.
```

After which, router never runs `doshutdown`. I even created a doshutdown script and put in `touch /etc/testfile` so i can see if it even runs, and it doesnt create the file
I can see that it does indeed create the `/etc/powerfail` file however i have to then manually run `/etc/apcupsd/apccontrol restartme` to clean it up together with the PID.
I do get email notification when the UPS goes on batteries. So it seems to call `/etc/apcupsd/onbattery` fine.
Ive tried using `LOGSTATS on` and havent seen anything significant in the Syslog file, besides the above notices

I know apcupsd works on debian and ubuntu systems, as i have installed and configured it countless times.

Please tell me i am doing something wrong and why doshutdown isnt getting called? or is this some kind of bug?
Would appreciate your reply.

Yes my router can indeed see the UPS when doing `apcaccess`.

tru7, 2021-03-14:
I must say that I never tested the paket on OpenWrt myself. I also run on apcupsd on my own ubuntu and just thought it was a good idea to provide apcupsd also on OpenWrt. I remember that I had to somehow redefine that system calls as they differ from linux. But I forgot the details. As I cannot test and I'm not a real expert I wont' be able to futher support this case. But I'd be happy to correct the paket if you find something for improvement.

rs6b, 2021-03-14:
I see...well apart from email notifications, nothing else seems to work in shutting down the router or powering off the UPS(which is what i want to evidently do). Im also not a real expert and the problem is, that im not sure how to check/read `/usr/sbin/apcupsd` so i can see whats going on, as its a binary file. I see that you simply copy it from somewhere in the makefile. I cannot understand at which line of code you have actually compiled it on?

Did you make any changes to `/usr/sbin/apcupsd` binary file previously?

Please tell me where to start looking and il be happy to check

tru7, 2021-03-14:
Well, the package is compiled from original source with some specific configure settings. The recipe is in the Makefile. The package is then built from the compiled binaries, some scripts and some config files as defined in the Makefile. The scripts and config files are a bit adjusted to OpenWrt environment and provided in the files directory. I just had another look at the config environment and it still looks OK to me.
At least there is no real manipulation of the original source code, just workarounds as defined in the patches directory.

As I understand the doshutdown, it is just a system call of the apccontrol script with argument doshutdown. Looking at your log lines it seems that apccontrol is not executed at all because the text is not echoed by the script but by something else.

I guess this would need some runtime debugging which is out of my scope. Maybe the way the external system command "apccontrol" is called does not really work on OpenWrt. I wonder if apccontrol is ever executed successfully at all. On the other hand it looks like the /etc/apcupsd/onbattery call works fine.