openwrt/packages #15140: libreswan: pluto selftest fail

weiyu10, 2021-03-16:
Maintainer: @lucize 
Environment:  openwrt-19.07.7-x86-64 virtualbox 

Description:

plutolog
```
Pluto initialized
Mar 16 11:26:19.533324: NSS DB directory: sql:/etc/ipsec.d
Mar 16 11:26:19.533494: Initializing NSS
Mar 16 11:26:19.533726: Opening NSS database "sql:/etc/ipsec.d" read-only
Mar 16 11:26:19.537941: NSS initialized
Mar 16 11:26:19.538090: NSS crypto library initialized
Mar 16 11:26:19.538167: FIPS HMAC integrity support [disabled]
Mar 16 11:26:19.538566: libcap-ng support [disabled]
Mar 16 11:26:19.538792: Linux audit support [disabled]
Mar 16 11:26:19.539039: Starting Pluto (Libreswan Version 3.27 XFRM(netkey) KLIPS FORK PTHREAD_SETSCHEDPRIO NSS) pid:21741
Mar 16 11:26:19.539331: core dump dir: /var/run/pluto
Mar 16 11:26:19.539840: secrets file: /etc/ipsec.secrets
Mar 16 11:26:19.539933: leak-detective disabled
Mar 16 11:26:19.540011: NSS crypto [enabled]
Mar 16 11:26:19.540186: XAUTH PAM support [disabled]
Mar 16 11:26:19.540941: NAT-Traversal support  [enabled]
Mar 16 11:26:19.541099: Initializing libevent in pthreads mode: headers: 2.1.11-stable (2010b00); library: 2.1.11-stable (2010b00)
Mar 16 11:26:19.541823: Encryption algorithms:
Mar 16 11:26:19.541953:   AES_CCM_16              IKEv1:     ESP     IKEv2:     ESP     FIPS  {256,192,*128}  aes_ccm, aes_ccm_c
Mar 16 11:26:19.542173:   AES_CCM_12              IKEv1:     ESP     IKEv2:     ESP     FIPS  {256,192,*128}  aes_ccm_b
Mar 16 11:26:19.542613:   AES_CCM_8               IKEv1:     ESP     IKEv2:     ESP     FIPS  {256,192,*128}  aes_ccm_a
Mar 16 11:26:19.543119:   3DES_CBC                IKEv1: IKE ESP     IKEv2: IKE ESP     FIPS  [*192]  3des
Mar 16 11:26:19.543365:   CAMELLIA_CTR            IKEv1:     ESP     IKEv2:     ESP           {256,192,*128}
Mar 16 11:26:19.544171:   CAMELLIA_CBC            IKEv1: IKE ESP     IKEv2: IKE ESP           {256,192,*128}  camellia
Mar 16 11:26:19.544408:   AES_GCM_16              IKEv1:     ESP     IKEv2: IKE ESP     FIPS  {256,192,*128}  aes_gcm, aes_gcm_c
Mar 16 11:26:19.544814:   AES_GCM_12              IKEv1:     ESP     IKEv2: IKE ESP     FIPS  {256,192,*128}  aes_gcm_b
Mar 16 11:26:19.545029:   AES_GCM_8               IKEv1:     ESP     IKEv2: IKE ESP     FIPS  {256,192,*128}  aes_gcm_a
Mar 16 11:26:19.545913:   AES_CTR                 IKEv1: IKE ESP     IKEv2: IKE ESP     FIPS  {256,192,*128}  aesctr
Mar 16 11:26:19.546233:   AES_CBC                 IKEv1: IKE ESP     IKEv2: IKE ESP     FIPS  {256,192,*128}  aes
Mar 16 11:26:19.546569:   SERPENT_CBC             IKEv1: IKE ESP     IKEv2: IKE ESP           {256,192,*128}  serpent
Mar 16 11:26:19.546889:   TWOFISH_CBC             IKEv1: IKE ESP     IKEv2: IKE ESP           {256,192,*128}  twofish
Mar 16 11:26:19.547213:   TWOFISH_SSH             IKEv1: IKE         IKEv2: IKE ESP           {256,192,*128}  twofish_cbc_ssh
Mar 16 11:26:19.547579:   NULL_AUTH_AES_GMAC      IKEv1:     ESP     IKEv2:     ESP           {256,192,*128}  aes_gmac
Mar 16 11:26:19.547849:   NULL                    IKEv1:     ESP     IKEv2:     ESP           []
Mar 16 11:26:19.548144:   CHACHA20_POLY1305       IKEv1:             IKEv2: IKE ESP           [*256]  chacha20poly1305
Mar 16 11:26:19.548480: Hash algorithms:
Mar 16 11:26:19.548649:   MD5                     IKEv1: IKE         IKEv2:
Mar 16 11:26:19.549038:   SHA1                    IKEv1: IKE         IKEv2:             FIPS  sha
Mar 16 11:26:19.549477:   SHA2_256                IKEv1: IKE         IKEv2:             FIPS  sha2, sha256
Mar 16 11:26:19.549894:   SHA2_384                IKEv1: IKE         IKEv2:             FIPS  sha384
Mar 16 11:26:19.550134:   SHA2_512                IKEv1: IKE         IKEv2:             FIPS  sha512
Mar 16 11:26:19.550462: PRF algorithms:
Mar 16 11:26:19.550534:   HMAC_MD5                IKEv1: IKE         IKEv2: IKE               md5
Mar 16 11:26:19.550765:   HMAC_SHA1               IKEv1: IKE         IKEv2: IKE         FIPS  sha, sha1
Mar 16 11:26:19.551003:   HMAC_SHA2_256           IKEv1: IKE         IKEv2: IKE         FIPS  sha2, sha256, sha2_256
Mar 16 11:26:19.551235:   HMAC_SHA2_384           IKEv1: IKE         IKEv2: IKE         FIPS  sha384, sha2_384
Mar 16 11:26:19.551470:   HMAC_SHA2_512           IKEv1: IKE         IKEv2: IKE         FIPS  sha512, sha2_512
Mar 16 11:26:19.551778:   AES_XCBC                IKEv1:             IKEv2: IKE         FIPS  aes128_xcbc
Mar 16 11:26:19.552003: Integrity algorithms:
Mar 16 11:26:19.552067:   HMAC_MD5_96             IKEv1: IKE ESP AH  IKEv2: IKE ESP AH        md5, hmac_md5
Mar 16 11:26:19.552280:   HMAC_SHA1_96            IKEv1: IKE ESP AH  IKEv2: IKE ESP AH  FIPS  sha, sha1, sha1_96, hmac_sha1
Mar 16 11:26:19.552496:   HMAC_SHA2_512_256       IKEv1: IKE ESP AH  IKEv2: IKE ESP AH  FIPS  sha512, sha2_512, hmac_sha2_512
Mar 16 11:26:19.552712:   HMAC_SHA2_384_192       IKEv1: IKE ESP AH  IKEv2: IKE ESP AH  FIPS  sha384, sha2_384, hmac_sha2_384
Mar 16 11:26:19.552925:   HMAC_SHA2_256_128       IKEv1: IKE ESP AH  IKEv2: IKE ESP AH  FIPS  sha2, sha256, sha2_256, hmac_sha2_256
Mar 16 11:26:19.553142:   HMAC_SHA2_256_TRUNCBUG  IKEv1:     ESP AH  IKEv2:         AH
Mar 16 11:26:19.553425:   AES_XCBC_96             IKEv1:     ESP AH  IKEv2: IKE ESP AH  FIPS  aes_xcbc, aes128_xcbc, aes128_xcbc_96
Mar 16 11:26:19.553619:   AES_CMAC_96             IKEv1:     ESP AH  IKEv2:     ESP AH  FIPS  aes_cmac
Mar 16 11:26:19.553958:   NONE                    IKEv1:     ESP     IKEv2:     ESP     FIPS  null
Mar 16 11:26:19.554460: DH algorithms:
Mar 16 11:26:19.554590:   NONE                    IKEv1:             IKEv2: IKE ESP AH  FIPS  null, dh0
Mar 16 11:26:19.554803:   MODP1024                IKEv1: IKE ESP AH  IKEv2: IKE ESP AH        dh2
Mar 16 11:26:19.555007:   MODP1536                IKEv1: IKE ESP AH  IKEv2: IKE ESP AH        dh5
Mar 16 11:26:19.555203:   MODP2048                IKEv1: IKE ESP AH  IKEv2: IKE ESP AH  FIPS  dh14
Mar 16 11:26:19.555400:   MODP3072                IKEv1: IKE ESP AH  IKEv2: IKE ESP AH  FIPS  dh15
Mar 16 11:26:19.555602:   MODP4096                IKEv1: IKE ESP AH  IKEv2: IKE ESP AH  FIPS  dh16
Mar 16 11:26:19.555798:   MODP6144                IKEv1: IKE ESP AH  IKEv2: IKE ESP AH  FIPS  dh17
Mar 16 11:26:19.555993:   MODP8192                IKEv1: IKE ESP AH  IKEv2: IKE ESP AH  FIPS  dh18
Mar 16 11:26:19.556192:   DH19                    IKEv1: IKE         IKEv2: IKE ESP AH  FIPS  ecp_256
Mar 16 11:26:19.556394:   DH20                    IKEv1: IKE         IKEv2: IKE ESP AH  FIPS  ecp_384
Mar 16 11:26:19.556621:   DH21                    IKEv1: IKE         IKEv2: IKE ESP AH  FIPS  ecp_521
Mar 16 11:26:19.556793:   DH31                    IKEv1: IKE         IKEv2: IKE ESP AH        curve25519
Mar 16 11:26:19.557346: NSS: AEAD decryption using AES_GCM_16_128 and PK11_Decrypt() failed (SECERR: 2 (0x2): security library: received bad data.)
Mar 16 11:26:19.557440: NSS: AEAD encryption using AES_GCM_16_128 and PK11_Encrypt() failed (SECERR: 2 (0x2): security library: received bad data.)
Mar 16 11:26:19.557777: NSS: AEAD decryption using AES_GCM_16_128 and PK11_Decrypt() failed (SECERR: 2 (0x2): security library: received bad data.)
Mar 16 11:26:19.557929: NSS: AEAD encryption using AES_GCM_16_128 and PK11_Encrypt() failed (SECERR: 2 (0x2): security library: received bad data.)
Mar 16 11:26:19.558182: NSS: AEAD decryption using AES_GCM_16_128 and PK11_Decrypt() failed (SECERR: 2 (0x2): security library: received bad data.)
Mar 16 11:26:19.558332: NSS: AEAD encryption using AES_GCM_16_128 and PK11_Encrypt() failed (SECERR: 2 (0x2): security library: received bad data.)
Mar 16 11:26:19.558586: NSS: AEAD decryption using AES_GCM_16_128 and PK11_Decrypt() failed (SECERR: 2 (0x2): security library: received bad data.)
Mar 16 11:26:19.558732: NSS: AEAD encryption using AES_GCM_16_128 and PK11_Encrypt() failed (SECERR: 2 (0x2): security library: received bad data.)
Mar 16 11:26:19.558938: ABORT: ASSERTION FAILED: test_gcm_vectors(&ike_alg_encrypt_aes_gcm_16, aes_gcm_tests) (in test_ike_alg() at ike_alg_test.c:34)
```

lucize, 2021-03-16:
Maybe you are missing some crypto kernel modules, not all modules are added as a dependency on the default installation

nlindq-maei, 2021-03-16:
I'm having the same issue. Noting your reply, I manually installed all available kmod-crypto modules (with the exception of the hardware offloading modules as they're not relevant) but the problem persists.

I did note that the installed version of libreswan is 3.27, and the installed libnss version is 3.55.

There seems to have been a libreswan issue revealed between libnss 3.51 and 3.52 reported [here](https://github.com/libreswan/libreswan/issues/334) and [here](https://github.com/libreswan/libreswan/issues/342) which later versions of libreswan were updated to handle; what version of libnss was the current libreswan built against?

My setup is OpenWRT 19.07.7 version running on Linksys WRT3200ACM, btw.

weiyu10, 2021-03-17:
@lucize 
I manually installed all available kmod-crypto modules,but the problem persists.
lsmod:
```
ablk_helper            12288  7 twofish_avx_x86_64,serpent_avx2,serpent_avx_x86_64,cast6_avx_x86_64,cast5_avx_x86_64,camellia_aesni_avx2,camellia_aesni_avx_x86_64
af_alg                 16384  2 algif_skcipher,algif_hash
af_key                 28672  0
ah4                    16384  0
ah6                    12288  0
akcipher               12288  1 rsa_generic
algif_hash             12288  0
algif_skcipher         12288  0
anubis                 16384  0
asn1_decoder           12288  1 rsa_generic
authenc                12288  0
blowfish_common        16384  1 blowfish_x86_64
blowfish_x86_64        20480  0
bnx2                   69632  0
button_hotplug         12288  0
camellia_aesni_avx_x86_64   28672  1 camellia_aesni_avx2
camellia_aesni_avx2    28672  0
camellia_generic       24576  0
camellia_x86_64        49152  2 camellia_aesni_avx2,camellia_aesni_avx_x86_64
cast_common            12288  4 cast6_avx_x86_64,cast6_generic,cast5_avx_x86_64,cast5_generic
cast5_avx_x86_64       49152  0
cast5_generic          20480  1 cast5_avx_x86_64
cast6_avx_x86_64       69632  0
cast6_generic          20480  1 cast6_avx_x86_64
cbc                    12288  0
ccm                    16384  0
ccp                    53248  0
cmac                   12288  0
crc_ccitt              12288  1 ppp_async
crypto_acompress       12288  1 deflate
cryptodev              36864  0
ctr                    12288  0
cts                    12288  0
deflate                 8192  0
des_generic            20480  0
drbg                   24576  0
e1000                  90112  0
e1000e                147456  0
ecdh_generic           20480  0
echainiv                8192  0
esp4                   16384  0
esp6                   16384  0
fcrypt                 12288  0
gcm                    16384  0
ghash_generic           8192  0
hmac                   12288  0
i2c_dev                12288  0
igb                   131072  0
ip_tables              16384  4 iptable_nat,iptable_mangle,iptable_filter
ip_tunnel              20480  1 ip_vti
ip_vti                 12288  0
ip6_tables             16384 24 ip6table_mangle,ip6table_filter
ip6_tunnel             28672  1 ip6_vti
ip6_vti                16384  0
ip6t_REJECT            12288  2
ip6table_filter        12288  1
ip6table_mangle        12288  1
ipcomp                 12288  0
ipcomp6                12288  0
ipt_MASQUERADE         12288  1
ipt_REJECT             12288  2
ipt_ah                  8192  0
iptable_filter         12288  1
iptable_mangle         12288  1
iptable_nat            12288  1
jitterentropy_rng      12288  0
[... 61 lines cut ...]
xfrm4_mode_transport    8192  0
xfrm4_mode_tunnel       8192  0
xfrm4_tunnel           12288  0
xfrm6_mode_beet         8192  0
xfrm6_mode_transport    8192  0
xfrm6_mode_tunnel       8192  0
xfrm6_tunnel           12288  1 ipcomp6
xt_CT                  12288  0
xt_FLOWOFFLOAD         12288  0
xt_LOG                 12288  0
xt_REDIRECT            12288  0
xt_TCPMSS              12288  4
xt_comment              8192127
xt_conntrack           12288 14
xt_esp                  8192  0
xt_limit               12288 20
xt_mac                  8192  0
xt_mark                 8192  0
xt_multiport            8192  0
xt_nat                 12288  0
xt_policy              12288  0
xt_state               12288  0
xt_tcpudp               8192 12
xt_time                12288  0
```

lucize, 2021-03-17:
I don't have at the moment a 19.07 test system, can you build your own packages ? try to copy the nspr,nss and libreswan from master and build against the same kernel it should work in replacing the release packages

nlindq-maei, 2021-03-17:
I don't currently have a build environment set up, but I might be able to do that. Is there a particular docker build image I should look for?
Alternatively, as you're not running 19.07 yourself, is there an older supported release I could install to get this working? Units aren't deployed yet so that's an option for me.

lucize, 2021-03-17:
newer versions of libreswan don't depend on kernel, you may try to use packages from master, but I'll prepare a vm also

lucize, 2021-03-17:
sadly it seems that the newer version needs the xfrm-interface kmod, so at least that driver needs to be backported to use the binary from master
the other dependencies are
libcap-ng
libevent27
libsqlite30
using just the newer nss and nspr didn't helped but now I'm a bit curious about this

nlindq-maei, 2021-03-17:
Have you tried applying the [patch for libreswan](https://github.com/libreswan/libreswan/commit/db7715407efa43cd2a66caed67c02d8f7bb90b35) referenced in one of my prior links? I expect the patch is against a newer version than 3.27, but may still work... (haven't finished setting up my own build environment yet, btw)

lucize, 2021-03-17:
@nlindq-maei yes https://github.com/openwrt/packages/blob/f98158b75518bc7096a96cc8ac7b461be149821e/net/libreswan/patches/030-fix_nss_3.52.patch fixes the build, so there is no other way than rebuild the package

lucize, 2021-03-17:
I still don't know how to apply the patch against stable branch

nlindq-maei, 2021-03-17:
I trust there's someone in the OpenWRT dev team who can help with that... in the meantime, would it be feasible to attach the updated .ipk to this ticket? I could download and apply the package manually to test the fix here; that'd get us by until stable branch is updated.

lucize, 2021-03-17:
search all the packages mentioned in the download section https://downloads.openwrt.org/snapshots/targets/

nlindq-maei, 2021-03-17:
I'm curious what the release philosophy for OpenWRT as it pertains to version upgrades might be--is it preferable to upgrade to the latest version of libreswan in the stable branch, or use the "release" version with the backported fix? If it works, I'm fine either way--just curious whether there are guidelines as to the desired approach.

lucize, 2021-03-17:
depends of your usage, I always use master on my scenario, but if you have some application that are not in tree, then you have to stick with compatibility

nlindq-maei, 2021-03-18:
I was able to install the snapshot build of libreswan 4.3-1 along with libcap-ng, libldns and libunbound, though I had to force install a couple of them.  There's no revised libevent package in the snapshot downloads, and for some reason it doesn't recognize the current stable libevent2-7 package as satisfying dependencies, but the dynamic library appears to be linked properly even so. 

It starts successfully and I was able to configure and establish a tunnel with another endpoint.

Do you know whether the "fixed" version for the stable branch will end up being 3.27 with the patch, or would you move to the latest release including the additional support libraries adapted for stable?

lucize, 2021-03-18:
I think it will be only with 3.27, libevent is in the base folder, not in packages

weiyu10, 2021-03-19:
I used [snapshot image](https://downloads.openwrt.org/snapshots/targets/x86/64/) to test the IPSec function, and the problem has been solved

nlindq-maei, 2021-03-24:
Do you know whether the patched 3.27 package has been integrated into stable yet? If not, do you know when?

lucize, 2021-03-28:
this should fix it https://github.com/openwrt/packages/pull/15295

nlindq-maei, 2021-03-29:
Yep! I uninstalled the snapshot libreswan and library dependencies I'd installed manually; then installed libreswan from the stable repo and everything is now working perfectly. Thanks very much!

cagney, 2025-04-17:
> Mar 16 11:26:19.558938: ABORT: ASSERTION FAILED: test_gcm_vectors(&ike_alg_encrypt_aes_gcm_16, aes_gcm_tests) (in test_ike_alg() at ike_alg_test.c:34)

FYI, this is libreswan testing NSS's AES_GCM implementation.  Since libreswan's been updated to  4.12, and NSS to 3.93 this bug is presumably fixed.