openwrt/packages #15143: jupp: Opening options results in "illegal operation" crash

Mrnofish, 2021-03-16:
Maintainer: @mirabilos 
Environment: armv7l, Netgear Nighthawk X4S R7800, OpenWrt 19.07.6

Description: Pressing ^O (Options) results in error message "**illegal instruction**" and **crash**.

mirabilos, 2021-03-16:
Mrnofish dixit:

>Maintainer: @mirabilos 
>Environment: armv7l, Netgear Nighthawk X4S R7800, OpenWrt 19.07.6
>
>Description: Pressing ^O (Options) results in error message "**illegal instruction**" and **crash**.

Erk.

Is this 3.1.40-1 ?

I guess I’ll have to build for a qemu/ARM target and see if I can
reproduce this… do you happen to be able to get a backtrace?

Thanks,
//mirabilos
-- 
<ch> you introduced a merge commit        │<mika> % g rebase -i HEAD^^
<mika> sorry, no idea and rebasing just fscked │<mika> Segmentation
<ch> should have cloned into a clean repo      │  fault (core dumped)
<ch> if I rebase that now, it's really ugh     │<mika:#grml> wuahhhhhh

Mrnofish, 2021-03-16:
3.1.38-1.

Don't think I can help with the backtrace, as I'm on an official image and I have no build environment installed.

mirabilos, 2021-03-16:
Mrnofish dixit:

>3.1.38-1.
>
>Don't think I can help with the backtrace, as I'm on an official image
>and I have no build environment installed.

Right… hmpf. I barely know enough about OpenWrt, this may take a bit.

Good news: I can reproduce this using the official binary image
19.07.6 for armvirt/32.

Bad news: I’ve got no idea where this could even generate SIGILL from…

bye,
//mirabilos
-- 
Solange man keine schmutzigen Tricks macht, und ich meine *wirklich*
schmutzige Tricks, wie bei einer doppelt verketteten Liste beide
Pointer XORen und in nur einem Word speichern, funktioniert Boehm ganz
hervorragend.		-- Andreas Bogk über boehm-gc in d.a.s.r

mirabilos, 2021-03-16:
Dixi quod…

>I guess I’ll have to build for a qemu/ARM target and see if I can

Erm… this is going to take a while:

| […]
| Checking 'file'... ok.
| Checking 'ldconfig-stub'... ok.
|
| Build dependency: Please install Python 2.x

There IS NO Python 2.x in Debian any more…

bye,
//mirabilos
-- 
„Cool, /usr/share/doc/mksh/examples/uhr.gz ist ja ein Grund,
mksh auf jedem System zu installieren.“
	-- XTaran auf der OpenRheinRuhr, ganz begeistert
(EN: “[…]uhr.gz is a reason to install mksh on every system.”)

Mrnofish, 2021-03-16:
> Good news: I can reproduce this using the official binary image 19.07.6 for armvirt/32.

BTW it's the same on 19.07.7.

> There IS NO Python 2.x in Debian any more…

I have an armel box running buster, definitely can install Python 2, OTOH jupp works fine there, probably no use then.

mirabilos, 2021-03-16:
Mrnofish dixit:

>> Good news: I can reproduce this using the official binary image 19.07.6 for armvirt/32.
>
>BTW it's the same on 19.07.7.

Yeah, I’m building from the tip of the 19.07 branch now…

>> There IS NO Python 2.x in Debian any more…
>
>I have an armel box running buster, definitely can install Python 2,

Yeah, I’ve set myself up a buster chroot for this now.

>OTOH jupp works fine there, probably no use then.

Yes, I test it on Debian on most platforms.

The OpenWrt build seems to behave a bit weird anyway,
or maybe it’s the combination of emulated serial console
and qemu. When editing the network config file, the tabs
are missing, whereas cat and vi show them. Maybe this is
better over ssh.

bye,
//mirabilos
-- 
22:20⎜<asarch> The crazy that persists in his craziness becomes a master
22:21⎜<asarch> And the distance between the craziness and geniality is
only measured by the success 18:35⎜<asarch> "Psychotics are consistently
inconsistent. The essence of sanity is to be inconsistently inconsistent

mirabilos, 2021-03-16:
Mrnofish dixit:

>3.1.38-1.

OK, I can reproduce this in a local build of 3.1.38-1 and it is fixed
when locally building 3.1.40-1, which is the latest package.

Looking at the changelog…

	http://www.mirbsd.org/jupp.htm#r3_1j40

… I’d have thought this to be the Coverity warnings, but it turns
out to be the “crash on Ins/^V on Mac OSX” fix; apparently musl is
as crash-happy with historically permitted strlcpy(3) usage as
Mac OSX is :/

My advice here is to upgrade, but I’ve got no idea how if you use
precompiled binary packages only…

bye,
//mirabilos
-- 
“It is inappropriate to require that a time represented as
 seconds since the Epoch precisely represent the number of
 seconds between the referenced time and the Epoch.”
	-- IEEE Std 1003.1b-1993 (POSIX) Section B.2.2.2

Mrnofish, 2021-03-16:
Tabs seem to be working fine over SSH.

I'm closing this now, thanks for looking into it.

closed 2021-03-16 by Mrnofish

mirabilos, 2021-03-17:
Thanks for the feedback and agreeing to close this as fixed.

To close the investigation, apparently OpenWrt uses [standalone `fortify-headers`](https://git.2f30.org/fortify-headers/log.html) which cause the `SIGILL` at runtime even for cases where it could have warned at compile time (constant), and it even embeds `ud2` instructions for cases it could have disproved at compile time.

I audited all remaining cases of `ud2` in a jupp binary built with them (all in “cold” code, so it knows they’re not normally hit), and they all are fine (and most of them provably so, but the compiler/toolchain doesn’t seem to be able to make use of enough context to elide them; gotta love half-arsed attempts at “security!!!1111oneeleven”).