openwrt/packages #15146: luci: package missing from release 19.07.7 packages

cris-, 2021-03-17:
Maintainer: @jow- 
Environment: 19.07.7 x86-64

Description: 
Im starting the image builder for 19.07.7 x86-64 with PACKAGES="... luci luci-ssl-openssl..." but it fails with the following error:

```
opkg_install_cmd: Cannot install package luci.
satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-ssl-openssl:
 	luci
opkg_install_cmd: Cannot install package luci-ssl-openssl.

```

The same build was working 1-2 days ago: am I wrong thinking that the luci package is missing from https://downloads.openwrt.org/releases/19.07.7/packages/x86_64/luci/ ?

Note: also reported to Luci repository: https://github.com/openwrt/luci/issues/4927

BKPepe, 2021-03-18:
Same as https://github.com/openwrt/luci/issues/4927#issuecomment-800883612. Closing.

closed 2021-03-18 by BKPepe

cris-, 2021-03-18:
Would somebody be so kind to share where such problems should be reported?

cielpy, 2021-03-19:
+1, where to feedback to the maintainer?

BKPepe, 2021-03-19:
Well, LuCI is not in this repository. This repository contains community maintained packages and we don't have anything to do with LuCI here. This could be related to upgrading build bot infrastructure and its changes, which was announced on `openwrt-devel` mailing list.  Also, [reporting bugs](https://openwrt.org/bugs) is documented.

cris-, 2021-03-19:
May I suggest to add a section for infrastructure related problems?

cielpy, 2021-03-19:
buildbot here https://downloads.openwrt.org/releases/faillogs-19.07/x86_64/luci/luci/check-compile.txt