openwrt/packages #15167: htop:

bolemo, 2021-03-19:
Maintainer: @\Etienne CHAMPETIER
Environment: cortex a15, Netgear R7800, Entware (based on OpenWRT source)

Description:

I am currently using htop 3.0.1-1 through Entware 3x for cortex a15 (router Netgear R7800).
I was told it is based on the OpenWRT source code.

While 3.0.1-1 works flawlessly (as did prior versions), after I upgraded htop to version 3.0.5-1 (using the opkg update the upgrade process), I noticed it is not working properly anymore.
Number of tasks reported is 0, and no tasks are shown at all in the list.
Other info (CPU, Mem, Swp, etc...) seems to be fine, and regular top works normal.

When downgraded back to 3.0.1-1, it worked again instantly (no need to restart).

champtar, 2021-03-19:
I don't know Entware at all, do they keep all versions they built ? ie can you test 3.0.2/3/4 ?

bolemo, 2021-03-19:
> I don't know Entware at all, do they keep all versions they built ? ie can you test 3.0.2/3/4 ?

The Entware releases I am getting are updated every 3 months, so no intermediaries.
However, Entware offers compiling tools that I have installed, so I could build any version I guess.

bolemo, 2021-03-20:
Ok, I was able to build htop 3.0.2 from the tarball release and it works.
Then I did the same with htop 3.0.3, and it shows the issue.

The problem appeared between versions 3.0.2 and 3.0.3.

champtar, 2021-03-20:
Hi @bolemo 
I just tried with a RaspberryPI 2, so cortex A7 but same instructions ARMv7-A and everything work fine with latest OpenWrt snapshot / htop 3.0.5-1
Can you try to remove `/root/.config/htop/htoprc` ?

champtar, 2021-03-20:
Also tested with 19.07
```
root@OpenWrt:~# opkg install htop
Installing htop (3.0.5-1) to root...
Downloading http://downloads.openwrt.org/releases/19.07.7/packages/arm_cortex-a7_neon-vfpv4/packages/htop_3.0.5-1_arm_cortex-a7_neon-vfpv4.ipk
Installing terminfo (6.1-5) to root...
Downloading http://downloads.openwrt.org/releases/19.07.7/packages/arm_cortex-a7_neon-vfpv4/base/terminfo_6.1-5_arm_cortex-a7_neon-vfpv4.ipk
Installing libncurses6 (6.1-5) to root...
Downloading http://downloads.openwrt.org/releases/19.07.7/packages/arm_cortex-a7_neon-vfpv4/base/libncurses6_6.1-5_arm_cortex-a7_neon-vfpv4.ipk
Configuring terminfo.
Configuring libncurses6.
Configuring htop.
```

bolemo, 2021-03-21:
> Hi @bolemo
> I just tried with a RaspberryPI 2, so cortex A7 but same instructions ARMv7-A and everything work fine with latest OpenWrt snapshot / htop 3.0.5-1
> Can you try to remove `/root/.config/htop/htoprc` ?

Yes (that was one of the first thing I tried).
It does not change anything.

So the bug is not in OpenWRT, so it is coming from differences between OpenWRT and Entware environment (libs, etc...)?
Any debug mode that could help?

champtar, 2021-03-21:
If you can I would try OpenWrt 19.07 to be sure, RPI2 has the same instruction set but likely many differences

bolemo, 2021-03-21:
> If you can I would try OpenWrt 19.07 to be sure, RPI2 has the same instruction set but likely many differences

Unfortunately, I only have one router, and I can’t mess too much with it...
Maybe I can figure what changed in the code between the two versions.

zyxmon, 2021-03-27:
@bolemo - entware has its own issues list here https://github.com/Entware/Entware/issues

zyxmon, 2021-03-27:
Try this - https://github.com/htop-dev/htop/issues/534#issuecomment-783322981

bolemo, 2021-03-27:
> Try this - [htop-dev/htop#534 (comment)](https://github.com/htop-dev/htop/issues/534#issuecomment-783322981)

Thank you. This does not work for me, as /proc is not a symlink in R7800 firmware.
However, since the problem is similar and appears from the same version, it is likely related somehow. I will post in the Entware place now.
Thank you again !

bolemo, 2021-03-28:
Problem identified, and fixed in Entware.

Since it is not directly OpenWRT related, and does not seem to affect OpenWRT releases, I will close this issue.

closed 2021-03-28 by bolemo

champtar, 2021-03-28:
Nice to hear, can you link to the fix / explanation ?

zyxmon, 2021-03-28:
>> Nice to hear, can you link to the fix / explanation ?
https://github.com/Entware/entware-packages/commit/2a2e578b095be74e8cfabbf7d00f168b84ba5bdb