openwrt/packages #15172: spice not compiling

neheb, 2021-03-20:
Maintainer: @yousong 
Environment: snapshot buildbots. https://downloads.openwrt.org/snapshots/faillogs/arm_cortex-a15_neon-vfpv4/packages/spice/compile.txt

ping @dhewg I don't know if the issue is the update to meson 57.1 or the update to Debian 10.

dhewg, 2021-03-21:
Even if I revert the recent 3 meson commits it doesn't build for me:

```
Traceback (most recent call last):
  File "/home/andre/src/openwrt/staging_dir/hostpkg/lib/meson/meson.py", line 26, in <module>
    from mesonbuild import mesonmain
  File "/tmp/openwrt/staging_dir/hostpkg/lib/meson/mesonbuild/mesonmain.py", line 25, in <module>
    from . import mconf, mdist, minit, minstall, mintro, msetup, mtest, rewriter, msubprojects, munstable_coredata, mcompile
  File "/tmp/openwrt/staging_dir/hostpkg/lib/meson/mesonbuild/mconf.py", line 16, in <module>
    from . import coredata, environment, mesonlib, build, mintro, mlog
  File "/tmp/openwrt/staging_dir/hostpkg/lib/meson/mesonbuild/environment.py", line 35, in <module>
    from . import compilers
  File "/tmp/openwrt/staging_dir/hostpkg/lib/meson/mesonbuild/compilers/__init__.py", line 133, in <module>
    from .c import (
  File "/tmp/openwrt/staging_dir/hostpkg/lib/meson/mesonbuild/compilers/c.py", line 19, in <module>
    from ..mesonlib import MachineChoice, MesonException, mlog, version_compare
ImportError: cannot import name 'mlog' from 'mesonbuild.mesonlib' (/tmp/openwrt/staging_dir/hostpkg/lib/meson/mesonbuild/mesonlib/__init__.py)
make[2]: *** [Makefile:85: /home/andre/src/openwrt/build_dir/target-mipsel_24kc_musl/spice-0.14.3/.configured_a17fb5ef857664f03cd0ce37cc5ea591] Error 1
make[2]: Leaving directory '/home/andre/src/openwrt/feeds/packages/libs/spice'
time: package/feeds/packages/spice/compile#0.39#0.05#0.42
    ERROR: package/feeds/packages/spice failed to build.
```

neheb, 2021-03-21:
What is this mlog?

neheb, 2021-03-22:
I'm getting this issue...

```
Directory does not contain a valid build tree:
/home/mangix/devstuff/openwrt/build_dir/target-mipsel_24kc_musl/pkgconf-1.7.4/openwrt-build
```

I reverted https://github.com/openwrt/packages/pull/15148/commits/f1c40e3840f6f51b3cf6a74331bd128eeb8b3961 and it compiles fine now.

neheb, 2021-04-05:
Fixed with https://github.com/openwrt/packages/commit/42baf1bac5932d54d9a19f5ce28001c0d745731a .

Seems upstream found several more bugs relating to this.

closed 2021-04-05 by neheb